dresden-elektronik/deconz-rest-plugin #6517: Human Presence Detector - ZY-M100-S

McKenzer, 2022-11-09:
## Device

- Product name: Human Presence Detector - ZY-M100-S
- Manufacturer: _TZE200_ztc6ggyl
- Model identifier: TS0601
- Device type : Sensor

## Screenshots

### Basic
[image]
[image]

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2022-11-09:
Ok so DDF to test

```
{
   "schema":"devcap1.schema.json",
   "manufacturername": "_TZE200_ztc6ggyl",
   "modelid":"TS0601",
   "product":"Human Presence Detector - ZY-M100-S",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_PRESENCE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0xef00"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name": "config/duration",
               "parse": {"fn": "tuya", "dpid": 101, "eval": "Item.val = Attr.val;" },
               "read": {"fn": "tuya"},
               "default": 0
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"config/sensitivity",
               "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = Attr.val;" },
               "read": {"fn": "none"},
               "default": false
            },
            {
               "name":"state/presence",
               "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
               "read": {"fn": "none"},
               "default": false
            }
         ]
      }
   ]
}
```

Only the state/presence is working
congif/delay and config/sensitivity are here for test, pls check if value are updated, I will enable the write feature.

Mimiix, 2022-11-09:
Can you please add all screenshots ?

McKenzer, 2022-11-09:
[image]

[image]
[image]

I have now inserted the ddf file

how can i test the sensor now?

in home assistant i can see the sensor but it doesn't work

I can't even see the sensor in the phoscon web app

[image]
[image]

Smanar, 2022-11-09:
I m not sure it have worked, for me the name need to change from plug to presence.

Have you used in the menu the "Hot reload" ? and from my memory deconz need to be in permit join "add new sensor") else can re-include the device.

You can see direclty the API in phoscon/help/API Information/sensor

McKenzer, 2022-11-09:
I have now deleted the sensor and reconnected it

[image]

[image]

Smanar, 2022-11-10:
Ok so config/duration and config/sensitivty are still 0 ?
The state/presence is working ?

Smanar, 2022-11-10:
Ok so nothing is working.

Can you take a look in log with "info" to find log with "TY_DATA_RESPONSE' for this device ?

McKenzer, 2022-11-10:
I still have the log running. I have now filtered for the device and found these entries.
I last checked at 7:05 p.m

`18:44:48:868 Node: id: 67, 94:34:69:ff:fe:61:a8:c6 (0x943469FFFE61A8C6) scene: -565,447184, -536,582121
18:44:49:044 Tuya debug 7 : Missing manufacture name for 0x943469fffe61a8c6
18:44:49:210 DEV found DDF for 0x943469FFFE61A8C6, path: /home/daniel/.local/share/dresden-elektronik/deCONZ/devices/Human Presence Detector - ZY-M100-S.json
18:44:49:213 DEV found DDF for 0x943469FFFE61A8C6, path: /home/daniel/.local/share/dresden-elektronik/deCONZ/devices/Human Presence Detector - ZY-M100-S.json
18:45:41:283 APS-DATA.request id: 97, addrmode: 0x03, addr: 0x943469fffe61a8c6, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
18:45:41:283 APS-DATA.request id: 98, addrmode: 0x03, addr: 0x943469fffe61a8c6, profile: 0x0000, cluster: 0x0033, ep: 0x00 -> 0x00 queue: 2 len: 2 tx.options 0x04
18:45:51:284 APS-DATA.request id: 166, addrmode: 0x03, addr: 0x943469fffe61a8c6, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
18:46:01:421 APS-DATA.request id: 9, addrmode: 0x03, addr: 0x943469fffe61a8c6, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
18:46:11:560 APS-DATA.request id: 79, addrmode: 0x03, addr: 0x943469fffe61a8c6, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
18:46:21:695 APS-DATA.request id: 173, addrmode: 0x03, addr: 0x943469fffe61a8c6, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
18:46:31:826 max transmit errors for node 0x943469FFFE61A8C6, last seen by neighbors 102 s
18:46:31:828 APS-DATA.request id: 244, addrmode: 0x03, addr: 0x943469fffe61a8c6, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
18:46:36:287 APS-DATA.request id: 15, addrmode: 0x03, addr: 0x943469fffe61a8c6, profile: 0x0000, cluster: 0x0033, ep: 0x00 -> 0x00 queue: 2 len: 2 tx.options 0x04
18:46:36:297 max transmit errors for node 0x943469FFFE61A8C6, last seen by neighbors 107 s
18:46:41:964 max transmit errors for node 0x943469FFFE61A8C6, last seen by neighbors 113 s
18:46:41:966 APS-DATA.request id: 69, addrmode: 0x03, addr: 0x943469fffe61a8c6, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
18:46:42:093 Node zombie state changed 0x943469fffe61a8c6`

Smanar, 2022-11-10:
You are sure there is not a problem with device ?
If you look at the second capture you have a cluster 0x0006 and this cluster is visible on your log

> 18:46:01:421 APS-DATA.request id: 9, addrmode: 0x03, addr: 0x943469fffe61a8c6, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 

But this cluster is not present on the capture on the first post.

McKenzer, 2022-11-10:
yes sorry i forget to say that i have removed the node for testing :D

Smanar, 2022-11-10:
Ok, but it don't explain the moving cluster, and why the device never use the tuya cluster ...

McKenzer, 2022-11-10:
[image]

[image]

McKenzer, 2022-11-10:
now i have installed deconz in a new container and only connected the sensor and changed the ddf file.
unfortunately the same result.

McKenzer, 2022-11-10:
new log entries

20:34:53:850 No Tuya productId entry found for manufacturername: _TZE200_ztc6ggyl
20:34:53:851 Tuya debug Request : Address 0x943469FFFE61A8C6, Endpoint 0x01, Command 0x02, Payload 00210902000400000060
20:34:53:851 Tuya debug 4 : Address 0x943469FFFE61A8C6 Payload 00210902000400000060
20:34:53:851 Tuya debug 5 : Status: 0 Transid: 33 Dp: 521 (0x02,0x09) Fn: 0 Data 96

20:34:42:865 Websocket 192.168.178.99:45012 send message: {"attr":{"id":"28","lastannounced":null,"lastseen":"2022-11-10T19:34Z","manufacturername":"_TZE200_ztc6ggyl","modelid":"TS0601","name":"Smart plug 28","swversion":null,"type":"Smart plug","uniqueid":"94:34:69:ff:fe:61:a8:c6-01"},"e":"changed","id":"28","r":"lights","t":"event","uniqueid":"94:34:69:ff:fe:61:a8:c6-01"} (ret = 318)

20:34:42:868 Websocket 192.168.178.99:45012 send message: {"attr":{"id":"82","lastannounced":null,"lastseen":"2022-11-10T19:34Z","manufacturername":"_TZE200_ztc6ggyl","modelid":"TS0601","name":"Presence 82","swversion":null,"type":"ZHAPresence","uniqueid":"94:34:69:ff:fe:61:a8:c6-01-ef00"},"e":"changed","id":"82","r":"sensors","t":"event","uniqueid":"94:34:69:ff:fe:61:a8:c6-01-ef00"} (ret = 328)

McKenzer, 2022-11-10:
sorry for the spamming :D

I have updated my conbee 2 and deconz to dev.
Now i have new log and the sensor has change his name

[image]

now I can also see TY_DATA_RESPONSE

https://pastebin.com/fy3x1KZq

Smanar, 2022-11-11:
I m not understand what you have done to have change.

>I have updated my conbee 2 and deconz to dev.

There is a dev version ? or you mean have installed the dev package ? but this package don't have impact on deconz itself ?

>Now i have new log and the sensor has change his name

And on your log the device was called "Presence 82", the name did not have changed previously too ?

>21:27:31:917 TY_DATA_REPORT: seq 221, dpid: 0x09, type: 0x02, length: 4, val: 95

I can'see the device that have made this request but this one (the dpid = 9) is the target distance for this device.
So it seem better, the value are still 0 or "none" ATM if you look inside the API (using phoscon/help/API information /ect ...)

Smanar, 2022-11-11:
```
{
   "schema":"devcap1.schema.json",
   "manufacturername": "_TZE200_ztc6ggyl",
   "modelid":"TS0601",
   "product":"Human Presence Detector - ZY-M100-S",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_PRESENCE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0xef00"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name": "config/duration",
               "parse": {"fn": "tuya", "dpid": 101, "eval": "Item.val = Attr.val / 10;" },
               "write": {"fn": "tuya", "dpid": 101, "dt": "0x2b", "eval": "Item.val = 10* Attr.val;"},
               "read": {"fn": "tuya"},
               "default": 0
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"config/sensitivity",
               "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = Attr.val;" },
               "write": {"fn": "tuya", "dpid": 2, "dt": "0x2b", "eval": "Item.val == Attr.val;"},
               "read": {"fn": "none"},
               "default": 0
            },
            {
               "name":"state/presence",
               "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
               "read": {"fn": "none"},
               "default": false
            }
         ]
      }
   ]
}
```

With this DDF you are able to set value you want you want for sensitivity and duration (if it work ofc)

BabaIsYou, 2022-11-23:
> There is a setting called "fading time" but I don't see what is it

It seems to be used to parameter the clearing delay [(see review of this device here)](https://smarthomescene.com/reviews/tuya-zigbee-human-presence-sensor-zy-m100-review/), then same as "duration" I guess ;-)

Smanar, 2022-11-23:
"Fading time" is 102.
I m using 101 for duration, it s "detection delay".
I can be wrong for the choice.

BabaIsYou, 2022-11-23:
As i understood about this device the « dectection delay » is related to the delay before reporting detected presence, « fading time » is related to the delay before resetting previous detection.

Smanar, 2022-11-24:
Ok so I think it's better to test this DDF

```
{
   "schema":"devcap1.schema.json",
   "manufacturername": "_TZE200_ztc6ggyl",
   "modelid":"TS0601",
   "product":"Human Presence Detector - ZY-M100-S",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_PRESENCE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0xef00"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name": "config/duration",
               "parse": {"fn": "tuya", "dpid": 102, "eval": "Item.val = Attr.val;" },
               "write": {"fn": "tuya", "dpid": 102, "dt": "0x2b", "eval": "Item.val;"},
               "read": {"fn": "tuya"},
               "default": 0
            },
            {
               "name": "config/delay",
               "parse": {"fn": "tuya", "dpid": 101, "eval": "Item.val = Attr.val/10;" },
               "write": {"fn": "tuya", "dpid": 101, "dt": "0x2b", "eval": "10*Item.val;"},
               "read": {"fn": "none"},
               "default": 0
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"config/sensitivity",
               "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = Attr.val;" },
               "write": {"fn": "tuya", "dpid": 2, "dt": "0x2b", "eval": "Item.val;"},
               "read": {"fn": "none"},
               "default": 0
            },
            {
               "name":"state/presence",
               "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
               "read": {"fn": "none"},
               "default": false
            }
         ]
      }
   ]
}
```

BabaIsYou, 2022-11-24:
Thanks. Will let @McKenzer try it, I've not yet received mine at this time ;-/

Smanar, 2022-11-24:
I have updated the previous post.
Only the state/lightlevel is enabled (need to check if working)

BabaIsYou, 2022-12-06:
> , I've not yet received mine at this time ;-/

I've received it today ;-)
But it's not exactly the same, it's _TZE204_ztc6ggyl (not TZE200) and model ZY-M100 (not ZY-M100-S). But all clusters are the same apparently.

Trying then adapted DDF
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE204_ztc6ggyl",
  "modelid": "TS0601",
  "product": "Human Presence Detector - ZY-M100",
  "sleeper": true,
  "status": "Silver",
  "path": "/devices/ZY-M100_human_breathing_presence.json",
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0400"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 84000,
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
[... 205 lines cut ...]

```
{
    "config": {
        "duration": 0,
        "on": true,
        "reachable": true
    },
    "etag": "fbe9c5a00605d0c4c7bf2f986fce695c",
    "lastannounced": "2022-12-06T01:12:56Z",
    "lastseen": "2022-12-06T01:33Z",
    "manufacturername": "_TZE204_ztc6ggyl",
    "modelid": "TS0601",
    "name": "Presence 61",
    "state": {
        "lastupdated": "none",
        "presence": null
    },
    "type": "ZHAPresence",
    "uniqueid": "70:ac:08:ff:fe:69:80:b9-ff-ef00"
}
```
BTW here are the parameter, values and range for this device
[image]
I surely missed something ... ;-)

BabaIsYou, 2022-12-06:
> You have again the endpoint 0xFF issue,

Didn't point it ... you're right. Weird because for another device the API display both of them, regular and 0xFF sensors.

> ```
> 20:13:02:135 70:ac:08:ff:fe:69:80:b9-01-ef00/config/sensitivity expression: Item.val = Attr.val/10; --> 0.1
> ```
> 
> 
>     
>   
> 
> On my DDF this value is not / 10 so 1 can be a normal value

Not sure about this because default value is 7 when looking at "documentation"

> DBG_Printf(DBG_INFO, "TY_DATA_%s: seq %u, dpid: 0x%02X, type: 0x%02X, length: %u, val: %d\n",

I've a lot a lines looking like this, but not exactly : 
```
12:37:11:726 APS-DATA.indication srcAddr: 0x2caf, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 212, rssi: -80
12:37:11:727 TY_DATA_REPORT: seq 30, dpid: 0x09, type: 0x02, length: 4, val: 78
12:37:11:728 TY_DATA_REPORT: seq 30, dpid: 0x09, type: 0x02, length: 4, val: 78
12:37:11:728 TY_DATA_REPORT: seq 30, dpid: 0x09, type: 0x02, length: 4, val: 78
12:37:11:729 TY_DATA_REPORT: seq 30, dpid: 0x09, type: 0x02, length: 4, val: 78
12:37:11:730 TY_DATA_REPORT: seq 30, dpid: 0x09, type: 0x02, length: 4, val: 78
12:37:11:730 APS-DATA.request id: 95, addrmode: 0x02, addr: 0x2caf, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 4 len: 5 tx.options 0x04
12:37:11:731 APS-DATA.request id: 96, addrmode: 0x03, addr: 0x00124b0025105427, profile: 0x0000, cluster: 0x0033, ep: 0x00 -> 0x00 queue: 5 len: 2 tx.options 0x04
12:37:11:761 APS-DATA.indication srcAddr: 0x2caf, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -76
12:37:11:763 TY_DATA_REPORT: seq 30, dpid: 0x09, type: 0x02, length: 4, val: 78
12:37:11:765 TY_DATA_REPORT: seq 30, dpid: 0x09, type: 0x02, length: 4, val: 78
12:37:11:765 TY_DATA_REPORT: seq 30, dpid: 0x09, type: 0x02, length: 4, val: 78
12:37:11:766 TY_DATA_REPORT: seq 30, dpid: 0x09, type: 0x02, length: 4, val: 78
12:37:11:766 TY_DATA_REPORT: seq 30, dpid: 0x09, type: 0x02, length: 4, val: 78
12:37:11:767 APS-DATA.request id: 98, addrmode: 0x02, addr: 0x2caf, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 6 len: 5 tx.options 0x04
12:37:11:802 APS-DATA.confirm id: 89, status: 0x00 SUCCESS
12:37:11:803 APS-DATA.confirm request id: 89 -> erase from queue
12:37:11:837 APS-DATA.confirm id: 91, status: 0x00 SUCCESS
12:37:11:838 APS-DATA.confirm request id: 91 -> erase from queue
12:37:11:844 APS-DATA.confirm id: 93, status: 0x00 SUCCESS
12:37:11:846 APS-DATA.confirm request id: 93 -> erase from queue
12:37:11:861 APS-DATA.indication srcAddr: 0x2caf, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 224, rssi: -79
12:37:11:862 APS-DATA.request id: 89 erase from queue
12:37:11:863 APS-DATA.request id: 91 erase from queue
12:37:11:863 APS-DATA.request id: 93 erase from queue
12:37:11:864 TY_DATA_REPORT: seq 31, dpid: 0x09, type: 0x02, length: 4, val: 94
12:37:11:865 TY_DATA_REPORT: seq 31, dpid: 0x09, type: 0x02, length: 4, val: 94
12:37:11:865 TY_DATA_REPORT: seq 31, dpid: 0x09, type: 0x02, length: 4, val: 94
12:37:11:865 TY_DATA_REPORT: seq 31, dpid: 0x09, type: 0x02, length: 4, val: 94
12:37:11:866 TY_DATA_REPORT: seq 31, dpid: 0x09, type: 0x02, length: 4, val: 94
12:37:11:866 APS-DATA.request id: 101, addrmode: 0x02, addr: 0x2caf, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 4 len: 5 tx.options 0x04

```

I can filter on NWK Address 0x2caf which is my device, then I got : 

```
12:37:09:663 APS-DATA.indication srcAddr: 0x2caf, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 156, rssi: -80
12:37:09:667 APS-DATA.request id: 67, addrmode: 0x02, addr: 0x2caf, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x04
12:37:09:886 APS-DATA.indication srcAddr: 0x2caf, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 192, rssi: -81
12:37:10:118 APS-DATA.request id: 76, addrmode: 0x02, addr: 0x2caf, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x04
12:37:10:160 APS-DATA.indication srcAddr: 0x2caf, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -74
12:37:10:170 APS-DATA.request id: 79, addrmode: 0x02, addr: 0x2caf, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 3 len: 5 tx.options 0x04
12:37:10:269 APS-DATA.indication srcAddr: 0x2caf, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -78
12:37:10:282 APS-DATA.request id: 84, addrmode: 0x02, addr: 0x2caf, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 4 len: 5 tx.options 0x04
12:37:11:646 APS-DATA.indication srcAddr: 0x2caf, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -76
12:37:11:655 APS-DATA.request id: 91, addrmode: 0x02, addr: 0x2caf, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x04
12:37:11:730 APS-DATA.request id: 95, addrmode: 0x02, addr: 0x2caf, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 4 len: 5 tx.options 0x04
```
But not sure is quite revelant ...

BabaIsYou, 2022-12-06:
Have deleted it and rejoining again ... Then no more endpoint 0xFF issue :-)

```
{
  "config": {
    "on": true,
    "reachable": true
  },
  "etag": "91ae9543a18e90cd33ed7a3ed1fad8bb",
  "lastannounced": null,
  "lastseen": "2022-12-06T18:12Z",
  "manufacturername": "_TZE204_ztc6ggyl",
  "modelid": "TS0601",
  "name": "LightLevel 62",
  "state": {
    "lastupdated": "2022-12-06T18:09:23.395",
    "lightlevel": 58
  },
  "swversion": "1.0.6",
  "type": "ZHALightLevel",
  "uniqueid": "70:ac:08:ff:fe:69:80:b9-01-0400",
  "id": "62",
  "deviceclass": "sensors"
}

```

```
{
  "config": {
    "delay": 0,
    "duration": 0,
    "on": true,
    "reachable": true,
    "sensitivity": 0
  },
  "etag": "5853714492d793d1b444405a38697850",
  "lastannounced": null,
  "lastseen": "2022-12-06T18:12Z",
  "manufacturername": "_TZE204_ztc6ggyl",
  "modelid": "TS0601",
  "name": "Presence 63",
  "state": {
    "lastupdated": "2022-12-06T18:09:14.855",
    "presence": true
  },
  "swversion": "1.0.6",
  "type": "ZHAPresence",
  "uniqueid": "70:ac:08:ff:fe:69:80:b9-01-ef00",
  "id": "63",
  "deviceclass": "sensors"
}
```

```
{
  "etag": "d118ab99f91e333567d7efce277c6955",
  "hascolor": false,
  "lastannounced": null,
  "lastseen": "2022-12-06T18:14Z",
  "manufacturername": "_TZE204_ztc6ggyl",
  "modelid": "TS0601",
  "name": "Smart plug 13",
  "state": {
    "alert": "none",
    "on": false,
    "reachable": true
  },
  "swversion": null,
  "type": "Smart plug",
  "uniqueid": "70:ac:08:ff:fe:69:80:b9-01",
  "id": "13",
  "deviceclass": "lights"
}

and here is all the log from insertion. Of course NWK address has changed and now it's 0x9bf3 [pastebin]( https://pastebin.com/snenGguc)
```

BabaIsYou, 2022-12-06:
> No more dpid ?

I found these in the log !

13:09:14:847 APS-DATA.indication srcAddr: 0x9bf3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 208, rssi: -79
13:09:14:848 TY_DATA_REPORT: seq 132, dpid: 0x01, type: 0x04, length: 1, val: 1

13:09:15:027 APS-DATA.indication srcAddr: 0x9bf3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 244, rssi: -78
13:09:15:028 TY_DATA_REPORT: seq 133, dpid: 0x04, type: 0x02, length: 4, val: 600

13:09:15:147 APS-DATA.indication srcAddr: 0x9bf3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 248, rssi: -77
13:09:15:152 TY_DATA_REPORT: seq 134, dpid: 0x02, type: 0x02, length: 4, val: 7

13:09:15:256 APS-DATA.indication srcAddr: 0x9bf3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 244, rssi: -78
13:09:15:258 TY_DATA_REPORT: seq 135, dpid: 0x03, type: 0x02, length: 4, val: 0

13:09:15:355 APS-DATA.indication srcAddr: 0x9bf3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 232, rssi: -78
13:09:15:358 TY_DATA_REPORT: seq 136, dpid: 0x06, type: 0x04, length: 1, val: 1

13:09:15:514 APS-DATA.indication srcAddr: 0x9bf3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 220, rssi: -77
13:09:15:518 TY_DATA_REPORT: seq 137, dpid: 0x65, type: 0x02, length: 4, val: 0

13:09:15:676 APS-DATA.indication srcAddr: 0x9bf3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -76
13:09:15:678 TY_DATA_REPORT: seq 138, dpid: 0x66, type: 0x02, length: 4, val: 600

13:09:15:837 APS-DATA.indication srcAddr: 0x9bf3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -77
13:09:15:839 TY_DATA_REPORT: seq 139, dpid: 0x09, type: 0x02, length: 4, val: 80

13:09:17:488 APS-DATA.indication srcAddr: 0x9bf3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -74
13:09:17:490 TY_DATA_REPORT: seq 141, dpid: 0x68, type: 0x02, length: 4, val: 5

BabaIsYou, 2022-12-07:
I tried this DFF afetr collecting several informatins from this thread : 
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": [
    "_TZE200_ztc6ggyl",
    "_TZE204_ztc6ggyl"
  ],
  "modelid": [
    "TS0601",
    "TS0601"
  ],
  "product": "Human Presence Detector - ZY-M100",
  "sleeper": true,
  "status": "Silver",
  "path": "/devices/ZY-M100_human_breathing_presence.json",
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0400"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 84000,
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lux",
          "description": "The current light intensity in Lux (max is 2000)",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 104,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 84000,
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/errorcode",
          "description": "Self check result code. \n0 = checking, 1 = check_success, 2 = check_failure, 3 = others, 4 = comm_fault, 5 = radar_fault",
           "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 6,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": "1"
        },
        {
          "name": "state/angle",
          "description": "Distance from detected target in cm (from 0 to 1000).",
           "read": {
            "fn": "none"
          },
           "parse": {
            "dpid": 9,
            "eval": "Item.va; = Attr.val;",
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "config/delay",
          "description": "Detection delay : target confirmation delay time in 10th of second (from 0 to 100). Default 0.1s",
          "read": {
            "fn": "none"
          },
          "write": {
            "dpid": 101,
            "dt": "0x2b",
            "eval": "Item.val;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 101,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 1
        },
        {
          "name": "config/duration",
          "description": "Fading time : target disapperance delay time in second (from 0 to 15000). Default 90s",
          "read": {
            "fn": "tuya"
          },
          "write": {
            "dpid": 102,
            "dt": "0x2b",
            "eval": "Item.val;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 102,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 90
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/sensitivity",
          "description": "Radar sensitivity (from 1 to 9). Default 7",
          "read": {
            "fn": "none"
          },
          "write": {
            "dpid": 2,
            "dt": "0x2b",
            "eval": "10*Item.val;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 2,
            "eval": "Item.val = Attr.val/10;",
            "fn": "tuya"
          },
          "default": 7
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 1,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0xEF00"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006"
    }
  ]
}

```
Will come back soon to see if values continues to be weird in json ...

BabaIsYou, 2022-12-08:
Data continue to get some strange values without doing anything with the device, could it be a wrong data type (even when reading) because dpis seems coorect when looking at logs ?
```
{
  "config": {
    "delay": 1968,
    "duration": 0,
    "on": true,
    "reachable": true,
    "sensitivity": 564
  },
  "etag": "7f32a46a0f9cf65db0fd451c3e89b768",
  "lastannounced": null,
  "lastseen": "2022-12-08T15:50Z",
  "manufacturername": "_TZE204_ztc6ggyl",
  "modelid": "TS0601",
  "name": "Presence Cinema",
  "state": {
    "angle": 0,
    "errorcode": "1",
    "lastupdated": "2022-12-08T15:50:31.080",
    "presence": false
  },
  "swversion": "1.0.6",
  "type": "ZHAPresence",
  "uniqueid": "70:ac:08:ff:fe:69:80:b9-01-ef00",
  "id": "63",
  "deviceclass": "sensors"
}
```

Smanar, 2022-12-08:
```
15:10:53:119 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
15:10:53:120 70:ac:08:ff:fe:69:80:b9-01-ef00/config/sensitivity expression: Item.val = Attr.val/10; --> 8388608.5
```

Yes It look like a wrong type convertion, you are using wich one deconz version ATM, realy old version already have this bug, but its realy old one.

But can you try without the "/10", the operation is 7/10, so can have a problem with float/int convertion value ?

BabaIsYou, 2022-12-08:
> realy old version already have this bug, but its realy old one.

V2.15.2-beta ... Not so old ... may be I'm wrong ;-)
Always a dilemna : updating with no assurance that it will be corrected and loosing a 7 months old PR not yet included, returning material ... or changing of interface that already support all of my stuff ... 
Keep trying to understand before changing DeConz.

> BTW can you enable the flag "APS" ? like that we can read the raw zigbee request just before, and see if the value is realy "7".

I already enabled it I thought ... Do you mean something like this : 
```
15:10:52:907 APS-DATA.indication srcAddr: 0x9bf3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -69
15:10:52:913 TY_DATA_REPORT: seq 161, dpid: 0x01, type: 0x04, length: 1, val: 0
15:10:52:914 TY_DATA_REPORT: seq 161, dpid: 0x01, type: 0x04, length: 1, val: 0
15:10:52:916 TY_DATA_REPORT: seq 161, dpid: 0x01, type: 0x04, length: 1, val: 0
15:10:52:917 TY_DATA_REPORT: seq 161, dpid: 0x01, type: 0x04, length: 1, val: 0
15:10:52:918 TY_DATA_REPORT: seq 161, dpid: 0x01, type: 0x04, length: 1, val: 0
15:10:52:921 70:ac:08:ff:fe:69:80:b9-01-ef00/state/presence expression: Item.val = Attr.val; --> 0
15:10:52:922 TY_DATA_REPORT: seq 161, dpid: 0x01, type: 0x04, length: 1, val: 0
15:10:52:923 TY_DATA_REPORT: seq 161, dpid: 0x01, type: 0x04, length: 1, val: 0
15:10:52:925 APS-DATA.request id: 191, addrmode: 0x02, addr: 0x9bf3, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x04
15:10:53:003 APS-DATA.confirm id: 191, status: 0x00 SUCCESS
15:10:53:004 APS-DATA.confirm request id: 191 -> erase from queue
15:10:53:024 APS-DATA.indication srcAddr: 0x9bf3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -69
15:10:53:028 APS-DATA.request id: 191 erase from queue
15:10:53:033 TY_DATA_REPORT: seq 162, dpid: 0x04, type: 0x02, length: 4, val: 600
15:10:53:034 TY_DATA_REPORT: seq 162, dpid: 0x04, type: 0x02, length: 4, val: 600
15:10:53:035 TY_DATA_REPORT: seq 162, dpid: 0x04, type: 0x02, length: 4, val: 600
15:10:53:036 TY_DATA_REPORT: seq 162, dpid: 0x04, type: 0x02, length: 4, val: 600
15:10:53:038 TY_DATA_REPORT: seq 162, dpid: 0x04, type: 0x02, length: 4, val: 600
15:10:53:039 TY_DATA_REPORT: seq 162, dpid: 0x04, type: 0x02, length: 4, val: 600
15:10:53:040 TY_DATA_REPORT: seq 162, dpid: 0x04, type: 0x02, length: 4, val: 600
15:10:53:041 APS-DATA.request id: 193, addrmode: 0x02, addr: 0x9bf3, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x04
15:10:53:097 APS-DATA.confirm id: 193, status: 0x00 SUCCESS
15:10:53:098 APS-DATA.confirm request id: 193 -> erase from queue
15:10:53:112 APS-DATA.indication srcAddr: 0x9bf3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -69
15:10:53:114 APS-DATA.request id: 193 erase from queue
15:10:53:117 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
15:10:53:117 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
15:10:53:118 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
15:10:53:119 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
15:10:53:120 70:ac:08:ff:fe:69:80:b9-01-ef00/config/sensitivity expression: Item.val = Attr.val/10; --> 8388608.5
15:10:53:120 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
15:10:53:479 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
15:10:53:480 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
15:10:53:481 APS-DATA.request id: 195, addrmode: 0x02, addr: 0x9bf3, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x04
15:10:53:530 APS-DATA.request id: 197, addrmode: 0x03, addr: 0x0022a30000253339, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x00
15:10:53:556 APS-DATA.indication srcAddr: 0x9bf3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -72
15:10:53:561 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
15:10:53:562 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
15:10:53:563 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
15:10:53:563 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
15:10:53:564 70:ac:08:ff:fe:69:80:b9-01-ef00/config/sensitivity expression: Item.val = Attr.val/10; --> 8388608.5
15:10:53:565 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
15:10:53:566 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
15:10:53:566 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
15:10:53:567 APS-DATA.request id: 199, addrmode: 0x02, addr: 0x9bf3, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 3 len: 5 tx.options 0x04
15:10:53:595 APS-DATA.confirm id: 195, status: 0x00 SUCCESS
15:10:53:596 APS-DATA.confirm request id: 195 -> erase from queue
15:10:53:613 APS-DATA.indication srcAddr: 0x9bf3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -68
15:10:53:615 APS-DATA.request id: 195 erase from queue
15:10:53:619 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
15:10:53:621 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
15:10:53:623 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
15:10:53:624 TY_DATA_REPORT: seq 163, dpid: 0x02, type: 0x02, length: 4, val: 7
[... cut ...]
15:10:53:796 TY_DATA_REPORT: seq 164, dpid: 0x03, type: 0x02, length: 4, val: 0
15:10:53:797 TY_DATA_REPORT: seq 164, dpid: 0x03, type: 0x02, length: 4, val: 0
15:10:53:798 TY_DATA_REPORT: seq 164, dpid: 0x03, type: 0x02, length: 4, val: 0
15:10:53:800 TY_DATA_REPORT: seq 164, dpid: 0x03, type: 0x02, length: 4, val: 0
15:10:53:801 TY_DATA_REPORT: seq 164, dpid: 0x03, type: 0x02, length: 4, val: 0
15:10:53:802 APS-DATA.request id: 207, addrmode: 0x02, addr: 0x9bf3, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 4 len: 5 tx.options 0x04
```

Smanar, 2022-12-09:
My bad it's APS + APS_L2
`
16:40:49:824 	asdu: 181301004086`

Ihis debug line is the raw zigbee request, you will see inside the dp , the value the type and ect .... but I think the problem is not here.
And I can understand your problem, so long time you are waiting for the merge.

@McKenzer can you try the DDF on a recent deconz version ? 2.18.X ? (avoid the 2.19.X for the moment)

BabaIsYou, 2022-12-09:
Here is an extract with APS+APS_L2+DDF. It seems only activate the DDF for attribute "angle" that I used to get "dpid": 9 that is "distance from target" ... And without any motion the values are strange ...

```
70:ac:08:ff:fe:69:80:b9-01-ef0011:44:38:012 70:ac:08:ff:fe:69:80:b9-01-ef00/state/angle expression: Item.val = Attr.val; --> 248
11:44:38:073 70:ac:08:ff:fe:69:80:b9-01-ef00/state/angle expression: Item.val = Attr.val; --> 248
11:44:38:104 70:ac:08:ff:fe:69:80:b9-01-ef00/state/angle expression: Item.val = Attr.val; --> 248
11:44:38:128 70:ac:08:ff:fe:69:80:b9-01-ef00/state/angle expression: Item.val = Attr.val; --> 248
11:44:38:215 50:0b:91:40:00:01:cb:9b-01-0402/state/temperature expression: Item.val = Attr.val + R.item('config/offset').val --> 1850
11:44:38:572 APS-DATA.confirm id: 184, status: 0x00 SUCCESS
11:44:38:575 APS-DATA.confirm request id: 184 -> erase from queue
11:44:38:596 APS-DATA.confirm id: 187, status: 0x00 SUCCESS
11:44:38:598 APS-DATA.confirm request id: 187 -> erase from queue
11:44:38:614 APS-DATA.confirm id: 189, status: 0x00 SUCCESS
11:44:38:616 APS-DATA.confirm request id: 189 -> erase from queue
11:44:38:628 APS-DATA.indication srcAddr: 0x9bf3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -55
11:44:38:630 APS-DATA.request id: 184 erase from queue
11:44:38:632 APS-DATA.request id: 187 erase from queue
11:44:38:633 APS-DATA.request id: 189 erase from queue
11:44:38:644 70:ac:08:ff:fe:69:80:b9-01-ef00/state/angle expression: Item.val = Attr.val; --> 1850
11:44:38:663 APS-DATA.request id: 196, addrmode: 0x02, addr: 0x9bf3, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 3 len: 5 tx.options 0x04
11:44:38:706 APS-DATA.indication srcAddr: 0x9bf3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -55
11:44:38:721 70:ac:08:ff:fe:69:80:b9-01-ef00/state/angle expression: Item.val = Attr.val; --> 1850
11:44:38:722 APS-DATA.request id: 199, addrmode: 0x02, addr: 0x9bf3, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 4 len: 5 tx.options 0x04
11:44:38:736 APS-DATA.indication srcAddr: 0x9bf3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -54
11:44:38:751 70:ac:08:ff:fe:69:80:b9-01-ef00/state/angle expression: Item.val = Attr.val; --> 1850
11:44:38:752 APS-DATA.request id: 201, addrmode: 0x02, addr: 0x9bf3, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 5 len: 5 tx.options 0x04
11:44:38:803 APS-DATA.confirm id: 191, status: 0x00 SUCCESS
11:44:38:804 APS-DATA.confirm request id: 191 -> erase from queue
11:44:38:837 APS-DATA.confirm id: 196, status: 0x00 SUCCESS
11:44:38:839 APS-DATA.confirm request id: 196 -> erase from queue
11:44:38:852 APS-DATA.confirm id: 199, status: 0x00 SUCCESS
11:44:38:854 APS-DATA.confirm request id: 199 -> erase from queue
11:44:38:874 aps request id: 191 finished, erase from queue
11:44:38:894 APS-DATA.confirm id: 201, status: 0x00 SUCCESS
11:44:38:895 APS-DATA.confirm request id: 201 -> erase from queue
11:44:38:953 aps request id: 196 finished, erase from queue
11:44:39:033 aps request id: 199 finished, erase from queue
11:44:39:113 aps request id: 201 finished, erase from queue
11:44:39:287 APS-DATA.indication srcAddr: 0x9bf3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -55
11:44:39:304 70:ac:08:ff:fe:69:80:b9-01-ef00/state/angle expression: Item.val = Attr.val; --> 1850
11:44:39:306 APS-DATA.request id: 205, addrmode: 0x02, addr: 0x9bf3, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x04
11:44:39:356 APS-DATA.confirm id: 205, status: 0x00 SUCCESS
11:44:39:358 APS-DATA.confirm request id: 205 -> erase from queue
11:44:39:433 aps request id: 205 finished, erase from queue
11:44:39:605 APS-DATA.request id: 207, addrmode: 0x03, addr: 0xb0ce1814036bcf99, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x00
11:44:39:606 	asdu (length: 5): 1007000000
11:44:39:649 APS-DATA.confirm id: 207, status: 0x00 SUCCESS
11:44:39:651 APS-DATA.confirm request id: 207 -> erase from queue
11:44:39:672 APS-DATA.indication srcAddr: 0x9fd3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -62
11:44:39:674 	asdu: 0807010000001001
11:44:39:675 APS-DATA.request id: 207 erase from queue
11:44:39:692 b0:ce:18:14:03:6b:cf:99-01/state/on expression: Item.val = Attr.val --> true
11:44:39:694 APS-DATA.request id: 210, addrmode: 0x02, addr: 0x9fd3, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x04
11:44:39:696 	asdu (length: 5): 10070b0100
11:44:39:732 APS-DATA.confirm id: 210, status: 0x00 SUCCESS
11:44:39:733 APS-DATA.confirm request id: 210 -> erase from queue
11:44:39:753 aps request id: 210 finished, erase from queue
11:44:40:204 APS-DATA.indication srcAddr: 0xbe7b, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -41
11:44:40:206 	asdu: 08250a00001000
11:44:40:216 b0:ce:18:14:03:6b:d3:71-01/state/on expression: Item.val = Attr.val --> false
11:44:40:217 APS-DATA.request id: 215, addrmode: 0x02, addr: 0xbe7b, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x04
11:44:40:218 	asdu (length: 5): 10250b0a00
11:44:40:247 APS-DATA.confirm id: 215, status: 0x00 SUCCESS
11:44:40:249 APS-DATA.confirm request id: 215 -> erase from queue
[... cut ...]
11:45:19:136 APS-DATA.request id: 89, addrmode: 0x02, addr: 0x9bf3, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 6 len: 5 tx.options 0x04
11:45:19:179 70:ac:08:ff:fe:69:80:b9-01-ef00/state/angle expression: Item.val = Attr.val; --> 1848
11:45:19:202 70:ac:08:ff:fe:69:80:b9-01-ef00/state/angle expression: Item.val = Attr.val; --> 1848
11:45:19:225 70:ac:08:ff:fe:69:80:b9-01-ef00/state/angle expression: Item.val = Attr.val; --> 1848
11:45:19:382 50:0b:91:40:00:01:cb:9b-01-0201/state/valve expression: Item.val = Attr.val; --> 9
11:45:19:448 70:ac:08:ff:fe:69:80:b9-01-ef00/state/angle expression: Item.val = Attr.val; --> 9
```

BabaIsYou, 2022-12-09:
Here is my actual DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": [
    "_TZE200_ztc6ggyl",
    "_TZE204_ztc6ggyl"
  ],
  "modelid": [
    "TS0601",
    "TS0601"
  ],
  "product": "Human Presence Detector - ZY-M100",
  "sleeper": true,
  "status": "Silver",
  "path": "/devices/ZY-M100_human_breathing_presence.json",
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0400"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 84000,
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lux",
          "description": "The current light intensity in Lux (max is 2000)",
          "refresh.interval": 5,
          "read": {
            "fn": "tuya"
          },
          "parse": {
            "dpid": 104,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 83999,
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/delay",
          "description": "Detection delay : target confirmation delay time in 10th of second (from 0 to 100). Default 0.1s",
          "read": {
            "fn": "none"
          },
          "write": {
            "dpid": 101,
            "dt": "0x2b",
            "eval": "Item.val;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 101,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 1
        },
        {
          "name": "config/duration",
          "description": "Fading time : target disapperance delay time in second (from 0 to 15000). Default 90s",
          "read": {
            "fn": "tuya"
          },
          "write": {
            "dpid": 102,
            "dt": "0x2b",
            "eval": "Item.val;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 102,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 90
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/sensitivity",
          "description": "Radar sensitivity (from 1 to 9). Default 7",
          "read": {
            "fn": "none"
          },
          "write": {
            "dpid": 2,
            "dt": "0x2b",
            "eval": "10*Item.val;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 2,
            "eval": "Item.val = Attr.val/10;",
            "fn": "tuya"
          },
          "default": 7
        },
        {
          "name": "state/angle",
          "description": "Distance from detected target in cm (from 0 to 1000).",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 9,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "state/errorcode",
          "description": "Self check result code. \n0 = checking, 1 = check_success, 2 = check_failure, 3 = others, 4 = comm_fault, 5 = radar_fault",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 6,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": "1"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "refresh.interval": 5,
          "read": {
            "fn": "tuya"
          },
          "parse": {
            "dpid": 1,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0xEF00"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006"
    }
  ]
}
```

Smanar, 2022-12-09:
```
11:44:43:521 	asdu: 093302002d090200040000015a
11:44:43:522 APS-DATA.request id: 2 erase from queue
11:44:43:538 70:ac:08:ff:fe:69:80:b9-01-ef00/state/angle expression: Item.val = Attr.val; --> 190
```

093302002d 09 02 00 04 0000015a

DPID 0x09
Type 0x02
0x015a = 346

Hu ? why 190 ?

There is serious convertion issue, but can come from your version, I hope @McKenzer can make a try.

manosioa, 2022-12-18:
Hey guys, 
Tomorrow I will have my hands on 2 of these sensors. Followed through your comments but I did not understand a lot. From what I understood, to make it work in HA I have to import a custom DDF (which I guess is something like a "driver" for deconz?) so that it recognises the devices attributes and functions. 
Is this right? 
If so, how can I import a DDF? 

Thank you :)

BabaIsYou, 2022-12-18:
DDF is a kind of descriptor for DeConz to dialog with devices that are not quite recognized by legacy code.

Not familiar with HA but as understand you have to put json DDF file under /data/.local/share/dresden-elektronik/deCONZ/devices/

Then it will recognized at reload (restart of DeConz)

shlomki, 2022-12-20:
> {
>   "schema": "devcap1.schema.json",
>   "manufacturername": [
>     "_TZE200_ztc6ggyl",
>     "_TZE204_ztc6ggyl"
>   ],
>   "modelid": [
>     "TS0601",
>     "TS0601"
>   ],
>   "product": "Human Presence Detector [...]

I just bought this sensor so I'll be trying the DDF that @BabaIsYou posted. Let me know if you'd like me to test something specific

BabaIsYou, 2022-12-20:
Hi,

which one do you have ? TZE200_ztc6ggyl or _TZE204_ztc6ggyl ?

Are attributes correctly reported ?

shlomki, 2022-12-20:
I have the _TZE204_ztc6ggyl
Looks like the attributes are not reported correctly, the only thing I'm seeing in attributes is:

`on: true`

BabaIsYou, 2022-12-20:
Nothing else in Phoscon ? Like this
```
{
    "config": {
        "on": true,
        "reachable": true
    },
    "etag": "546dd1e1eb3ec15f03e90c1bb8335ae1",
    "lastannounced": null,
    "lastseen": "2022-12-20T13:24Z",
    "manufacturername": "_TZE204_ztc6ggyl",
    "modelid": "TS0601",
    "name": "Luminosité Cinema",
    "state": {
        "lastupdated": "2022-12-20T13:17:19.332",
        "lux": 0
    },
    "swversion": "1.0.6",
    "type": "ZHALightLevel",
    "uniqueid": "70:ac:08:ff:fe:69:80:b9-01-0400"
}
```
and
```
{
    "config": {
        "delay": 2192,
        "duration": 1,
        "on": true,
        "reachable": true,
        "sensitivity": 219
    },
    "etag": "f5d6cb845b1ff60d0877a475fbffbf88",
    "lastannounced": null,
    "lastseen": "2022-12-20T13:30Z",
    "manufacturername": "_TZE204_ztc6ggyl",
    "modelid": "TS0601",
    "name": "Presence Cinema",
    "state": {
        "angle": 1,
        "errorcode": "1",
        "lastupdated": "2022-12-20T13:27:18.484",
        "presence": false
    },
    "swversion": "1.0.6",
    "type": "ZHAPresence",
    "uniqueid": "70:ac:08:ff:fe:69:80:b9-01-ef00"
}
```

manosioa, 2022-12-20:
> DDF is a kind of descriptor for DeConz to dialog with devices that are not quite recognized by legacy code. Not familiar with HA but as understand you have to put json DDF file under /data/.local/share/dresden-elektronik/deCONZ/devices/ Then it will recognized at reload (restart of DeConz)

Thank you for this! I also have _TZE204_ztc6ggyl.
I open the DDF file, click on Hot reload while the "Permit Join is Enabled" but it does not change. It stays as a smart plug.

Am I missing something?

Smanar, 2022-12-20:
> I open the DDF file, click on Hot reload while the "Permit Join is Enabled" but it does not change. It stays as a smart plug.

Yeah ^^ sorry, there is some bad magic ATM, need to set a procedure.

But you still have the "Smart plug 10" and this one is not in the DDF, so I think you have ATM a conflict with the old entry with legacy code, and the new one with the DDF core.

I think if you just delete the device and re-include it, the smart plug will be removed (without action on DDF)

BabaIsYou, 2022-12-20:
> I think if you just delete the device and re-include it, the smart plug will be removed (without action on DDF)

I did it a lot of time and still have the smart plug entry, but may be it's about my version ;-)

I have the same screens that @manosioa but with a version 1.0.6 in the lightlevel sensor.

Smanar, 2022-12-20:
The swversion can happen later, 84000s poll, can take time.

```
        {
          "name": "attr/swversion",
          "refresh.interval": 84000,
          "read": {
            "at": "0x0001",
```

But on recent deconz version the "smart plug" need to be removed

```
    "state": {
        "lastupdated": "2022-12-20T13:17:19.332",
        "lux": 0
```

Ha yes lux is not working on your side.

BabaIsYou, 2022-12-20:
Perhaps a conversion issue, will wait for results from @manosioa because mine are still strange.

manosioa, 2022-12-20:
> Here is my actual DDF
>
> ```
> {
>   "schema": "devcap1.schema.json",
>   "manufacturername": [
>     "_TZE200_ztc6ggyl",
>     "_TZE204_ztc6ggyl"
>   ],
>   "modelid": [
>     "TS0601",
>     "TS0601"
>   ], [...]

Ok, I used the quoted DDF. 

I was hoping you could tell me what you need and where I can find it. I'm a fast learner (in 24h I made it work and learned all about HA's docker containers, SSH shell access, what DDF is, and how to make a custom one and load it on a device.)

HA is reporting only "Delay" and "Duration". Distance, sensitivity and light are not reported.
[image]

What is the next step? What do you need me to try?

BabaIsYou, 2022-12-20:
Well, can you connect to Phoscon, Help, API information, and then select sensors radio button, and by selecting the sensors in the list, catching the information about those related to your device (may be the two last of the list I presume) .
Then you will got something like [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6517#issuecomment-1359352740](url)

[image]

BabaIsYou, 2022-12-20:
If you didn't change default value you should have default values  for sensitivity, duration and delay in the presence sensor : [https://user-images.githubusercontent.com/85300265/205787373-6c18f4dc-f79b-49cf-bc37-2be8b4b170af.png](url).

And these values must be stable. That not my case ;-(

manosioa, 2022-12-20:
> Well, can you connect to Phoscon, Help, API information, and then select sensors radio button, and by selecting the sensors in the list, catching the information about those related to your device [...]

Great, here are my screenshot (should you need the actual logs I can provide them to you)
[image]
[image]

BabaIsYou, 2022-12-20:
By default, sensitivity should be 7. Do you have the same values at each refresh ?

manosioa, 2022-12-20:
Yes, sensitivity is stable at 1 in each and every refresh. I did not change anything regarding the sensitivity (I'm not sure where I can do that).
The other sensor has the same sensitivity as well (1).
[image]

Smanar, 2022-12-20:
The DDF work realy better on his version :(
I m sorry @BabaIsYou but manup is back from holiday, hoping for a massive PR merge soon

BabaIsYou, 2022-12-20:
Yes, it seems more stable, and it's good news BTW. 
If PR #6011 could be included in the next batch that would be perfect ... If not I will upgrade anyway and redo the chhanges manually.

@manosioa coukd you please modify the DFF to : 
1. suppress the /10 in parse section of sensitivity item 
```
          "parse": {
            "dpid": 2,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
```
2.  as @smanar told earlier, you can remove in DDF for lux

`"refresh.interval": 5,`

Then I think that a PR could be made for this device.

manosioa, 2022-12-20:
> Yes, it seems more stable, and it's good news BTW. 
> 
> If PR #6011 could be included in the next batch that would be perfect ... If not I will upgrade anyway and redo the chhanges manually.
> 
> 
> 
> @manosioa coukd you please modify the DFF to : 
> 
> 1. suppress the /10 in parse section of sensitivity item . 1 in place of 7 is may be the cause.
> 
> ```
> 
>           "parse": {
> 
>             "dpid": 2,
> 
>             "eval": "Item.val = Attr.val;",
> 
>             "fn": "tuya"
> 
>           },
> 
> ```
> 
> 2.  as @smanar told earlier, you can remove in DDF for lux
> 
> 
> 
> `"refresh.interval": 5,`
> 
> 
> 
> Then I think that a PR could be made for this device (#6621)
> 
> 
> 
> In fact here is the actual DDF I use : 
> 
> 
> 
> ```
> 
> {
> 
>   "schema": "devcap1.schema.json",
> 
>   "manufacturername": [
> 
>     "_TZE200_ztc6ggyl",
> 
>     "_TZE204_ztc6ggyl"
> 
>   ],
> 
>   "modelid": [
> 
>     "TS0601",
> 
>     "TS0601"
> 
>   ],
> 
>   "product": "Human Presence Detector - ZY-M100",
> 
>   "sleeper": true,
> 
>   "status": "Silver",
> 
>   "subdevices": [
> 
[... 503 lines cut ...]
> 
>     },
> 
>     {
> 
>       "bind": "unicast",
> 
>       "src.ep": 1,
> 
>       "cl": "0x0006"
> 
>     }
> 
>   ]
> 
> }
> 
> 
> 
> ```

I did use the DDF you provided and that led to sensors not updating presence.
I re added them from scratch using the same DDF you provided but still no presence detection. 
[image]
[image][image]

BabaIsYou, 2022-12-20:
It seems that you have the FF issue one more time … "uniqueid":"70:ac:08:ff:fe:21:2b:6c-ff-0400" instead of "uniqueid":"70:ac:08:ff:fe:21:2b:6c-01-0400" ;-(

manosioa, 2022-12-21:
What does "ff issue" mean? I understand that the ff appears in the unique id, but how is that a problem?
If it is how can I fix it? 
What is the next step?

BabaIsYou, 2022-12-21:
I already had it for this device (and still have it with another one - a thermostat). You have to delete and repair the device. 

At this time nobody found where this bug come from. It’s here and there sometimes in different version ;-(

manosioa, 2022-12-21:
I did delete and repair both sensors but something strange happens.
Even after I delete them from the Phoscon environment and they are deleted in deCONZ environment as well, when I restart deCONZ add on, I can see both sensors in Phoscon environment but not in deCONZ environment.
[image]
[image]

Everytime I restart both sensors re appear in Phoscon. It's like it doesn't save the changes.
Tried also restarting the GW with no luck.

Smanar, 2022-12-21:
And I think the "magic sensor" that can't be deleted are with the endpoint "FF" ?

I have compared both DDF (but IDK why the last one is not working)
- RStateErrorCode is a string, not a value, so defaut value can't be 1 but "1"
- From [the code](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5868) you need to have at least one

```
          "read": {
            "fn": "tuya"
          },
```

and for all other

```
          "read": {
            "fn": "none"
          },
```

> 
The read function is special and must only be specified on one item (here state/on). It doesn't actually read a single value but triggers the Tuya device to report all datapoints.

But don't ask me the impact ^^

But they are minor changes, I don't see why the last DDF is not used ....

[157 more replies]

PR #7184 (2023-08-31): DDF add support for TS0601_TZE204_qasjif9e & fix other mmWave Radar Presence DDF
Add support for _TZE204_qasjif9e and add two custom items to avoid side effect of reusing `config/duration` for instance [#6998](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6998#issuecomment-1677574695)

Fix #6998  (breaking change)

Close #7183 
Close #6517
closed 2023-08-31 by manup