dresden-elektronik/deconz-rest-plugin #6523: Frient MOSZB-141

Idaho947, 2022-11-10:
Hello,

I just buy a new sensor module and it seems  to not be in deconz

## Device

- Product name: Frient motion sensor
- Manufacturer: Frient
- Model identifier: MOSZB-141
- Device type : Sensor

## Screenshots

  Required screenshots:
[image]
[image]

### Basic
[image]
[image]

### Identify
[image]

### Other
[image]

Smanar, 2022-11-10:
I have found the DDF for the model MOSZB-140, but I don't see how the DDF can work ...
Can you try this clean one ?

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "frient A/S",
  "modelid": "MOSZB-141",
  "vendor": "Frient",
  "product": "MOSZB-141 motion sensor pro",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x23",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x23",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername",
          "awake": true
        },
        {
          "name": "attr/modelid",
          "awake": true
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 35,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 300,
          "max": 43200,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 35,
      "cl": "0x0500"
    }
  ]
}
```

You will probably miss battery

Idaho947, 2022-11-11:
This one doesnt work. The sensor stay like that even if remove and reinclude it.
[image]

Smanar, 2022-11-11:
I have a doubt because this sensor have 4 endpoint, but from that I m reading it use only the 0x23.

Can you take a look in the API to get the device json (in phoscon/help/API Information/sensors/X)
The enrollement can have failed.

Else will be usefull to check logs if the device use the cluster 0x0406, but for the moment I m sure it's not the situation.

Idaho947, 2022-11-11:
I have this on the json :
`                    {
    "config": {
        "battery": 0,
        "on": true,
        "pending": [],
        "reachable": true
    },
    "ep": 35,
    "etag": "f7fe4468763dd67c2394bfae780e255e",
    "lastannounced": null,
    "lastseen": "2022-11-11T10:12Z",
    "manufacturername": "frient A/S",
    "modelid": "MOSZB-141",
    "name": "MOSZB-141",
    "state": {
        "lastupdated": "none",
        "presence": false
    },
    "type": "ZHAPresence",
    "uniqueid": "00:15:bc:00:1a:01:ac:3b-23-0500"
}`

Smanar, 2022-11-11:
Ok my bad, I have totaly missed the enrollement field.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "frient A/S",
  "modelid": "MOSZB-141",
  "vendor": "Frient",
  "product": "MOSZB-141 motion sensor pro",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x23",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x23",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 35,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 300,
          "max": 43200,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 35,
      "cl": "0x0500"
    }
  ]
}
```

Idaho947, 2022-11-11:
Same result. I try to hot reload the ddf but nothing more happen.

Smanar, 2022-11-11:
Can you try to re-include it to go faster.
To explan it's an IAS device used for alarm, thoses device need a special inclusion, the enrollement, I have totaly skipped this part on the first DDF.
On the second, the filed is present but hidden

```
        {
          "name": "config/enrolled",
          "public": false
        },
```

It's possible to make it visible, but I think a re-inclusion will make the job, thoses device are sleeper, so need to awake them to configure them, the "hot relaod" can take hour to success the configuration.
During the re-inclusion, the device is awake and deconz remake all the configuration.

Idaho947, 2022-11-11:
I'll try but nothing change.

Smanar, 2022-11-11:
SwoopX have just made a PR for it ^^

Idaho947, 2022-11-11:
Top. But still no battery after 4 hours

Smanar, 2022-11-11:
Can you try the SwoopX DDF ?

Idaho947, 2022-11-11:
Ok I try it.
When a motion is detected the motion pass 1 during one second then pass 0 direct. It's normal ?

Smanar, 2022-11-11:
I m asking him too https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6524#issuecomment-1311837076
Because his DDF generate more than 1 ZHAPresence, so idk how it work.

SwoopX, 2022-11-12:
> Ok I try it. Whith this ddf : -When a motion is detected the motion pass 1 during one second then pass 0 direct. It's normal ? -No batterie at this time. Wait few hours.

If you do not configure the sensor, that's what you get.
And you won't get any battery values unless either the DDF is part of the deconz package and therefore has access to the required js files or you copy the js file to the current DDF location.

Idaho947, 2022-11-12:
Goood ! Thanks for your answer. Can you give me the documentation for the sensor configuration ? I don't know what I have to search...

SwoopX, 2022-11-12:
https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/sensors/

> The ddf is in : /home/mika/.local/share/dresden-elektronik/deCONZ/devices it's not OK ?

Read again what I worte.

Idaho947, 2022-11-12:
Thanks. Ok I see it you talk about tu batterie js not th ddf... I'll want for the next stable. Thanks for your help.

closed 2022-12-11 by github-actions[bot]

Idaho947, 2023-04-04:
Hello, I'm in 2.21.2 and no batterie. It's normal ?

Smanar, 2023-04-04:
Hello, You haven't the field or it's not working ?

Idaho947, 2023-04-04:
It stay at null. I will try an exclusion / reinclusion in the week. I ve nearly the same issue with the consumption frient module (no field in the case).

Smanar, 2023-04-04:
I m not sure it's working for this device.
On Z2M they are using only the lowbattery field.

```
    {
        zigbeeModel: ['MOSZB-141'],
        model: 'MOSZB-141',
        vendor: 'Develco',
        description: 'Motion sensor',
        fromZigbee: [fz.ias_occupancy_alarm_1],
        toZigbee: [],
        exposes: [e.occupancy(), e.battery_low()],
    },
```

Idaho947, 2023-04-05:
After an exclusion\reinclusion it's OK.

Smanar, 2023-04-05:
??? So you have a battery level with this device ?

Idaho947, 2023-04-05:
Yes it's ok.