dresden-elektronik/deconz-rest-plugin #6525: Changing color temperature on IKEA Tradfri LED with IKEA STYRBAR switch

nachevn, 2022-11-12:
## Describe the bug

Changing the color temperature on IKEA Tradfri LED with IKEA STYRBAR switch seems not to work. In contrast, changing it over the Phoscon App is working.

I have multiple Zigbee Lights at home and I'm trying to control them over external Zigbee switches. I tried two different setups:
1) Two IKEA Tradfri LED bulbs, "modelid": "TRADFRIbulbE14WScandleopal470lm" + IKEA STYRBAR switch
2) Two LIDL Livarno Lux bulbs,  "modelid": "HG06106B", + IKEA STYRBAR switch

Changing the color temperature on the Livarno lights over the styrbar switch is working. Changing the color on the tradfri bulbs unfortunately not.

STYRBAR Switch:
```                    
{
    "config": {
        "alert": "none",
        "battery": 100,
        "group": "8",
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "71564ee28502bfe310ede0681d433d0e",
    "lastannounced": null,
    "lastseen": "2022-11-12T10:53Z",
    "manufacturername": "IKEA of Sweden",
    "mode": 1,
    "modelid": "Remote Control N2",
    "name": "STYRBAR Switch",
    "state": {
        "buttonevent": 3003,
        "lastupdated": "2022-11-12T10:34:56.444"
    },
    "swversion": "1.0.024",
    "type": "ZHASwitch",
    "uniqueid": "00:3c:84:ff:fe:e0:02:da-01-1000"
}
```

TRADFRIbulbE14WScandleopal470lm:
```
{
    "colorcapabilities": 0,
    "ctmax": 65279,
    "ctmin": 0,
    "etag": "420f4c3262a653aa91659dbb1cbd8736",
    "hascolor": true,
    "lastannounced": "2022-11-12T08:54:28Z",
    "lastseen": "2022-11-12T10:54Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRIbulbE14WScandleopal470lm",
    "name": "IKEA TEST",
    "state": {
        "alert": "none",
        "bri": 254,
        "colormode": "ct",
        "ct": 385,
        "on": true,
        "reachable": true
    },
    "swversion": "1.0.032",
    "type": "Color temperature light",
    "uniqueid": "04:0d:84:ff:fe:98:67:48-01"
}
```

LIDL Livarno Lux:
```
{
    "ctmax": 65279,
    "ctmin": 0,
[... 91 lines cut ...]
12:00:16:744 rule event /sensors/3/state/lastupdated: 0 -> 0
12:00:16:744 trigger rule 1 - Rule PUSHDIMUP_ON
12:00:17:254 [INFO] - Button 3001 - Remote Control N2, broadcast to: 0x0004, endpoint: 0x01, cluster: SCENES (0x0005), action: Move ct colder, payload: 010D00, zclSeq: 99
12:00:17:254 rule event /sensors/3/state/lastupdated: 0 -> 0
12:00:17:254 trigger rule 7 - Rule DIMCTDOWN_START
12:00:22:478 ZCL attribute report 0x842E14FFFE37DF5B for cluster: 0x0300, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
12:00:22:488 ZCL attribute report 0x680AE2FFFEA33756 for cluster: 0x0300, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
12:00:23:062 [INFO] - Button 3003 - Remote Control N2, broadcast to: 0x0004, endpoint: 0x01, cluster: SCENES (0x0005), action: Stop ct colder, payload: D716, zclSeq: 100
12:00:23:063 rule event /sensors/3/state/lastupdated: 0 -> 0
12:00:23:063 trigger rule 8 - Rule DIMCTDOWN_STOP
12:00:27:434 [INFO] - Button 4001 - Remote Control N2, unicast to: 0x0000, endpoint: 0x01, cluster: SCENES (0x0005), action: Move ct warmer, payload: 000D00, zclSeq: 104
12:00:27:434 rule event /sensors/3/state/lastupdated: 0 -> 0
12:00:27:434 trigger rule 9 - Rule DIMCTUP_START
12:00:32:637 ZCL attribute report 0x842E14FFFE37DF5B for cluster: 0x0300, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
12:00:32:653 ZCL attribute report 0x680AE2FFFEA33756 for cluster: 0x0300, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
12:00:33:334 [INFO] - Button 4003 - Remote Control N2, broadcast to: 0x0004, endpoint: 0x01, cluster: SCENES (0x0005), action: Stop ct warmer, payload: 6984, zclSeq: 105
12:00:33:335 rule event /sensors/3/state/lastupdated: 0 -> 0
12:00:33:335 trigger rule 10 - Rule DIMCTUP_STOP
12:00:35:596 [INFO] - Button 2002 - Remote Control N2, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: On, payload: None, zclSeq: 106
12:00:35:597 rule event /sensors/3/state/lastupdated: 0 -> 0
12:00:35:597 trigger rule 5 - Rule PUSHDIMDOWN_OFF
```
## Additional context

I would be glad to provide more information if required.

ebaauw, 2022-11-12:
Are the IKEA lights members of group 8?

The Styrbar sends direct commands to that group.  These are IKEA-specific, so other lights will ignore them, but IKEA lights would react to them.  With the IKEA firmware bug that it ignores new commands while a previous command is still transitioning, the commands sent from the rules would be ignored.

Last I checked, the CWS light that was bundled with the Styrbar still reacted to the direct commands, even when paired with deCONZ.  Older IKEA light models seem to lose their capability to react to these commands, once deCONZ configures them.

nachevn, 2022-11-12:
@ebaauw : Thank you for answering so fast. As far as I can see there is no such group with id 8. They should be, together with the switch, as far as I understand that in the group with an id 1. But how then is the on/off as well as changing the brightness working, when they are in different groups?

That is the group with ID=1. As mentioned already, there is no such group with id 8:
```
{
    "action": {
        "alert": "none",
        "bri": 242,
        "colormode": "ct",
        "ct": 0,
        "effect": "none",
        "hue": 0,
        "on": false,
        "sat": 127,
        "scene": null,
        "xy": [
            0,
            0
        ]
    },
    "devicemembership": [],
    "etag": "0b3ce132dd2847965b59a319e4865612",
    "hidden": false,
    "id": "1",
    "lights": [
        "10",
        "11",
        "12"
    ],
    "lightsequence": [],
    "multideviceids": [],
    "name": "The name of the group",
    "scenes": [],
    "state": {
        "all_on": false,
        "any_on": false
    },
    "type": "LightGroup"
}
```

Regarding the IKEA lights, I bought them yesterday, I hope that they didn't lose anything by just connecting them to deCONZ.

ebaauw, 2022-11-13:
You’re using rules to control the lights.  The Styrbar’s commands are picked up by deCONZ and reported as button events.  The button event that triggers a rule, which sends commands to the lights and/or groups (a different group than the Styrbar sends its commands to).
I’m not sure what’s not working.  In your first log, the Trådfri lights report changes for _Color Control_ after pressing the colour temperature buttons.

nachevn, 2022-11-13:
@ebaauw : I think I didn't make it clear, if you have a look at the logs, everything looks fine but in real nothing happen. The color temperature of the lights does not change by pressing the button on the switch.

ebaauw, 2022-11-13:
Could you list rules 37 thru 40?

nachevn, 2022-11-13:
All of the rules with address /groups/1:
```
    "25": {
        "actions": [
            {
                "address": "/groups/1/action",
                "body": {
                    "on": true
                },
                "method": "PUT"
            }
        ],
        "conditions": [
            {
                "address": "/sensors/00:3c:84:ff:fe:e0:02:da-01-1000/state/buttonevent",
                "operator": "eq",
                "value": "1002"
            },
            {
                "address": "/sensors/00:3c:84:ff:fe:e0:02:da-01-1000/state/lastupdated",
                "operator": "dx"
            }
        ],
        "created": "2022-11-12T08:29:58",
        "etag": "8baef7673f0b2d7ec3b1ab99ed403404",
        "lasttriggered": "2022-11-13T08:11:03",
        "name": "Rule PUSHDIMUP_ON",
        "owner": "4B0B8CAEB0",
        "periodic": 0,
        "status": "enabled",
        "timestriggered": 21
    },
    "26": {
        "actions": [
            {
                "address": "/groups/1/action",
                "body": {
                    "bri_inc": 0
                },
                "method": "PUT"
            }
        ],
        "conditions": [
            {
                "address": "/sensors/00:3c:84:ff:fe:e0:02:da-01-1000/state/buttonevent",
                "operator": "eq",
                "value": "1003"
            },
            {
                "address": "/sensors/00:3c:84:ff:fe:e0:02:da-01-1000/state/lastupdated",
                "operator": "dx"
            }
        ],
        "created": "2022-11-12T08:29:58",
        "etag": "27d71e613a849aca459f71ba68a24abc",
        "lasttriggered": "2022-11-12T15:09:30",
        "name": "Rule PUSHDIMUP_STOP",
        "owner": "4B0B8CAEB0",
        "periodic": 0,
        "status": "enabled",
        "timestriggered": 12
    },
    "27": {
        "actions": [
            {
                "address": "/groups/1/action",
                "body": {
                    "bri_inc": 254,
                    "transitiontime": 30
                },
[... 212 lines cut ...]
                },
                "method": "PUT"
            }
        ],
        "conditions": [
            {
                "address": "/sensors/00:3c:84:ff:fe:e0:02:da-01-1000/state/buttonevent",
                "operator": "eq",
                "value": "4003"
            },
            {
                "address": "/sensors/00:3c:84:ff:fe:e0:02:da-01-1000/state/lastupdated",
                "operator": "dx"
            }
        ],
        "created": "2022-11-12T08:33:38",
        "etag": "339464a191d496f9de2be1b4aabbb72c",
        "lasttriggered": "2022-11-12T11:49:20",
        "name": "Rule DIMCTUP_STOP",
        "owner": "4B0B8CAEB0",
        "periodic": 0,
        "status": "enabled",
        "timestriggered": 2
    }
```

ebaauw, 2022-11-13:
Ok, so you set a long `transitiontime` from the rules.  WIth the IKEA firmware bug, that means the Trådfri lights become unresponsive for 3 to 6 seconds, after pressing a button.  You might also want to double-check in the GUI that the Trådfri lights support _Step Color Temperature_ (which is sent by `ct_inc`) in addition to _Move to Color Temperature_ (which is sent by setting `ct`).  Also, Trådfri lights have a colour temperature range from 250 to 454, to the maximum `ct_inc` values would be +/- 204.  I don't think the API plugin would mind, though, but you might want to check in the GUI that the Trådfri accepts a _Step Size_ of 500 or -500 with _Step Color Temperature_.

nachevn, 2022-11-15:
Sorry for my late answer. To be honest I don't know where to check that. It took me some time to make the VNC connection working but here is it:
[image]
At the end, as mentioned in the description of the issue, changing the color temperature over the deCONZ App is working. Changing it over the styrbar switch isn't working.

I would appreciate if you could give an advise where to look and how to check what you described in your last comment.

Thanks again for helping.

ebaauw, 2022-11-15:
Enable the _Cluster Info_ panel from the _Panels_ menu, select the. _Color Control_ cluster on the Trådfri light.  See the user manual under _Help_ for details.

nachevn, 2022-11-15:
Just tested it:
[image]
I can set a value of 500 or 0, that changes the color temperature of the light. Also tried with -500 but nothing happened.

ebaauw, 2022-11-15:
That’s _Move to Color Temperature_, setting an absolute colour temperature value, corresponding to `state/ct` in the API.  That cannot be negative, of course.

You want to check _Step Color Temperature_ further down, setting a relative colour temperature value, corresponding to `state/ct_inc`.  That is what you use in the rules triggered by the Styrbar.

nachevn, 2022-11-15:
Ok, that was new for me. Changing the color temperature down (from cold to warm) is working either over the deCONZ GUI:
[image]
as well as over the styrbar switch.

Changing it up (warm --> cold) isn't working. Also not over the deCONZ GUI.

The steps from my side were:
Working scenario:
1) Move to color temperature 0 over the deCONZ GUI
2) Using Step Color Temperature, Step Mode down, Step size 40, after 5 exec the max value was reached, which means executing further steps didn't change anything.
Same scenario but with the Styrbar Switch works as well.

As mentioned already the other way around is not working. Not over the deCONZ GUI nor over the switch.

ebaauw, 2022-11-15:
Huh?  Step mode _Down_ should decrease the _Color Temperature_ mired value (turning the light cooler, or increasing the temperature in K); _Up_ should increase the mired (turning it warmer, or decreasing the temerature in K).  The _Step Value_ needs to be positive in both cases.

I'm not too sure what the min/max values are supposed to do here, you might want to set max to 500 (or 454) just to be sure.

Next, try setting a _Transition Time_ of 100 (10s) and see if the light reacts to another command (e.g. _Stop_) while transitioning.  If it has the firmware bug, it won't.

nachevn, 2022-11-15:
ok, thats seem to be the case, with a _Transition Time_ of 100, nothing works. As you mentioned, even the stop isn't working.

As far as I got that from you, there is a firmware bug and there seems to be also no firmware update fixing it? Am I right so far? Is it just because of the transition time and can that be set to 0? What I didn't get is, why is the one way working over the deCONZ GUI working, I mean the _Step Color Temperature_ down and the other way not?

ebaauw, 2022-11-15:
So the (other) issue with the Trådfri firmware bug is that the light doesn't accept commands, while a transition for a previous command is still underway.  With the default `transitiontime` of 0.4s (raw value 4), this typically only happens when you combine multiple rest attributes in the same API call (e.g. PUTting a state of `{"bri": 254, "ct", 360}').  However, when you specify longer transition times, next API calls might be ignored as well.  So with your rules, you'll see that the first button press works, but then the lights is unresponsive for 6 seconds.  So if you change brightness and immediately after change colour temperature, the latter is ignored.  If you wait a little, and try again, it should work.

Adding `"transitiontime": 0` to the action body will workaround the bug, but results in very ugly (abrupt) transitions.

The firmware bug has been present since forever on older Trådfri lights, over multiple firmware versions.  I'm afraid, it doesn't look like IKEA will be fixing it.  The newer color white spectrum light that came bundled with my Styrbar is the first (and only I got my hands on) Trådfri light without this bug.

nachevn, 2022-11-16:
@ebaauw changed and tested. Now Up and Down is working as expected over the deCONZ GUI. Still changing it over the switch, at least the down button it isn't. Even after changing the transition time:
```
$curl https://some.domain/api/XXXXXXXXXX/rules/39 | jq 
{
  "actions": [
    {
      "address": "/groups/1/action",
      "body": {
        "ct_inc": -40,
        "transitiontime": 10
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/sensors/00:3c:84:ff:fe:e0:02:da-01-1000/state/buttonevent",
      "operator": "eq",
      "value": "4001"
    },
    {
      "address": "/sensors/00:3c:84:ff:fe:e0:02:da-01-1000/state/lastupdated",
      "operator": "dx"
    }
  ],
  "created": "2022-11-12T08:33:38",
  "etag": "ebc92ad80c0daaae4223465f01fa80ab",
  "lasttriggered": "2022-11-16T10:34:02",
  "name": "Rule DIMCTUP_START",
  "owner": "4B0B8CAEB0",
  "periodic": 0,
  "status": "enabled",
  "timestriggered": 2
}

$curl https://some.domain/api/XXXXXX/rules/37 | jq

{
  "actions": [
    {
      "address": "/groups/1/action",
      "body": {
        "ct_inc": 40,
        "transitiontime": 10
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/sensors/00:3c:84:ff:fe:e0:02:da-01-1000/state/buttonevent",
      "operator": "eq",
      "value": "3001"
    },
    {
      "address": "/sensors/00:3c:84:ff:fe:e0:02:da-01-1000/state/lastupdated",
      "operator": "dx"
    }
  ],
  "created": "2022-11-12T08:33:29",
  "etag": "0e2ce59476b0ea63cdd4a93ef6230631",
  "lasttriggered": "2022-11-16T10:35:36",
  "name": "Rule DIMCTDOWN_START",
  "owner": "4B0B8CAEB0",
  "periodic": 0,
  "status": "enabled",
  "timestriggered": 7
}
```

ebaauw, 2022-11-16:
You’re using the absolute values with `ct_inc` instead of relative values.  After two presses the lights will be at max `ct` value and not change anymore.

are you sure you want to set rules on the x001 event?  This is the hold event on a long press, not a single press.  It wouldn’t trigger the rule on a normal press/release.

closed 2022-12-15 by github-actions[bot]