dresden-elektronik/deconz-rest-plugin #6528: HUE Smart Switch ROM001 does not work

IgnacioHR, 2022-11-12:
## Describe the bug

HUE Smart Switch ROM001 does not work

## Steps to reproduce the behavior

Having deCONZ installed with versions 2.18.0 or higher, open DEBUG and select INFO_2. Press the HUE Switch button, and notice NO Websocket output is generated.

Repeat the process with 2.17.1 installed and you'll see this in the log

```
13:30:18:776 Websocket 127.0.0.1:40804 send message: {"e":"changed","id":"21","r":"sensors","state":{"buttonevent":1002,"eventduration":1,"lastupdated":"2022-11-12T12:30:18.768"},"t":"event","uniqueid":"00:17:88:01:08:93:7f:3b-01-fc00"} (ret = 16)
13:30:18:777 Websocket 192.168.1.29:60858 send message: {"e":"changed","id":"21","r":"sensors","state":{"buttonevent":1002,"eventduration":1,"lastupdated":"2022-11-12T12:30:18.768"},"t":"event","uniqueid":"00:17:88:01:08:93:7f:3b-01-fc00"} (ret = 16)
```

## Expected behavior

Websocket events should be raised when using the HUE Smart Switch button ROM001

## Screenshots

No screenshots needed, the bind process works and the deCONZ UI related with the switch works well. The issue is with the WebSocket and thus with the integration with Home Assistant or any other software using it.

## Environment

In my case it is Raspbian + Raspbee II + Home Assistant in a separate Raspberry PI

## deCONZ Logs

Provided above

## Additional context

I tested with versions: 2.19.2, 2.19.1 and 2.18.2 and in all of them my HUE Smart Switch ROM001 from Philips doesn't work. The last version they word was 2.17.1

I notice this once I upgraded from 2.17.1 to 2.19.1. I reported here:

https://forum.phoscon.de/t/hue-smart-switch-rom001-no-longer-work-in-version-2-19-01/2706

But someone pointed me to:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6461

But after the merge applied to 6461 my problem persist.

In fact, there is this change between 2.17.1 and 2.18.0 that is more likely the source of the problem

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6164

ebaauw, 2022-11-12:
See #6461: I cannot reproduce this issue.  Please execute the checks in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6461#issuecomment-1300977974 and list the output here.

Mimiix, 2022-11-12:
@ebaauw is this already being in Progress in the issue you mentioned or is it worth keeping this one open?

@IgnacioHR can you please fill the template as asked? Environment is missing details.

IgnacioHR, 2022-11-12:
Hi @ebaauw 

Test results:

1. Installed 2.19.02
[image]

2. Button actions received in the GUI:
[image]

Works

3. Basic Cluster

[image]

The same

4. DDF

[image]

5. ¿?¿?¿? !!! HUE Switches started working !!! I rebooted the RPI and reinstalled 2.19.2 so maybe for some reason the update I did yesterday using `sudo apt update ; sudo apt upgrade` failed to to date to 2.19.2? Today I installed 2.19.02 downloading the .deb and using `dpkg -i deconz [file]`

So, I'm going to close this and sorry for the confusion.

IgnacioHR, 2022-11-13:
Well, maybe I was too fast closing this issue.

I've 3 HUE Smart Switches at home. 2 are working as expected but 1 does not work for sending WebSocket events.

When I start deCONF, the Node Info page of the one that doesn't work shows like this:
[image]

The I force read the Basic cluster and it shows like this:

[image]

And going back to see the Node Info the values are changed
[image]

The information is not stored in persistent memory because the next time I restart deCONZ-gui the old information is back.

De device does not emit socket events even after the update.

IgnacioHR, 2022-11-13:
Additional information:

a REST GET to obtain device information:

Just after starting deCONZ gui, the returned JSON is:

```
{
    "config": {
        "on": true,
        "reachable": true
    },
    "etag": "62522a9d8d7b470716de0eb168e01c9c",
    "lastannounced": null,
    "lastseen": "2022-11-12T07:46Z",
    "manufacturername": "Philips",
    "mode": 1,
    "modelid": "ROM001",
    "name": "Salon on/off pie",
    "state": {
        "buttonevent": 1002,
        "lastupdated": "2022-11-07T20:13:36.742"
    },
    "type": "ZHASwitch",
    "uniqueid": "00:17:88:01:08:02:a4:5a-ff-fc00"
}
```

After force reading the Basic cluster it changes to:

```
{
    "config": {
        "battery": 100,
        "on": true,
        "reachable": true
    },
    "etag": "b54158aec36e38990f3cbcd984c0be72",
    "lastannounced": null,
    "lastseen": "2022-11-13T12:36Z",
    "manufacturername": "Signify Netherlands B.V.",
    "mode": 1,
    "modelid": "ROM001",
    "name": "Salon on/off pie",
    "state": {
        "buttonevent": 1002,
        "lastupdated": "2022-11-07T20:13:36.742"
    },
    "swversion": "2.47.8_h2F968624",
    "type": "ZHASwitch",
    "uniqueid": "00:17:88:01:08:02:a4:5a-ff-fc00"
}
```

ebaauw, 2022-11-13:
So force-read the _Basic_ cluster, wait a couple of minutes for deCONZ to write the newly read manufacturer name to the database, restart deCONZ, and it should work.

IgnacioHR, 2022-11-13:
I wait the hole night already. I discover the issue yesterday afternoon and I repeat the process this morning to document it. 

`lsof -p ` reveals there is a file open:

`deCONZ  16656   pi   16u      REG      179,2   356352  125804 /home/pi/.local/share/dresden-elektronik/deCONZ/zll.db`

that most likely is the database.

That file modified time is just now!

[image]

Now I restart deCONZ, and this is what I see for that node

[image]

ebaauw, 2022-11-13:
Yes, that's the database alright.  I'm sorry, I don't know why the manufacturer name would revert to "Philips" on restart.  Personally, I would check and patch the database (make a backup first!).  Maybe it's safer to delete, reset, and repair the button.

IgnacioHR, 2022-11-13:
Thank you

I finally deleted the node, removed the battery on the switch, put the battery again and re-paired with deCONZ, now it works

closed 2022-11-13 by IgnacioHR

Mimiix, 2022-11-13:
If you have any issues with this in the future, please open an topic on the forums as user requests are handled there.