dresden-elektronik/deconz-rest-plugin #6529: Conbee I, all deconz > 2.18.01 (2.18.02 - 2.19.02-beta) make system become unresponsive,  ON/OFF ends in "timeout"

dxxxm, 2022-11-13:
## Describe the bug

Regardless of whether trying to turn ON/OFF a light via hardware or deconz, the command ends with a timeout, i.e. eventually unresponsive.

## Steps to reproduce the behavior

All docker images tested from v2.18.02 onwards show the same behaviour. (inlc 2.19.02-beta)

## Expected behavior

Responsiveness as e.g. v2.18.01

## Screenshots

## Environment

 - Host system: Raspberry Pi 3 Model B Plus Rev 1.3 
 - Running method: deconz-community / deconz-docker
 - Firmware version: 0x26400500
 - deCONZ version: 2.18.02 - 2.19.02-beta
 - Device: ConBee I
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? Yes.  msata2usb (AS2115) Port 2: Dev 6, If 0, Class=Mass Storage, Driver=usb-storage, 480M

## deCONZ Logs

pastebin [debug log](https://pastebin.com/zVxLmaQc)

## Additional context
Running fine up to v2.18.01. 
Downgraded to it after another fail with the latest release v2.19.02-beta.

Mimiix, 2022-11-13:
Hi,

As this is not a bug report as described in #5113 i am closing this issue.

Additionally, there's already a issue about this :  https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6445

Please update your conbee firmware. If that doesn't helpt , post in that issue so we can reopen that one.

Kind regards,

closed 2022-11-13 by Mimiix

Ltek, 2022-11-18:
@Mimiix you close both cases reporting the same issue (it seems) -- with a circular reference? You listed THIS issue when closing --> https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6445

This issue has been reported by several people, including @Smanar  in that other thread. Is it being addressed in a different issue posted somewhere? This is a showstopper bug across platforms... Docker, Windows 10 (what I run), etc

Mimiix, 2022-11-18:
> @Mimiix you close both cases reporting the same issue (it seems) -- with a circular reference? You listed THIS issue when closing --> #6445
>
> This issue has been reported by several people, including [...]

The issue is closed as it was merged with a pull request. If it's jot fixed, we should open it again.