dresden-elektronik/deconz-rest-plugin #6530: Energy meter - Hiking -Tomzn DDS238-2 Zigbee _TZE200_bkkmqmyo TS0601

aIA78, 2022-11-13:
## Device
Please, add support to this device:

- Product name: Hiking - Tomzn - DDs238-2 Zigbee
- Manufacturer: _TZE200_bkkmqmyo
- Model identifier: TS0601
- Device type : 
    - Other: Energy meter, with internal switch
    - https://zigbee.blakadder.com/Hiking_DDS238-2.html
   [image]

## Screenshots

### Basic

[image]

[image]

[image]

### Identify

### Alarms

### Device Temperature

### Groups

[image]

[image]

[image]

### Scenes

[image]

[image]

[image]

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

[image]

[image]

[image]

Smanar, 2022-11-14:
Hello can you try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_lsanae15", "_TZE200_bkkmqmyo", "_TZE200_eaac7dkw"],
  "modelid": ["TS0601", "TS0601", "TS0601"],
  "product": "Single Phase Din Rail Smart Energy Meter",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
      {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 16, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 16, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "parse": {"fn": "tuya", "dpid": 17, "eval": "Item.val = Attr.val * 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/current",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "Item.val = Attr.val / 1000;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/voltage",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "Item.val = Attr.val / 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/power",
          "parse": {"fn": "tuya", "dpid": 103, "eval": "Item.val = Attr.val / 10;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}
```

I have some problems for current and tension, need to see how they look in logs (but will see later)

aIA78, 2022-11-17:
Hi, thank you very much, i’m going to try at home later…  I will report soon

Smanar, 2022-11-23:
Yep, for HA > https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839
The persistent file is hidden `/data/.local/share/dresden-elektronik/deCONZ/devices/`

aIA78, 2022-11-30:
Thank you for your comment. After few days testing I finally can access to HomeAssistan host through SSH. I have created a json file with the content that you suggested and done “hot reload” in VNC deconz interface, but nothing happens… I can’t reset host until tonight, it’s running some automations… I’m trying to reset tonight and see. Thank you again…. I will report soon.

Smanar, 2022-11-30:
>and done “hot reload” in VNC deconz interface

So it mean when you have done "edit DDF" the good file was selected ? Or you have used the "load" feature ?
To see if it work the node name go from 0xXXXX to a real name (visible on the GUI) and as it s a wired device, no problem to use the "hot relaod" feature.

aIA78, 2022-11-30:
Well... I haven't used "load", only pushed on "hot reload".

This morning mains from the company went out so... full hard reset!!

After restart homeassistant host now i can see that DDF file was loaded.

I paste some captueres from phoscon, deconz and home assistant.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

Well, the switch works perfectly and the power and compsumtion are 0 because theres no have load conected yet... tomorrow I'm going to connect something to the output.

How can I get the logs you need?

Greetins and thank you for you help again!

Smanar, 2022-12-01:
>How can I get the logs you need?

Mimiix, 2023-01-27:
Sure.

Smanar, 2023-01-27:
In this part

```
        {
          "name": "state/consumption",
          "parse": {"fn": "tuya", "dpid": 17, "eval": "Item.val = Attr.val * 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
```

Use 102 as dpid.

aIA78, 2023-01-27:
Hello, I have tried your recommendation but it has not worked.
Then I found this github post (https://github.com/Koenkk/zigbee2mqtt/issues/13688) and I decided to try the first value (dpid = 1) and it worked!!! HA shows in consumption the same value as the lcd of the device. It is not necessary to apply any multiplication or division factor.
Could you help me to get to show more parameters (current, voltage, frequency, etc)?

Smanar, 2023-01-28:
Ha yes right 102 is for "reverse energy".

It seem we need to use the dpid 6, but value need to be decoded for exemple like here https://github.com/jasonacox/tinytuya/issues/55#issuecomment-864506722

Can you enable log "aps"+"aps_l2" and give me the logs when you receive data, I need raw value (will be something like "asnu 123456677887")
To locate them you can enable in same time "DDF", you will something about state/current as it's the field that use this dpid ATM

aIA78, 2023-02-02:
Hi, I'm still digging into the Deconz logs and the Home Assistant add-on logs.  I have found packages with information for almost all the dpids listed in the GitHub link I posted.
In the logs the dpids are in hex while in the list and in the DDF file they are in decimal (for example the dpid for power is 103 in decimal and in the logs it looks like 0x67 in hexadecimal).
I just don't see packages for dpid 6 anywhere. I'll keep looking to see if it is retransmitted with another dpid number.
The question at the moment is... how can I add in the DDF file so that HomeAssistantbme shows another entity with for example the frequency (dpid:105) or the power factor (dpid:111)? 
Thanks in advance!

Smanar, 2023-02-02:
>I just don't see packages for dpid 6 anywhere. I'll keep looking to see if it is retransmitted with another dpid number.

Ha ? I can be wrong, have travalled a lot in zigbee project, but have started with your manufacture name "_TZE200_bkkmqmyo". But there is so much tuya clone ...

https://github.com/zigpy/zha-device-handlers/pull/1878

To use DDF on home assistant, you need to create the DDF in the hidden folder `/data/.local/share/dresden-elektronik/deCONZ/devices/`

Long story with all the procedure https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

If you have the GUI, can update the DDF with the editor (hot reload) else need to restart deconz.

aIA78, 2023-02-03:
As you told me before, I have edited the DDF file by repeating this block and editing it as follows:

        {
          "name": "state/power_factor",
          "parse": {"fn": "tuya", "dpid": 111, "eval": "Item.val = Attr.val /1000;" },
          "read": {"fn": "none"},
          "default": 0
        },

However, it seems that deconz does not recognize what I have written (could it be because the power_factor state does not exist in deconz's available states? If so, how can I make that entity available to HA?)

Smanar, 2023-02-03:
>However, it seems that deconz does not recognize what I have written (could it be because the power_factor state does not exist in deconz's available states?

Exactly.

It's easy, like I do here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6673/files
But need to be merged in c++ code, and can take long time.

aIA78, 2023-02-06:
Hello, I continue with my investigations.
I have found this whole piece of text in the Deconz log. It has a different structure than the messages that contain the DPID, but the source addresses are those corresponding to the device I am studying (srcAddr: 0x0e9f and addr: 0x943469fffed98d7d). Do you think that some value can be obtained from here that corresponds to Voltage and Current, or to a combination of both as you indicated before?

`21:44:56:695 APS-DATA.indication srcAddr: 0x0e9f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -69`
`21:44:56:696 	asdu: 095b0200401100001e00000000000000000000006400010e0000aa00000000000a000000000000`
`21:44:56:701 APS-DATA.request id: 177, addrmode: 0x02, addr: 0x0e9f, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x04`
`21:44:56:702 	asdu (length: 5): 105b0b0200`
`21:44:56:744 APS-DATA.confirm id: 177, status: 0x00 SUCCESS`
`21:44:56:745 APS-DATA.confirm request id: 177 -> erase from queue`
`21:44:56:773 aps request id: 177 finished, erase from queue`
`21:44:56:778 APS-DATA.indication srcAddr: 0x0e9f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -70`
`21:44:56:780 	asdu: 095c02004112000011000500640005001e003c00000000000000`
`21:44:56:784 Websocket 172.30.32.1:55824 send message: {"attr":{"id":"4","lastannounced":null,"lastseen":"2023-02-06T20:44Z","manufacturername":"_TZE200_bkkmqmyo","modelid":"TS0601","name":"base monitor casa","swversion":null,"type":"On/Off output","uniqueid":"94:34:69:ff:fe:d9:8d:7d-01"},"e":"changed","id":"4","r":"lights","t":"event","uniqueid":"94:34:69:ff:fe:d9:8d:7d-01"} (ret = 323)`
`21:44:56:788 Websocket 172.30.32.1:55824 send message: {"attr":{"id":"72","lastannounced":null,"lastseen":"2023-02-06T20:44Z","manufacturername":"_TZE200_bkkmqmyo","modelid":"TS0601","name":"Power 72","swversion":null,"type":"ZHAPower","uniqueid":"94:34:69:ff:fe:d9:8d:7d-01-0b04"},"e":"changed","id":"72","r":"sensors","t":"event","uniqueid":"94:34:69:ff:fe:d9:8d:7d-01-0b04"} (ret = 322)`
`21:44:56:792 Websocket 172.30.32.1:55824 send message: {"attr":{"id":"71","lastannounced":null,"lastseen":"2023-02-06T20:44Z","manufacturername":"_TZE200_bkkmqmyo","modelid":"TS0601","name":"Consumption 71","swversion":null,"type":"ZHAConsumption","uniqueid":"94:34:69:ff:fe:d9:8d:7d-01-0702"},"e":"changed","id":"71","r":"sensors","t":"event","uniqueid":"94:34:69:ff:fe:d9:8d:7d-01-0702"} (ret = 334)`
`21:44:56:793 APS-DATA.request id: 179, addrmode: 0x02, addr: 0x0e9f, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x04`
`21:44:56:795 	asdu (length: 5): 105c0b0200`
`21:44:56:866 APS-DATA.confirm id: 179, status: 0x00 SUCCESS`
`21:44:56:867 APS-DATA.confirm request id: 179 -> erase from queue`
`21:44:56:934 APS-DATA.request id: 180, addrmode: 0x03, addr: 0x943469fffed98d7d, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00`
`21:44:56:935 	asdu (length: 2): 330f`
`21:44:56:936 aps request id: 179 finished, erase from queue`
`21:44:56:976 APS-DATA.confirm id: 180, status: 0x00 SUCCESS`
`21:44:56:977 APS-DATA.confirm request id: 180 -> confirmed, timeout 344890062`
`21:44:57:015 APS-DATA.indication srcAddr: 0x0e9f, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -70`
`21:44:57:016 	asdu: 3300190f030b7b07ffff2e2100b3975dfeff14430c7fa125020fb50b7b07ffff2e2100841e40feff14430cb4a325020f520b7b07ffff2e2100bf955dfeff14430c6eb125020f96`
`21:44:57:017 APS-DATA.indication request id: 180 -> finished`
`21:44:57:017 APS-DATA.request id: 180 erase from queue`

I can't find any message that contains DPID 6 or any other new one regarding the DPIP list that I put previously, except for 10 which I think is equivalent to 16 (relay status). There are some that do not appear in my logs (6, 16, 17, 18 and 108).

1: total forward energy
6: Phase A (probably voltage, but it is encoded)
9: fault
12: clear energy
16: state (reading of the state of the on/off relay)
17: alarm set 1
18: alarm set 2
101: total active energy (I think they mean the balance of forward and reverse energy)
102: reverse energy
103: power
105: frequency
108: breaker (maybe you can activate the relay with this dp?)
109: reactive energy (KVArh)
110: reactive power (kVAr)
111: power factor

(Copy and pasted from previus linked post)

In my logs, the messages in which a DPID is clearly identified have the following structure:

`21:44:58:626 APS-DATA.indication srcAddr: 0x0e9f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -67`
`21:44:58:627 	asdu: 095d020042690200040000138b`
`21:44:58:632 TY_DATA_REPORT: seq 66, dpid: 0x69, type: 0x02, length: 4, val: 5003`
`21:44:58:633 TY_DATA_REPORT: seq 66, dpid: 0x69, type: 0x02, length: 4, val: 5003`
`21:44:58:634 TY_DATA_REPORT: seq 66, dpid: 0x69, type: 0x02, length: 4, val: 5003`
`21:44:58:635 TY_DATA_REPORT: seq 66, dpid: 0x69, type: 0x02, length: 4, val: 5003`
`21:44:58:637 TY_DATA_REPORT: seq 66, dpid: 0x69, type: 0x02, length: 4, val: 5003`
`21:44:58:638 APS-DATA.request id: 191, addrmode: 0x02, addr: 0x0e9f, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x04`
`21:44:58:638 	asdu (length: 5): 105d0b0200`

Regarding the modification of the code to include 'available states' I think I could do it with a little help, but what I don't know is how to recompile the code and replace the files created in the Deconz addon container for HomeAssistant.

To edit the files I think I have to clone the deconz-rest-api repository onto a raspberry pi running bullseyes os and edit the files to include the states I need.
If you give me some guidance on how to start I try to do it myself.

Greetings and thank you very much.

Smanar, 2023-02-09:
Ha ?
you don't have line with 

```
        DBG_Printf(DBG_INFO, "Tuya debug 4 : Address 0x%016llX Payload %s\n", ind.srcAddress().ext(), qPrintable(zclFrame.payload().toHex()));
        DBG_Printf(DBG_INFO, "Tuya debug 5 : Status: %u Transid: %u Dp: %u (0x%02X,0x%02X) Fn: %u Data %d\n", status, transid, dp, dp_type, dp_identifier, fn, data);
```

It's probalby because you can't use in same time the legacy code and the DDF one ....

```
21:44:56:695 APS-DATA.indication srcAddr: 0x0e9f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -69
21:44:56:696 	asdu: 095b0200401100001e00000000000000000000006400010e0000aa00000000000a000000000000
```

I don't see why this one is not decrypted like other ?
`095b0200 40 1100 00 1e 00000000000000000000006400010e0000aa00000000000a000000000000`
It's dpid 0x11 type 0x00 len = 30 with data `00000000000000000000006400010e0000aa00000000000a000000000000`

And yes can be the "0x06 for your device", I will make take a look at this string this WE 

Compile the code on the HA docker ? you like suffer ^^ ?
It's thericaly possible https://github.com/franciscogouveia/docker-deconz-rest-plugin-test
But I realy can't help you for that. The compilation is device dependant, so a compilation an another hardware can be not compatible.
If you have a raspberry can use it with a cloned deconz installation (out of docker), and can configure HA to use this installation too.

Smanar, 2023-02-12:
Your device have a display, it's not possible to compare value in the dpid 0x11 with some info from screen ?
Because the string don't have enought value for me to be usefull.

aIA78, 2023-03-05:
Hello, continuing my investigations and following your suggestions I have looked at all those data traces that have not been decoded with the corresponding DPIP... and I have found the following!!!

`23:10:34:663 APS-DATA.indication srcAddr: 0x0e9f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -64`
`23:10:34:665 	asdu: 09420200e50600000f0000080b0000016003d50022880945`

Focusing here:

`asdu: 09420200e50600000f0000080b0000016003d50022880945`

Comparing to previous capture and decode

`21:44:58:627 	asdu: 095d020042690200040000138b`
`21:44:58:632 TY_DATA_REPORT: seq 66, dpid: 0x69, type: 0x02, length: 4, val: 5003`

095d020042690200040000138b

095d0200 42 69 02 0004 0000138b

095d0200 - ??

42 - Seq (in HEX): 66 (in DEC)

69 - DPID (in HEX): 105 (in DEC)

02 - Type (in HEX): 02 (in DEC)

0004 - Length (in HEX): 4 (in DEC) ( 4 bytes)

0000138b - Value (in HEX): 5003 (in DEC)

This DPID corresponds to the network frequency. Just divide by 100 to get Hertz: 50.03 Hz.

Continuing with our example:

09420200e50600000f0000080b0000016003d50022880945

09420200 e5 06 00000f 0000080b0000016003d50022880945

09420200 - ????

e5 - Seq (in HEX): 229 (in DEC)

06 - DPID (in HEX): 06 (in DEC)

00 - Type (in HEX): 00 (in DEC)

000f - Length (in HEX): 15 (in DEC) (15 bytes)

The rest of the information I had to go making groups of different lengths, by trial and error, until I found something coherent and that coincided with the values ​​on the display of the device:

0000080b0000016003d50022880945

8 bytes: 0000080b (in HEX): 2059 (in DEC): Active Power (W)
 
8 bytes: 00000160  (in HEX): 352 (in DEC): Reactive Power (VAr) - ??? Not sure

2 bytes: 03d5 (in HEX): 981 (in DEC): 0.981 Power factor

3 bytes: 002288 (in HEX): 8840 (in DEC): 8.840 Current (A)

2 bytes: 0945 (in HEX): 2373 (in DEC): 237.3 Voltage (V)

To check the values ​​obtained I apply the basic power formulas:

V x I x cosφ = Active Power; Power Factor = cos φ

237.3 * 8.840 * 0.981=2057.87 (match)

Power Factor = cos φ

φ = arccos (Power Factor); φ = arccos(0.981) = 11.18676

V x I x sinφ = Reactive Power; sinφ = sin (11.18676) = 0.194

237.3 * 8.840 * 0.194 = 406.96 VAr (not match)

I will do more tests to see if the value that does not match is or is not the reactive power.

Also in a few days I will try to install deconz on a separate raspberry pi to test modifications in the code.

Smanar, 2023-03-05:
0942 02 00 e5 0600 000f 0000080b0000016003d50022880945

02 is probably the command  0x02 is the tuya answer on request, 0x01 is the normal reporting.
00 is a status, always 0x00

BTW I don't remember, are able to compile the code to test a code change to support RAW request ? The 0x00 type

>00 - Type (in HEX): 00 (in DEC)

aIA78, 2023-03-05:
Currently I can't compile code because I use Deconz in a HomeAssistant addon.
I am going to try to install deconz externally on another Raspberry to be able to compile code. I think I'll have it configured in a few days.

Smanar, 2023-03-05:
Ok, have started the code https://github.com/Smanar/deconz-rest-plugin/tree/tuya_raw
But tell me before making compilation.

closed 2023-04-04 by github-actions[bot]

BabaIsYou, 2023-05-29:
> "manufacturername": ["_TZE200_lsanae15", "_TZE200_bkkmqmyo", "_TZE200_eaac7dkw"],

It seems that these devices doesn't really shared the same DPIPs : 
_TZE200_bkkmqmyo & _TZE200_eaac7dkw https://github.com/sjorge/zigbee-herdsman-converters/blob/3c42ea14c74f3e23c9329b8ddcf6c6ef1fe7bbc0/devices/tuya.js#L3317

_TZE200_lsanae15 https://github.com/sjorge/zigbee-herdsman-converters/blob/3c42ea14c74f3e23c9329b8ddcf6c6ef1fe7bbc0/devices/tuya.js#L3347

BabaIsYou, 2023-05-30:
> Ok, have started the code https://github.com/Smanar/deconz-rest-plugin/tree/tuya_raw

Hi. Do you think that something looking like this in js could do the trick without compiling for values depending on dpid 6 ?

const buffer = Buffer.from(Attr.val, 'hex'); (not quite sure about the real encoding we received from "parse")
R.item('config/voltage').val = Math.max(0, (buffer[1] | buffer[0] << 8) / 10)
R.item('config/current').val = Math.max(0,buffer[4] | buffer[3] << 8) / 1000)
R.item('config/power').val = Math.max(0,(buffer[7] | buffer[6] << 8))

BabaIsYou, 2023-05-30:
I know it's quite old, but we had a new request 2 days ago #7014

Smanar, 2023-05-30:
Ha yes, so will see with DDF test.
If he have same issue you will be not be able to see usefull logs.