dresden-elektronik/deconz-rest-plugin #6532: New Device  Tuya Smoke Alarm YG400A

FelixHaimmerer, 2022-11-14:
## Device
https://zigbee.blakadder.com/Tuya_YG400A.html

- Product name: Smoke Alarm YG400A
- Manufacturer: Tuya
- Model identifier: _TZE200_ntcy3xu1
- Device type :
  - Sensor
  - Siren

## Screenshots
[image]

### Basic
- detect smoke
- show batterie state

### Alarms
- ring smoke alarm

FelixHaimmerer, 2022-12-09:
Sorry for the late response. 

here are the requested screenshots:

[image]
[image]
[image]

Thanks!

Mimiix, 2022-12-09:
Please add all screenshots.

FelixHaimmerer, 2022-12-09:
[image]

Sorry I have never done that bevore. Ist this what you meant? Thanks! :)

Mimiix, 2022-12-16:
> [image]
> 
> Sorry I have never done that bevore. Ist this what you meant? Thanks! :)

We need to have a screenshot of every cluster. https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

Mimiix, 2022-12-16:
Nevermind, i'll close this issue and re-open : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6236

closed 2022-12-16 by Mimiix