dresden-elektronik/deconz-rest-plugin #6536: Hue Festavia

ebaauw, 2022-11-15:
## Device

- Product name: Hue white and color ambiance gradient festavia string lights 250 LEDs
- Manufacturer: Signify Netherlands B.V.
- Model identifier: LCX012
- Device type : Light

## Screenshots
Will follow once I've paired it with deCONZ.

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

## Hue bridge
Under API v1, the Hue bridge exposes the gradient light string as:
```
{
  "capabilities": {
    "certified": true,
    "control": {
      "colorgamut": [
        [
          0.6915,
          0.3083
        ],
        [
          0.17,
          0.7
        ],
        [
          0.1532,
          0.0475
        ]
      ],
      "colorgamuttype": "C",
      "ct": {
        "max": 500,
        "min": 153
      },
      "mindimlevel": 5000
    },
    "streaming": {
      "proxy": true,
      "renderer": true
    }
  },
  "config": {
    "archetype": "stringlight",
    "direction": "omnidirectional",
    "function": "decorative",
    "startup": {
      "configured": true,
      "mode": "safety"
    }
  },
  "manufacturername": "Signify Netherlands B.V.",
  "modelid": "LCX012",
  "name": "Hue gradient string light 1",
  "productid": "Philips-LCX012-1-DSLECLv1",
  "productname": "Hue gradient string light",
  "state": {
    "alert": "select",
    "bri": 254,
    "colormode": "xy",
    "ct": 500,
    "effect": "none",
    "hue": 62234,
    "mode": "homeautomation",
    "on": true,
    "reachable": true,
    "sat": 215,
    "xy": [
      0.5623,
      0.2792
    ]
  },
  "swconfigid": "6729D92A",
  "swupdate": {
    "lastinstall": "2022-11-15T14:07:44",
    "state": "noupdates"
  },
  "swversion": "1.100.4",
  "type": "Extended color light",
  "uniqueid": "00:17:88:01:0d:62:77:e0-0b"
}
```

The v2 API reports:
```
{
    "errors": [],
    "data": [
        {
            "id": "993e4f28-c62e-4a2f-8ac0-a03541caa568",
            "id_v1": "/lights/3",
            "owner": {
                "rid": "802dddea-06ef-4545-a28d-5c074a8665d1",
                "rtype": "device"
            },
            "metadata": {
                "name": "Hue gradient string light 1",
                "archetype": "string_light"
            },
            "on": {
                "on": true
            },
            "dimming": {
                "brightness": 100.0,
                "min_dim_level": 5.0
            },
            "dimming_delta": {},
            "color_temperature": {
                "mirek": null,
                "mirek_valid": false,
                "mirek_schema": {
                    "mirek_minimum": 153,
                    "mirek_maximum": 500
                }
            },
            "color_temperature_delta": {},
            "color": {
                "xy": {
                    "x": 0.5624,
                    "y": 0.2792
                },
                "gamut": {
                    "red": {
                        "x": 0.6915,
                        "y": 0.3083
                    },
                    "green": {
                        "x": 0.17,
                        "y": 0.7
                    },
                    "blue": {
                        "x": 0.1532,
                        "y": 0.0475
                    }
                },
                "gamut_type": "C"
            },
            "dynamics": {
                "status": "none",
                "status_values": [
                    "none",
                    "dynamic_palette"
                ],
                "speed": 0.0,
                "speed_valid": false
            },
            "alert": {
                "action_values": [
                    "breathe"
                ]
            },
            "signaling": {},
            "mode": "normal",
            "gradient": {
                "points": [
                    {
                        "color": {
                            "xy": {
                                "x": 0.5624,
                                "y": 0.2792
                            }
                        }
                    },
                    {
                        "color": {
                            "xy": {
                                "x": 0.3677,
                                "y": 0.1621
                            }
                        }
                    },
                    {
                        "color": {
                            "xy": {
                                "x": 0.1975,
                                "y": 0.0691
                            }
                        }
                    },
                    {
                        "color": {
                            "xy": {
                                "x": 0.1589,
                                "y": 0.0504
                            }
                        }
                    },
                    {
                        "color": {
                            "xy": {
                                "x": 0.1548,
                                "y": 0.1083
                            }
                        }
                    }
                ],
                "mode": "interpolated_palette",
                "points_capable": 5,
                "mode_values": [
                    "interpolated_palette",
                    "random_pixelated"
                ],
                "pixel_count": 250
            },
            "effects": {
                "status_values": [
                    "no_effect",
                    "candle",
                    "fire",
                    "sparkle"
                ],
                "status": "no_effect",
                "effect_values": [
                    "no_effect",
                    "candle",
                    "fire",
                    "sparkle"
                ]
            },
            "timed_effects": {
                "status_values": [
                    "no_effect",
                    "sunrise"
                ],
                "status": "no_effect",
                "effect_values": [
                    "no_effect",
                    "sunrise"
                ]
            },
            "type": "light"
        }
    ]
}
```
Wauw, a pixel_count of 250, suggesting each of the LEDs is individually addressable (at least by the device firmware).

## Hue app
The Hue app shows an additional _Sparkle_ effect, as expected.  This seems to be encoded like the other effects, but with hex value 0x0a.  The gradient light strip doesn't accept this value, however the Hue bridge is currently upgrading the led string firmware - maybe there's new firmware for the gradient led strip as well?

The app shows a new _Style_ setting with values _Linear_ and _Scattered_.  When selecting _Linear_, the 0th point becomes 02:00:00 instead of 00:00:00.

ebaauw, 2022-11-19:
```
{
  "capabilities": {
    "alerts": [
      "none",
      "select",
      "lselect",
      "blink",
      "breathe",
      "okay",
      "channelchange",
      "finish",
      "stop"
    ],
    "bri": {
      "min_dim_level": 5
    },
    "color": {
      "ct": {
        "computes_xy": true,
        "max": 500,
        "min": 153
      },
      "effects": [
        "none",
        "colorloop",
        "candle",
        "fireplace",
        "loop",
        "sunrise",
        "sparkle"
      ],
      "gamut_type": "C",
      "gradient": {
        "max_segments": 31,
        "pixel_count": 250,
        "pixel_length": 800,
        "styles": [
          "linear",
          "scattered",
          "mirrored"
        ]
      },
      "modes": [
        "ct",
        "effect",
        "gradient",
        "hs",
        "xy"
      ],
      "xy": {
        "blue": [
          0.1532,
          0.0475
        ],
        "green": [
          0.17,
          0.7
        ],
        "red": [
          0.6915,
          0.3083
        ]
      }
    }
  },
  "colorcapabilities": 31,
  "config": {
    "bri": {
      "execute_if_off": true,
      "startup": "previous"
    },
    "color": {
      "ct": {
        "startup": "previous"
      },
      "execute_if_off": true,
      "gradient": {
        "reversed": false
      },
      "xy": {
        "startup": "previous"
      }
    },
    "groups": [
      "0",
      "256"
    ],
    "on": {
      "startup": false
    }
  },
  "ctmax": 500,
  "ctmin": 153,
  "etag": "9369379178014d87ab4ed3b6fb9b9c62",
  "hascolor": true,
  "lastannounced": null,
  "lastseen": "2022-11-19T20:40Z",
  "manufacturername": "Signify Netherlands B.V.",
  "modelid": "LCX012",
  "name": "Hue Festavia",
  "productid": "Philips-LCX012-1-DSLECLv1",
  "state": {
    "alert": "none",
    "bri": 241,
    "colormode": "effect",
    "ct": 367,
    "effect": "sunrise",
    "gradient": {
      "color_adjustment": 6,
      "offset": 31,
      "offset_adjustment": 2,
      "points": [
        [
          0.4771,
          0.4137
        ],
        [
          0.4633,
          0.4114
        ],
        [
          0.4511,
          0.4085
        ],
        [
          0.44,
          0.4052
        ],
        [
          0.4299,
          0.4019
        ]
      ],
      "segments": 31,
      "style": "linear"
    },
    "hue": 8109,
    "on": true,
    "reachable": true,
    "sat": 168,
    "xy": [
      0.4475,
      0.4074
    ]
  },
  "swversion": "1.101.1",
  "type": "Extended color light",
  "uniqueid": "00:17:88:01:0d:62:77:e0-0b"
}
```

TheJulianJES, 2022-12-28:
> OTAU Image Type is 0x0123. The Hue bridge did update the firmware to 1.101.1, while I had the sniffer running, but it didn't seem to capture the firmware file location. Did they change the download logic?

Yes, update and server communication logic was completely reworked and is no longer using the ICP portal.
All updated Hue images are now available [here](https://github.com/TheJulianJES/zigbee-OTA/blob/b569796dcc29f8a4f96f17ef3ef996612be4c749/index.json). (Search for URLs with meethue.com)

(Note: [The PR](https://github.com/Koenkk/zigbee-OTA/pull/212) was temporarily reverted due to Z2M being unable to parse the new Hue ZCL v8 images. ZHA works fine though. As soon as the issue is fixed in Z2M (and the fix released), the images should be added back into the main repo.)

ebaauw, 2022-12-29:
Thanks.  How did you discover the URLs, especially the UUID part?

closed 2023-01-07 by manup