dresden-elektronik/deconz-rest-plugin #6538: DDF Editor not reloading DDF from disk

Zehir, 2022-11-16:
## Describe the bug

DDF editor don't reload the DDF from the file when it's was updated with an external editor.

## Steps to reproduce the behavior

I open it in the DDF Editor, I see the correct model ID.
I update the model ID with a text editor and save. If I open it again in VSCode it's correctly updated.
I close the DDF Editor and open the DDF Editor. (Optional, happen without doing that too).
I press File -> Open and load the file SE_shutter.json
The model ID is still the old value.
I rename the file to SE_shutter_v2.json.
I open the v2 file with the DDF Editor. I correctly see the new Model ID.

## Expected behavior

I expect that the model ID was updated in the GUI

## Screenshots

[image]

## Environment

 - Host system: (Raspberry Pi / PC / NAS) Raspberry Pi 4 / Windows 11
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine) Home Assistant OS / Windows 11
 - Firmware version: (26xxyy00) 0x26780700
 - deCONZ version: (2.xx.yy) 2.16.1 / 2.19.1, 2.19.2
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II) ConBee II
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II Yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? No

## deCONZ Logs

## Additional context

BabaIsYou, 2022-11-23:
This may explain some strange behavior that I have seen with handling several DDFs over the past few months.

Smanar, 2022-12-09:
Another issue with the editor (not the same) https://forum.phoscon.de/t/deconz-ddf-editor-looses-some-information/2971

Zehir, 2022-12-09:
> Another issue with the editor (not the same) https://forum.phoscon.de/t/deconz-ddf-editor-looses-some-information/2971

Yes I had that issue too but forgot about it

Zehir, 2022-12-31:
Up

BabaIsYou, 2023-01-23:
Up

BabaIsYou, 2023-02-14:
Up