dresden-elektronik/deconz-rest-plugin #6539: [Sensor] SOLE HOME Smart Wireless Temperature and Humidity Sensor

Mephisto090, 2022-11-16:
## Device

- Product name: SOLE HOME Smart Wireless Temperature and Humidity Sensor
- Manufacturer: _TZ3000_bguser20
- Model identifier: TS0201
- Device type : Sensor

## Product Parameters
Quiescent current: <= 10uA
Alarm current: <= 15mA
Low voltage and undervoltage <=2,6V
Zigbee working Frequency: 2.400GHz - 2.484GHz
Detection Temp: -10°C - +60°C
Detection Hum.: 0 - 100% RH
Working Temp: -10°C - +50°C
Working Hum.: max. 95% RH

## Screenshots

[image]

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature
[image]
[image]

### Groups

### Scenes

### On/Off
[image]

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2022-11-16:
Hello, can you try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZ3000_yd2e749y", "_TZ3000_fllyghyj", "_TZ3000_lbtpiody", "_TZ3000_5nrcorgu", "_TZ3000_ena8vgqb", "_TZ3000_bguser20"],
  "modelid": ["TS0201", "TS0201", "TS0201", "TS0201", "TS0201", "TS0201"],
  "vendor": "Tuya",
  "product": "Temperature and humidity sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          }
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          }
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    }
  ]
}
```

Mephisto090, 2022-11-16:
Where can I find instructions on how to test it?

Smanar, 2022-11-16:
Depend of OS

> All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

If you have HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

You just need to create a text file caled what_you_want_.json in the "devices" folder.
Then restart deconz and include the device.

Mephisto090, 2022-11-16:
The manufacturer still looks wrong. I'll see if the temperature and humidity update automatically

[image]
[image]

Smanar, 2022-11-16:
>The manufacturer still looks wrong

Nono, it's the good one ^^

But idk why the picture is a motion sensor ....

Have added your device here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6244

Mephisto090, 2022-11-16:
All my Multisensors are motion sensor

[image]

Mephisto090, 2022-11-16:
> All my Multisensors are motion sensor. The Other one all Sonoff
> 
> [image]

Should I create a new issues for this?

Smanar, 2022-11-16:
For the picture error ?
It's from phoscon, we haven't impact on this application, there is another github for it.

If it's not a problem for you, you can let it as it.

Mephisto090, 2022-11-16:
It looks like it doesn't automatically read the temperature. Even a manual "read" does not work. See screenshot. I'll wait 10 minutes to see if the Phoscon app shows an update

[image]

Mephisto090, 2022-11-16:
> For the picture error ? It's from phoscon, we haven't impact on this application, there is another github for it.
> 
> If it's not a problem for you, you can let it as it.

Do you have a link, so I can open a ticket there?

Smanar, 2022-11-16:
This device is sleeping like all sensors, you need to awake it in same time you press the "read" button (with using the reset button generaly)

So the phoscon github is closed https://github.com/dresden-elektronik/phoscon-app-beta and they say to use now https://forum.phoscon.de/c/phoscon/17

And yes 15mn is a lot, if the temperature move. Have you tried to heat the sensor with your hands ?

I m seing this device need the tuya unlock sequence, so if the values don't move, I will be not possible to add it to deconz.

Mephisto090, 2022-11-16:
> And yes 15mn is a lot, if the temperature move. Have you tried to heat the sensor with your hands ?

Yes that didn't work

[image]

Smanar, 2022-11-16:
Ok, so was the missing "tuya unlock sequence".
Sorry but this device can't be added for the moment.

Mephisto090, 2022-11-20:
What do you mean?

Smanar, 2022-11-20:
This device need the coordinator ask for 5/6 attributes in a good order to unlock all feature on it.

closed 2022-12-20 by github-actions[bot]