dresden-elektronik/deconz-rest-plugin #6541: False detection in every 55 minutes with Lumi Motion Sensors

tosion, 2022-11-16:
## Describe the bug

I have a strange behavior with couple Xiaomi Mi Smart Home Occupancy Sensors. They are outside in my summer cabin and when detected movement security cameras send me a notification of the area. I have got false detection but I have thought that they are just because of sun or other reasons. Until now:
I realized the false detection follow a pattern – they appear in every 55 minutes when they start but they don’t detect all the time (a pic below). I do not know any reason for that. I have more of those same sensors in summer cabin and home which do not follow this pattern. Any idea what is going on here?

Below logs for two identical LUMI sensors: Motion11 without false positives and Motion12 with false positives. Every 55 minutes I can see two identical logs (in blue below). In addition to Motion12 (red below) a log with “presence”:true following one what I did not copy with “presence”:false. Very same time with this red log I get the false detection.

So what should I think about this: there is maybe an update in every 55 minutes which causes these false detection but how to get rid of them is the Question.

## Steps to reproduce the behavior
False detection in every 55 minutes

## Expected behavior
No false detection

## Screenshots
[image]

## Environment

 - Host system: NUC
 - Running method: Home Assistant deCONZ Add-on 
 - Firmware version: (26xxyy00)
 - deCONZ version: 2.19.1
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? Eaton UPS

## deCONZ Logs
[image]

## Additional context
No additional context

tosion, 2022-12-15:
This is still a problem so let's open it.

As a quick fix I created an input_boolean.blocksensor which will turn on every 55 minutes for few seconds after the LUMI-sensor turns on. I use that as a condition to prevent these “false alarms”.

tosion, 2022-12-15:
It is still a problem so please do not close it ...

Closed #6541<https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6541> as completed.

ebaauw, 2022-12-15:
Those aren’t deCONZ logs.  We would need to see the communication between deCONZ and the sensors.

But in all likelihood, the sensor is acting funny because it’s battery is running low.  Don’t trust the reported battery on button-cell-powered devices.  Use a battery tester to check whether the cell still provides at least 3.0V, or simply replace the battery.

tosion, 2022-12-15:
Thanks for a response! I think the logs are Deconz logs from Homeassistant. Is there some other logs I can get with HA-addon? I sure have changed the battery and I always measure with my Fluke-multimeter voltage. I have around 100 Zigbee sensors for couple years so I am quite familiar with the the problem we have with battery charge.

Below the method I used for log in HA:

[image]

tosion, 2022-12-15:
This is really not a big problem for me – I have extra new sensors in my closet and I can easily replace these ”funny acting sensors”. I just wanted to raise up this problem because it took me a very long time to understand that these false detections have a pattern. With this ”input_boolean.block sensor” I get way less disturbing false alarms now. And I still have many other motion sensors in my real estate to be aware of the status there.

But really I would like to educate myself how to get ”better logs” for Deconz with HA-addon. In my previous post is is ”my way”,  so please if there is a ”other way” let me know.

ebaauw, 2022-12-15:
The 55 minutes pattern suggests the Xiaomi special attribute report, and your log shows `battery` at 95, so I would double-check the battery.

I don't know HA, but deCONZ logs to stdout/stderr.  Depending on how you start it, that output is captured in some log file.  You can specify the level of logging with startup parameters (see https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-command-line-parameters#available-debug-parameters), or in the _Debug View_ in the deCONZ GUI.  We would need APS and APS_L2 to see the Zigbee traffic from the sensor.

tosion, 2022-12-15:
OK - Thanks!

Mimiix, 2022-12-15:
Re-opened.

closed 2023-01-13 by github-actions[bot]