dresden-elektronik/deconz-rest-plugin #6544: LK Wiser Dubble relay (545D6514) not working

mihaimaersk, 2022-11-17:
## Describe the bug

LK Wiser Double relay (545D6514) not working (i tried two different devices and same thing). It pairs with the deconz gateway but i can't send commands to it. If i get into deconz into the cluster info and try to toggle on/off i either get timeout or send failed.

## Steps to reproduce the behavior

1. Reset the LK Wiser unit by pressing the upper corner button 3 times then the 4th time hold it down until it blinks red
2. Enable adding devices in deconz
3. Tripple click upper left corner button 
4. Wait till the device get's added to deconz.

## Expected behavior

I would expect that i could control the device.

## Screenshots

[image]

[image]

[image]

## Environment

 - Host system: (Raspberry Pi )
 - Running method: (Raspbian )
 - Firmware version: (26400500)
 - deCONZ version: (2.19.2)
 - Device: (RaspBee I)

## deCONZ Logs

15:11:30:000 APS-DATA.request id: 18, addrmode: 0x03, addr: 0x000b57fffe38fcb3, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
15:11:30:122 APS-DATA.confirm id: 18, status: 0x00 SUCCESS
15:11:30:124 APS-DATA.confirm request id: 18 -> erase from queue
15:11:30:150 APS-DATA.indication srcAddr: 0x02cd, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -52
15:11:30:151 APS-DATA.request id: 18 erase from queue
15:11:30:199 APS-DATA.request id: 20, addrmode: 0x03, addr: 0x000b57fffe38fcb3, profile: 0x0104, cluster: 0x0300, ep: 0x01 -> 0x01 queue: 0 len: 9 tx.options 0x00
15:11:30:300 APS-DATA.confirm id: 20, status: 0x00 SUCCESS
15:11:30:301 APS-DATA.confirm request id: 20 -> erase from queue
15:11:30:332 aps request id: 20 finished, erase from queue
15:11:30:334 APS-DATA.indication srcAddr: 0x02cd, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0300, lqi: 255, rssi: -52
15:11:31:099 APS-DATA.request id: 22, addrmode: 0x03, addr: 0x00158d00013c7c98, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
15:11:31:200 APS-DATA.confirm id: 22, status: 0x00 SUCCESS
15:11:31:201 APS-DATA.confirm request id: 22 -> erase from queue
15:11:31:229 APS-DATA.indication srcAddr: 0x0406, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -52
15:11:31:230 APS-DATA.request id: 22 erase from queue
15:11:31:234 APS-DATA.request id: 24, addrmode: 0x02, addr: 0x0406, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x04
15:11:31:309 APS-DATA.confirm id: 24, status: 0x00 SUCCESS
15:11:31:310 APS-DATA.confirm request id: 24 -> erase from queue
15:11:31:327 aps request id: 24 finished, erase from queue
15:11:32:399 APS-DATA.request id: 25, addrmode: 0x03, addr: 0x680ae2fffea51c89, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
15:11:32:499 APS-DATA.confirm id: 25, status: 0x00 SUCCESS
15:11:32:500 APS-DATA.confirm request id: 25 -> erase from queue
15:11:32:549 aps request id: 25 finished, erase from queue
15:11:32:550 APS-DATA.indication srcAddr: 0x042a, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 191, rssi: -67
15:11:33:309 APS-DATA.indication srcAddr: 0x9dd9, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0x0001, lqi: 255, rssi: -52
15:11:33:499 APS-DATA.request id: 28, addrmode: 0x03, addr: 0x0017880108c9b7a2, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 0 len: 5 tx.options 0x00
15:11:33:631 APS-DATA.confirm id: 28, status: 0x00 SUCCESS
15:11:33:632 APS-DATA.confirm request id: 28 -> erase from queue
15:11:33:659 aps request id: 28 finished, erase from queue
15:11:33:660 APS-DATA.indication srcAddr: 0x06f8, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 191, rssi: -67
15:11:34:800 APS-DATA.request id: 30, addrmode: 0x03, addr: 0x0017880108ce1ce8, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 0 len: 5 tx.options 0x00
15:11:34:899 APS-DATA.confirm id: 30, status: 0x00 SUCCESS
15:11:34:900 APS-DATA.confirm request id: 30 -> erase from queue
15:11:34:949 APS-DATA.indication srcAddr: 0x411b, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 239, rssi: -61
15:11:34:950 APS-DATA.request id: 30 erase from queue
15:11:36:000 APS-DATA.request id: 32, addrmode: 0x03, addr: 0x0017880108c9b16a, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 0 len: 5 tx.options 0x00
15:11:36:141 APS-DATA.confirm id: 32, status: 0x00 SUCCESS
[... 316 lines cut ...]
15:12:30:436 APS-DATA.confirm id: 129, status: 0x00 SUCCESS
15:12:30:437 APS-DATA.confirm request id: 129 -> erase from queue
15:12:30:476 aps request id: 129 finished, erase from queue
15:12:30:478 APS-DATA.indication srcAddr: 0x042a, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 191, rssi: -67
15:12:30:481 ZCL got data for node=0x042A, cl=0x0006, at=0x0000, status=0x00, type=0x10
15:12:31:400 APS-DATA.request id: 131, addrmode: 0x03, addr: 0x0017880108c9b7a2, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 0 len: 5 tx.options 0x00
15:12:31:499 APS-DATA.confirm id: 131, status: 0x00 SUCCESS
15:12:31:500 APS-DATA.confirm request id: 131 -> erase from queue
15:12:31:529 APS-DATA.indication srcAddr: 0x06f8, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 239, rssi: -61
15:12:31:531 APS-DATA.request id: 131 erase from queue
15:12:31:531 ZCL got data for node=0x06F8, cl=0x0006, at=0x0000, status=0x00, type=0x10
15:12:32:600 APS-DATA.request id: 133, addrmode: 0x03, addr: 0x0017880108ce1ce8, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 0 len: 5 tx.options 0x00
15:12:32:689 APS-DATA.confirm id: 133, status: 0x00 SUCCESS
15:12:32:691 APS-DATA.confirm request id: 133 -> erase from queue
15:12:32:695 aps request id: 133 finished, erase from queue
15:12:32:749 APS-DATA.indication srcAddr: 0x411b, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 239, rssi: -61
15:12:32:751 ZCL got data for node=0x411B, cl=0x0006, at=0x0000, status=0x00, type=0x10
15:12:33:800 APS-DATA.request id: 135, addrmode: 0x03, addr: 0x0017880108c9b16a, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 0 len: 5 tx.options 0x00
15:12:33:889 APS-DATA.confirm id: 135, status: 0x00 SUCCESS
15:12:33:891 APS-DATA.confirm request id: 135 -> erase from queue
15:12:33:895 aps request id: 135 finished, erase from queue
15:12:33:939 APS-DATA.indication srcAddr: 0x44d1, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 127, rssi: -75
15:12:33:941 ZCL got data for node=0x44D1, cl=0x0006, at=0x0000, status=0x00, type=0x10

## Additional context

Smanar, 2022-11-17:
On your capture the name is grayed no ? Mean the device have DC.

mihaimaersk, 2022-11-17:
The device have AC. I tried ZHA with another bridge and the device itself works fine just not with deconz for some unknown reason.

Smanar, 2022-11-17:
This device have too a reset and a different pairing procedure ?

Mimiix, 2022-11-17:
Looks like something else is in play here 

cluster: 0x0006 cmd: 0x01
15:12:18:677 APS-DATA.request rejected, destination 0x0000000000000000 is zombie node

Smanar, 2022-11-18:
You have used the same raspbee 1 during your test with the others apps ?

```

1-    Reset the LK Wiser unit by pressing the upper corner button 3 times then the 4th time hold it down until it blinks red
2-   Enable adding devices in deconz
3-    Tripple click upper left corner button
4-    Wait till the device get's added to deconz.
```

Can you try with switching 1 and 2 ?

donnib, 2022-11-19:
No, the other gateway was a sonoff ZigBee gateway. I have tried enabling pairing in deconz then reset and as I mentioned the device does get paired but behaves in a way that's not useable. It gets added to deconz but then dies and can't communicate with it.

mihaimaersk, 2022-11-19:
No, the other gateway was a sonoff ZigBee gateway. I have tried enabling pairing in deconz then reset and as I mentioned the device does get paired but behaves in a way that's not useable. It gets added to deconz but then dies and can't communicate with it.

mihaimaersk, 2022-12-08:
@manup any ideas ? I mean it should work according to your change https://github.com/dresden-elektronik/deconz-rest-plugin/commit/57c60a53f28bb29bef861441bc4b39744e8fa54c

also usage info when/if we get it to work, do you know if we will get events from the buttons itself separately from the switch itself ?

hanskroner, 2022-12-08:
> 
> @hanskroner are you using one of these yourself ? if so does it work for you ?

Sorry, no. I was provided a gateway, a dimmer, and a double relay a while back by another user but he wanted the hardware back once the functionality he was interested in got implemented. The only Wiser devices I have are the 4-button battery ones.

closed 2023-01-06 by github-actions[bot]