dresden-elektronik/deconz-rest-plugin #6546: deCONZ REST API allows clashing device names

endreszabo, 2022-11-18:
## Describe the bug

The REST API allows changing the device names using the PUT method on the `name` attribute. This works fine. However, it does not check if there is another device with the same name. The Phoscon app checks it (using client-side JS) and does not let me submit the form as long as the name-to-be clashes with another device.

## Steps to reproduce the behavior

Rename a device to another existing device using REST. deCONZ renames the device, which will then be indistinguishable from the former one. Not sure about any other problems this might cause.

## Expected behavior

REST API call should return an error, stating that another device exists with the same name.

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: custom (native)
 - Firmware version: 26780700
 - deCONZ version: 2.18.02-1
 - Device: ConBee II
 - Do you use a USB extension cable: yes
 - Are any other USB or serial devices connected to the host system? nope

## deCONZ Logs

not applicable to the problem

## Additional context

Smanar, 2022-11-18:
IDK, I don't see the problem, if you want to call all your device "stuff", it's your choice.
Device have handled with the ID, never with the name.

endreszabo, 2022-11-21:
Fair enough, I don't depend on them being unique, either. Thanks for confirming it. :+1:

closed 2022-11-21 by endreszabo