dresden-elektronik/deconz-rest-plugin #6552: "ConBee II" light appearing with "Unknown" type instead of "Configuration tool"

Thomas-Vos, 2022-11-20:
## Describe the bug
For some users the "Conbee II" light is appearing with type "Unknown". Should this not be "Configuration tool"? On my setup it is.

Report from a user:
```
    "manufacturername": "dresden elektronik",
    "modelid": "ConBee II",
    "name": "Unknown 13",
    "state": {
      "reachable": true
    },
    "swversion": "0x26780700",
    "type": "Unknown",
```

## Steps to reproduce the behavior
None

## Expected behavior
Type equals `Configuration tool`

## Screenshots
None

## Environment

 - Host system: Raspberry Pi, maybe others.
 - Running method: Unsure
 - Firmware version: Unsure
 - deCONZ version: Unsure
 - Device: All were reported to me: ConBee I / ConBee II / RaspBee I / RaspBee II
 - Do you use an USB extension cable: not relevant
 - Is there any other USB or serial devices connected to the host system? If so: Which? Unsure

## deCONZ Logs

None

## Additional context

Mimiix, 2022-11-20:
It's like that on mine . Can confirm this happens.

popy2k14, 2022-11-21:
I can also confirm this.
(I think i reported to @Thomas-Vos 😅 )

ebaauw, 2022-11-21:
As I mentioned on Discord, please double-check the _Endpoints_ tab of the _deCONZ Network Settings_ in the GUI.

Smanar, 2022-11-23:
Yep, but lot of users, haven't the GUI, and this issue can causes problem on third app that check for "Configuration tool"

popy2k14, 2022-11-23:
CHecked the Endpoints tab of the deCONZ Network Settings in the GUI.
There are just hex numbers and no string indicating "unknown" or anything else.

What to look for?

ebaauw, 2022-11-23:
Please post a screenshot.  Basically these numbers define how the device advertises itself on the Zigbee network: what endpoints with what profile, device type, input (server) and output (client) clusters.  The definitions of these numbers (translating them to text) is in `general.xml`.

ebaauw, 2022-11-23:
I should like this:
[image]

As I posted on Discord: _Device ID_ 0x0005 in the Home Automation profile (0x0104) is the Configuration Tool.  If memory serves, you might need to Save and then leave and rejoin the network before the settings take.  It's a bit messy.  Haven't touched this in years and would advise anyone to stay away from it.

popy2k14, 2022-11-23:
It's exactly the same on my system like in @ebaauw screenshot.
Here is the info from the api (removed unessasary things):

```
{
   "manufacturername":"dresden elektronik",
   "modelid":"ConBee II",
   "name":"Configuration tool",
   "state":{
      "reachable":true
   },
   "swversion":"0x26780700",
   "type":"Unknown",
}
```

In my case, name is correct but type is unknown.

ebaauw, 2022-11-23:
And it shows as _Configuration Tool_ in the GUI?

Mimiix, 2022-11-23:
@ebaauw  My screen:
[image]

        "13": {
            "etag": "8beb967da3792b291ecec206862dee42",
            "hascolor": false,
            "lastannounced": null,
            "lastseen": "2022-11-23T19:00Z",
            "manufacturername": "dresden elektronik",
            "modelid": "ConBee II",
            "name": "Unknown 13",
            "state": {
                "reachable": true
            },
            "swversion": "0x26780700",
            "type": "Unknown",
            "uniqueid": "00:21:2e:ff:ff:04:d7:94-01"
        }

SwoopX, 2022-11-23:
Got the same as Dennis

ebaauw, 2022-11-24:
I’m at a loss.  The GUI, `uniqueid`, and absence of `state/on` and `state/alert` all indicate that the API plugin recognises the Configuration tool on startup.  But `type` doesn’t seem to be updated by `setHaEndpoint()` for previously existing `lights` resources.  I suspect it’s wrong in the database.  Normally I’d suggest deleting the node and repairing the device, but the node for the coordinator cannot be deleted.  Force-reading the descriptors from the GUI won’t update the `type` either.  The only possible action seems to be patching the database, not something I’d normally recommend.

Even more elusive is how the type came to be Unknown in the first place.  Had it always been that?  Or did it change with an upgrade, backup/restore, replacement of the coordinator device, move deCONZ to another server, or something?  Start of deCONZ without it finding ConBee maybe?  Has anyone seen this on a RaspBee?

Mimiix, 2022-11-24:
> Even more elusive is how the type came to be Unknown in the first place. Had it always been that? Or did it change with an upgrade, backup/restore, replacement of the coordinator device, move deCONZ [...]

Sometime it changed i suppose. However, it might have happened when my groups also went apeshit.  This was between 12.6 and 13.4. My groups numbers out of the sudden: 
[image]

That's the only time something major changed in my network.

Nevertheless: I am not going to patch my database.

ebaauw, 2022-11-24:
> But type doesn’t seem to be updated by setHaEndpoint() for previously existing lights resources. I suspect it’s wrong in the database.

I think it's actually set alright, but it's overwritten by `LightNode::jsonToResourceItems()`, which is called from `sqliteLoadLightNodeCallback()`.

ebaauw, 2022-11-25:
PR should resolve the symptoms, in that the value for `type` is no longer restored from the database, so the type will be set from the _Device Type_ when (re-) creating the `lights` resource on startup.  Still no clue how it came to be "Unknown" in the first place.

Thomas-Vos, 2022-12-17:
Great, thanks for fixing this Erik.

Thomas-Vos, 2023-01-16:
Not fixed

closed 2023-02-05 by manup