dresden-elektronik/deconz-rest-plugin #6555: Tuya Scene switches

machgy, 2022-11-20:
## Device

- Product name: IH-L400-P4
- Manufacturer:  _TZ3000_itb0omhv
- Model identifier: TS0041
- Device type : Switch
- https://www.aliexpress.com/item/1005004248050569.html?spm=a2g0o.order_list.0.0.2b7a1802Sy9qET

## Screenshots
[image]

[image]

[image]

### Basic
[image]

### Power
[image]

### Unknown
[image]

### On/Off
[image]

## No button maps
[image]

## DDF
[image]
[image]

## Other info
I have 1, 2 and 3 gang version, they are very same.

### 1 gang
- Product name: IH-L400-P4
- Manufacturer:  _TZ3000_itb0omhv
- Model identifier: TS0041
- Device type : Switch

### 2 gang

- Product name: IH-L400-P4
- Manufacturer:  _TZ3000_tzvbimpq
- Model identifier: TS0042
- Device type : Switch

### 3 gang

- Product name: IH-L400-P4
- Manufacturer:  _TZ3000_sj7jbgks
- Model identifier: TS0043
- Device type : Switch

Smanar, 2022-11-23:
Hello can you try with existing DDF and adding your model if ? ( you need to ad dit in the "modelid" and the "manufacturename"

2 gang > https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_t8hzpgnd_2gang_bat_switch.json

3 gangs > https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_gbm10jnj_3gang_remote.json

for the 1 gangs

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": " _TZ3000_itb0omhv",
  "modelid": "TS0041",
  "product": "Tuya 1 gangs switch battery",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },        
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}
```

machgy, 2022-11-23:
Hello,
I tried it, but after the restart, in the deCONZ the device started with "Untitled file". 
[image]
[image]
[image]
[image]
[image]

machgy, 2022-11-23:
> "manufacturername": " _TZ3000_itb0omhv",

One step closer, there is an unwanted space in the DDF: "manufacturername": " _TZ3000_itb0omhv",

machgy, 2022-11-23:
> Hello can you try with existing DDF and adding your model ? ( you need to add it in the "modelid" and the "manufacturename"
> 
> 2 gang > https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_t8hzpgnd_2gang_bat_switch.json
> 
> ```
>   "manufacturername": ["_TZ3000_t8hzpgnd", "_TZ3000_tzvbimpq"],
>   "modelid": ["TS0042", "TS0042"],
> ```
> 
> 3 gangs > https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_gbm10jnj_3gang_remote.json
> 
> ```
>   "manufacturername": ["_TZ3000_gbm10jnj", "_TZ3000_sj7jbgks"],
>   "modelid": ["TS0043", "TS0043"],
> ```
> 
> for the 1 gangs
> 
> ```
> {
>   "schema": "devcap1.schema.json",
>   "manufacturername": " _TZ3000_itb0omhv",
>   "modelid": "TS0041",
>   "product": "Tuya 1 gangs switch battery",
>   "sleeper": false,
>   "status": "Gold",
>   "subdevices": [
>     {
>       "type": "$TYPE_SWITCH",
>       "restapi": "/sensors",
>       "uuid": [
>         "$address.ext",
>         "0x01",
>         "0x0006"
>       ],
>       "items": [
>         {
>           "name": "attr/id"
>         },
>         {
>           "name": "attr/lastannounced"
>         },
>         {
>           "name": "attr/lastseen"
>         },
>         {
>           "name": "attr/manufacturername"
>         },
>         {
>           "name": "attr/modelid"
>         },
>         {
>           "name": "attr/name"
>         },
>         {
>           "name": "attr/swversion"
>         },
>         {
>           "name": "attr/type"
>         },
>         {
>           "name": "attr/uniqueid"
>         },
>         {
>           "name": "config/battery",
>           "awake": true,
>           "read": {
>             "at": "0x0021",
>             "cl": "0x0001",
>             "ep": 1,
>             "fn": "zcl"
>           },
>           "parse": {
>             "at": "0x0021",
>             "cl": "0x0001",
>             "ep": 1,
>             "eval": "Item.val = Attr.val / 2"
>           },
>           "default": 0
>         },        
>         {
>           "name": "config/on"
>         },
>         {
>           "name": "config/reachable"
>         },
>         {
>           "name": "state/buttonevent"
>         },
>         {
>           "name": "state/lastupdated"
>         }
>       ]
>     }
>   ],
>   "bindings": [
>     {
>       "bind": "unicast",
>       "src.ep": 1,
>       "cl": "0x0001",
>       "report": [
>         {
>           "at": "0x0021",
>           "dt": "0x20",
>           "min": 60,
>           "max": 3600,
>           "change": "0x00000001"
>         }
>       ]
>     },
>     {
>       "bind": "unicast",
>       "src.ep": 1,
>       "dst.ep": 1,
>       "cl": "0x0006"
>     }
>   ]
> }
> ```

same res

machgy, 2022-11-23:
>
Same results:

[image]
[image]
[image]

Smanar, 2022-11-23:
Oups, right my bad.

And yes, have checked again, I have already removed some of them, IDK why my copy/paste add a space evey time.

In the "devices" folder you will found a file button_maps.json (I haven't asked what is your OS, but it seem you don't have problem with it)
Inside you need to add the 3 devices

```
        "Tuya3gangMap": {
            "vendor": "Tuya",
            "doc": "3-gang remote",
            "modelids": ["_TZ3000_ee8nrt2l", "_TZ3000_ygvf9xzp", "_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0", "_TZ3000_rco1yzb1", "_TZ3000_gbm10jnj", "_TZ3000_itb0omhv", "_TZ3000_tzvbimpq", "_TZ3000_sj7jbgks"],
            "map": [
```

machgy, 2022-11-23:
You can see my map below. I use hassio with deCONZ plugin. The OS run the plugin in a docker container, so the next problem is loosing the modifications afrer a restart. Do I need restart to reload the map?

 "Tuya3gangMap": {
            "vendor": "Tuya",
            "doc": "3-gang remote",
            "modelids": ["_TZ3000_ee8nrt2l", "_TZ3000_ygvf9xzp", "_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_ad000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0", "_TZ3000_rco1yzb1", "_TZ3000_gbm10jnj" "_TZ3000_itb0omhv", "_TZ3000_tzvbimpq", "_TZ3000_sj7jbgks"],
            "map": [
                [1, "0x01", "ONOFF", "0xfd", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "B1 short"],
                [1, "0x01", "ONOFF", "0xfd", "1", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "B1 double"],
                [1, "0x01", "ONOFF", "0xfd", "2", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "B1 long"],
                [1, "0x02", "ONOFF", "0xfd", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "B2 short"],
                [1, "0x02", "ONOFF", "0xfd", "1", "S_BUTTON_2", "S_BUTTON_ACTION_DOUBLE_PRESS", "B2 double"],
                [1, "0x02", "ONOFF", "0xfd", "2", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "B2 long"],
                [1, "0x03", "ONOFF", "0xfd", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "B3 short"],
                [1, "0x03", "ONOFF", "0xfd", "1", "S_BUTTON_3", "S_BUTTON_ACTION_DOUBLE_PRESS", "B3 double"],
                [1, "0x03", "ONOFF", "0xfd", "2", "S_BUTTON_3", "S_BUTTON_ACTION_LONG_RELEASED", "B3 long"],
                [1, "0x04", "ONOFF", "0xfd", "0", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "B4 short"],
                [1, "0x04", "ONOFF", "0xfd", "1", "S_BUTTON_4", "S_BUTTON_ACTION_DOUBLE_PRESS", "B4 double"],
                [1, "0x04", "ONOFF", "0xfd", "2", "S_BUTTON_4", "S_BUTTON_ACTION_LONG_RELEASED", "B4 long"]

machgy, 2022-11-23:
> You can see my map below. I use hassio with deCONZ plugin. The OS run the plugin in a docker container, so the next problem is loosing the modifications afrer a restart. Do I need restart to reload [...]

WO restart I got the same result:
[image]

Smanar, 2022-11-23:
>You can see my map below. I use hassio with deCONZ plugin. The OS run the plugin in a docker container, so the next problem is loosing the modifications afrer a restart. Do I need restart to reload the map?

Yes sorry, and IDK if it's possible on the HA OS, I know DDF can be put in a persistent path, but there is only 1 possible file for the buttonmap.json.

But as I think it will be same (all tuyas are working same) if you don't found how to test, I can make a PR with the 3 DDF and the modification on the buttonmap.

Will be untested, but for me no reason for it don't work.

machgy, 2022-11-24:
> > You can see my map below. I use hassio with deCONZ plugin. The OS run the plugin in a docker container, so the next problem is loosing the modifications afrer a restart. Do I need restart to reload the map?
> 
> Yes sorry, and IDK if it's possible on the HA OS, I know DDF can be put in a persistent path, but there is only 1 possible path for the buttonmap.json.
> 
> But as I think it will be same (all tuyas are working same) if you don't found how to test, I can make a PR with the 3 DDF and the modification on the buttonmap.
> 
> Will be untested, but for me no reason for it don't work.

Hi, after the restart I losed all of the moidifications. So I cannot test it. What does PR mean? Is it a kind of udate?
I'm looking forward for it.

Smanar, 2022-11-24:
I have added your device here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6244
Without tests, but I hope it will be ok.

MacroScooter, 2022-12-21:
Did anybody get this device working?

TS0043 by _TZ3000_sj7jbgks 3 button scene switch

Smanar, 2022-12-21:
The DDF is not yet official, you have it here https://github.com/dresden-elektronik/deconz-rest-plugin/blob/b2761cd529982fa13eaf4b56059221220a7f3237/devices/tuya/_TZ3000_gbm10jnj_3gang_remote.json

MacroScooter, 2022-12-21:
Thank you for your help!
I already used this DDF before, in Phoscon the switch is disvovered and appears BUT its not possible to integrate it in Phoscon-Scenes and also not possible to use it as Thing in Openhab (my main use). Thing there is online, but links don`t work.
Have i something to add in button_maps.json too?

Smanar, 2022-12-21:
Ha no, sorry, deconz is a thing phoscon another.
But on third app like openHAB it need to work

And yes, need to edit too button_maps.json, you have the complete PR here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6244

MacroScooter, 2022-12-21:
Ok, seems tricky...
Used your DDF, adapted button_maps.json (as is was, with 4 buttons?)
Workaround with Openhanb deconz-binding -> switch is online, last seen, batterylevel, low battery are WORKING
BUT: the 3 switches and last timestamp NOT

Smanar, 2022-12-21:
>Ok, seems tricky...

Yeah but you don't need all the PR, the usefull part is this one

```
        "Tuya3gangMap": {
            "vendor": "Tuya",
            "doc": "3-gang remote",
            "modelids": ["_TZ3000_ee8nrt2l", "_TZ3000_ygvf9xzp", "_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0", "_TZ3000_rco1yzb1", "_TZ3000_gbm10jnj", "_TZ3000_dziaict4", "_TZ3000_famkxci2", "_TZ3000_itb0omhv", "_TZ3000_tzvbimpq", "_TZ3000_sj7jbgks", "_TZ3000_mh9px7cq"],
```

Perhaps a deconz restart ? the utton_maps.json is only loaded at start, and yes it s the same buton map for all tuya switches, 1 to 4 gangs.

It need to update state/buttonevent and state/lastupdated when pressing a button, have you access to deconbz logs ?

MacroScooter, 2022-12-22:
Yess, here we go, log after pressing button 1 (hope not long or short enough)
Node named "Dreifachschalter 1"
The "null" ia also have in Openhab

`07:36:04:814 APS-DATA.indication srcAddr: 0x0333, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -39
07:36:04:814 	asdu: 01fdfd00
07:36:04:814 APS-DATA.indication from child 0x0333
07:36:04:815 [INFO] - No button map for: TS0043, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: LIDL (0xFD), payload: 00, zclSeq: 253
07:36:04:816 Websocket 192.168.0.12:45888 send message: {"attr":{"id":"21","lastannounced":null,"lastseen":"2022-12-22T06:36Z","manufacturername":"_TZ3000_sj7jbgks","modelid":"TS0043","name":"Dreifachschalter 1","swversion":"1.0.3","type":"ZHASwitch","uniqueid":"a4:c1:38:c4:b1:85:1e:d1-01-0006"},"e":"changed","id":"21","r":"sensors","t":"event","uniqueid":"a4:c1:38:c4:b1:85:1e:d1-01-0006"} (ret = -1097456176)
07:36:04:817 APS-DATA.request id: 47, addrmode: 0x02, addr: 0x0333, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 3 len: 5 tx.options 0x04
07:36:04:817 	asdu (length: 5): 18fd0bfd00
07:36:04:915 APS-DATA.request id: 48, addrmode: 0x03, addr: 0xa4c138c4b1851ed1, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 4 len: 5 tx.options 0x00
07:36:04:915 	asdu (length: 5): 1025000100
07:36:04:915 APS-DATA.request id: 49, addrmode: 0x03, addr: 0xa4c138c4b1851ed1, profile: 0x0000, cluster: 0x0033, ep: 0x00 -> 0x00 queue: 5 len: 2 tx.options 0x04
07:36:04:915 	asdu (length: 2): 0100
07:36:04:931 APS-DATA.confirm id: 47, status: 0x00 SUCCESS
07:36:04:931 APS-DATA.confirm request id: 47 -> erase from queue
07:36:04:965 aps request id: 47 finished, erase from queue
07:36:05:017 APS-DATA.confirm id: 48, status: 0x00 SUCCESS
07:36:05:017 APS-DATA.confirm request id: 48 -> erase from queue
07:36:05:046 aps request id: 48 finished, erase from queue
07:36:05:128 APS-DATA.indication srcAddr: 0x0333, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 255, rssi: -35
07:36:05:128 	asdu: 1825010100002043
07:36:05:128 APS-DATA.indication from child 0x0333
07:36:05:129 APS-DATA.request id: 51, addrmode: 0x03, addr: 0xa4c138c4b1851ed1, profile: 0x0104, cluster: 0x0001, ep: 0x01 -> 0x01 queue: 4 len: 5 tx.options 0x00
07:36:05:129 	asdu (length: 5): 1026002100
07:36:05:130 Websocket 192.168.0.12:45888 send message: {"attr":{"id":"21","lastannounced":null,"lastseen":"2022-12-22T06:36Z","manufacturername":"_TZ3000_sj7jbgks","modelid":"TS0043","name":"Dreifachschalter 1","swversion":"1.0.3","type":"ZHASwitch","uniqueid":"a4:c1:38:c4:b1:85:1e:d1-01-0006"},"e":"changed","id":"21","r":"sensors","t":"event","uniqueid":"a4:c1:38:c4:b1:85:1e:d1-01-0006"} (ret = -1097456176)
07:36:05:153 APS-DATA.confirm id: 49, status: 0x00 SUCCESS
07:36:05:153 APS-DATA.confirm request id: 49 -> confirmed, timeout 0
07:36:05:249 APS-DATA.indication srcAddr: 0x0333, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8033, lqi: 255, rssi: -37
07:36:05:249 	asdu: 0100050002d11e85b1c438c1a4010000030a5d09ffff2e210001d11e85b1c438c1a4010100030a5d09ffff2e210001
07:36:05:249 APS-DATA.indication request id: 49 -> finished
07:36:05:249 APS-DATA.indication from child 0x0333
07:36:05:249 APS-DATA.request id: 49 erase from queue
07:36:05:249 APS-DATA.request id: 53, addrmode: 0x03, addr: 0xa4c138c4b1851ed1, profile: 0x0000, cluster: 0x0033, ep: 0x00 -> 0x00 queue: 4 len: 2 tx.options 0x04
07:36:05:249 	asdu (length: 2): 0202
07:36:05:267 APS-DATA.confirm id: 51, status: 0x00 SUCCESS
07:36:05:268 APS-DATA.confirm request id: 51 -> erase from queue
07:36:05:286 aps request id: 51 finished, erase from queue
07:36:05:403 APS-DATA.indication srcAddr: 0x0333, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0001, lqi: 255, rssi: -37
07:36:05:403 	asdu: 18260121000020c8
07:36:05:403 APS-DATA.indication from child 0x0333
07:36:05:404 Websocket 192.168.0.12:45888 send message: {"config":{"battery":100,"on":true,"reachable":true},"e":"changed","id":"21","r":"sensors","t":"event","uniqueid":"a4:c1:38:c4:b1:85:1e:d1-01-0006"} (ret = -1097456176)
07:36:05:408 APS-DATA.confirm id: 53, status: 0x00 SUCCESS
07:36:05:408 APS-DATA.confirm request id: 53 -> confirmed, timeout 0
07:36:05:495 APS-DATA.indication srcAddr: 0x0333, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8033, lqi: 255, rssi: -37
07:36:05:495 	asdu: 0200050202d11e85b1c438c1a4010600030a5d09ffff2e210001d11e85b1c438c1a4020600030a5d09ffff2e210001
07:36:05:495 APS-DATA.indication request id: 53 -> finished
07:36:05:495 APS-DATA.indication from child 0x0333
07:36:05:495 APS-DATA.request id: 53 erase from queue
07:36:05:495 APS-DATA.request id: 56, addrmode: 0x03, addr: 0xa4c138c4b1851ed1, profile: 0x0000, cluster: 0x0033, ep: 0x00 -> 0x00 queue: 3 len: 2 tx.options 0x04
07:36:05:495 	asdu (length: 2): 0304
07:36:06:304 APS-DATA.confirm id: 56, status: 0x00 SUCCESS
07:36:06:304 APS-DATA.confi`

MacroScooter, 2022-12-22:
Ooooops, sorry - My mistake!

I thougt because it is "unofficial" that I have to make a whole new button in button_maps.json at the end, so it was twice......
Mine wasn`t read any more, because it was already defined.

I added the manfacturer at the first button and everything now is working.

Thank you for your time!

machgy, 2022-12-22:
> > Ok, seems tricky...
> 
> Yeah but you don't need all the PR, the usefull part is this one
> 
> ```
>         "Tuya3gangMap": {
>             "vendor": "Tuya",
>             "doc": "3-gang remote",
>             "modelids": ["_TZ3000_ee8nrt2l", "_TZ3000_ygvf9xzp", "_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0", "_TZ3000_rco1yzb1", "_TZ3000_gbm10jnj", "_TZ3000_dziaict4", "_TZ3000_famkxci2", "_TZ3000_itb0omhv", "_TZ3000_tzvbimpq", "_TZ3000_sj7jbgks", "_TZ3000_mh9px7cq"],
> ```
> 
> Perhaps a deconz restart ? the button_maps.json is only loaded at start, and yes it s the same buton map for all tuya switches, 1 to 4 gangs.
> 
> It need to update state/buttonevent and state/lastupdated when pressing a button, have you access to deconbz logs ?

Dear Smanar,

Do you know the date of the update release of hassio DeCONZ addon? Would be the button map part of it? I have few remotes and I'd like to use them:)

Thanks and Best Regards, Gyula.

Smanar, 2022-12-22:
I don't know what is doing HA, but yes from my memory it's possible adding DDF in the deconz HA plugin, but not make change on the button_maps.json.

There is critic issue on the 2.19.X version, so it's the priority ATM, I think after we will have a merge for all small PR, them the bigger.

closed 2023-01-09 by manup