dresden-elektronik/deconz-rest-plugin #6557: Lights officially unavailable but can still receive commands

airbone42, 2022-11-21:
## Describe the bug
I'm using deconz with a Conbee II stick in a docker container using a raspi 4 since 2.5 years now. The system has grown to 72 nodes. Unfortunately, since around 4 weeks I have the issue that different lights (various brands and types) are marked as unavailable in deconz and homeassistant and usually are available again after either switching them off and on or just waiting several minutes to hours. Interstingly when sending a command to one off these lights e.g. to turn on they still receive the command and work properly.

Unfortunately, the error message in the logs for this behavior is pretty standard and I'm not sure how to handle this.

## Steps to reproduce the behavior
I can't reproduce it, it seems to be randomly multiple times a day on different lights

## Expected behavior
The lights should still be available.

## Screenshots
n/a

## Environment
 - Host system: Raspberry Pi
 - Running method: Marthoc Docker container
 - Firmware version: 26780700
 - deCONZ version: 2.19.01 / 19.9.2022
 - Device: ConBee II
 - Do you use an USB extension cable:yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? Yes a SSD hard drive, but also connected with another extension cable as this preivously lead to issues with the Zigbee connectivity. And a Z-Wave dongle, also with another extension cable.

## deCONZ Logs
`
08:03:01:796 0x0017880106FD923C error APSDE-DATA.confirm: 0xE9 on task 
08:03:01:797 max transmit errors for node 0x0017880106FD923C, last seen by neighbors 47 s
`

## Additional context
n/a

Mimiix, 2022-11-21:
Can you share some more logs?

I am temped to close this one as it is not as described in #5113, but it might has to do with something new i keep seeing with users. Logs can tell me if that's the case.

We need APS , APS L2, Error, Error L2, info, Info L2.

airbone42, 2022-11-21:
Sure, also happy to share the whole log, but it's pretty big ;)

There's nothing when grepping for Error, or Info, but these APS repeating:

`
08:00:19:907 0x0017880108317879 error APSDE-DATA.confirm: 0xE9 on task
08:00:33:300 0x0017880106FD923C error APSDE-DATA.confirm: 0xE9 on task
08:00:35:326 0x00178801082F075D error APSDE-DATA.confirm: 0xE9 on task
08:00:45:716 0x0017880108317879 error APSDE-DATA.confirm: 0xE9 on task
08:00:56:822 0x0017880106FD923C error APSDE-DATA.confirm: 0xE9 on task
08:00:58:808 0x00178801082F075D error APSDE-DATA.confirm: 0xE9 on task
08:01:09:005 0x0017880108317879 error APSDE-DATA.confirm: 0xE9 on task
08:01:23:140 0x0017880108317879 error APSDE-DATA.confirm: 0xE9 on task
08:01:31:399 0x0017880106FD923C error APSDE-DATA.confirm: 0xE9 on task
08:01:33:320 0x00178801082F075D error APSDE-DATA.confirm: 0xE9 on task
08:01:46:397 0x0017880108317879 error APSDE-DATA.confirm: 0xE9 on task
08:01:59:514 0x0017880108317879 error APSDE-DATA.confirm: 0xE9 on task
08:02:09:799 0x0017880106FD923C error APSDE-DATA.confirm: 0xE9 on task
08:02:11:797 0x00178801082F075D error APSDE-DATA.confirm: 0xE9 on task
08:02:23:402 0x0017880108317879 error APSDE-DATA.confirm: 0xE9 on task
08:02:33:314 0x0017880106FD923C error APSDE-DATA.confirm: 0xE9 on task
08:02:35:351 0x00178801082F075D error APSDE-DATA.confirm: 0xE9 on task
08:02:46:509 0x0017880108317879 error APSDE-DATA.confirm: 0xE9 on task
08:03:01:796 0x0017880106FD923C error APSDE-DATA.confirm: 0xE9 on task
08:03:03:812 0x00178801082F075D error APSDE-DATA.confirm: 0xE9 on task
08:03:11:997 0x0017880108317879 error APSDE-DATA.confirm: 0xE9 on task
08:03:25:126 0x0017880106FD923C error APSDE-DATA.confirm: 0xE9 on task
08:03:26:410 0x00178801082F075D error APSDE-DATA.confirm: 0xE9 on task
08:03:30:418 0x0017880108317879 error APSDE-DATA.confirm: 0xE9 on task
08:03:47:796 0x0017880106FD923C error APSDE-DATA.confirm: 0xE9 on task
08:03:50:908 0x0017880108317879 error APSDE-DATA.confirm: 0xE9 on task
08:03:51:707 0x00178801082F075D error APSDE-DATA.confirm: 0xE9 on task
08:04:15:811 0x0017880106FD923C error APSDE-DATA.confirm: 0xE9 on task
08:04:17:725 0x00178801082F075D error APSDE-DATA.confirm: 0xE9 on task
08:04:38:931 0x0017880106FD923C error APSDE-DATA.confirm: 0xE9 on task
08:04:40:968 0x00178801082F075D error APSDE-DATA.confirm: 0xE9 on task
08:26:38:751 0x0017880108A13E31 error APSDE-DATA.confirm: 0xE9 on task
08:49:19:028 0x00178801082F05A2 error APSDE-DATA.confirm: 0xE9 on task
`

Mimiix, 2022-11-21:
I really want to have the full logs, not just a part.

airbone42, 2022-11-21:
Sure, here you go: https://drive.google.com/file/d/1qQYwsZ2wkzcPvpwrvCmPc-Cj2M2VufrB/view?usp=share_link

airbone42, 2022-11-21:
Done, will give you a new log file as soon as the error appeared again.

airbone42, 2022-11-23:
So it happened again, this time with device 0x0017880108a19226. At the end it was unavailable but still reacted to on/off commands.

Logs at https://drive.google.com/file/d/1w4ziCuaCAjtlgazmng_yuKMa2RKwOONt/view?usp=share_link

beginnersguide, 2022-12-07:
I have the same issue with one lamp. The lamp greyed out and unavailable but still react to on/off by my FoH Switch SR-ZG2833PAC-FOH
I have to power off and on the electricity of the lamp to manage it again. I will try to get some logs as soon as the error appeare again.

Mimiix, 2022-12-07:
Can you try the latest version 2.19.3?

beginnersguide, 2022-12-07:
I have updated to latest version and report after a few days

airbone42, 2022-12-29:
> Can you try the latest version 2.19.3?

I updated and the issue doesn't appear that often anymore (only 2-3 times a week), but it's still there.

edvjangabriel, 2022-12-30:
I have the same problem. Since some weeks and with 2.19.3 too some lights went offline and don´t react. After minutes to hours they return and fulfill then the requested action. I do not have logs now but try to fetch some next time. Just wanted to confirm that bug on my side too.

closed 2023-01-28 by github-actions[bot]