dresden-elektronik/deconz-rest-plugin #6561: Device name

Idaho947, 2022-11-21:
## Device

- Product name: Climax technology Vibration sensor SVGS-1ZBS
- Manufacturer: Climax technology
- Model identifier: SVGS_00.00.03.08TC
- Device type : Sensor

## Screenshots

### Basic
[image]

### Identify
[image]

### Alarms
[image]

Mimiix, 2022-11-21:
Can you please  add a screenshot of the total cluster list?

Idaho947, 2022-11-21:
Ah yes. Just this line is missing :
[image]

Idaho947, 2022-11-21:
Ah ok sorry.
[image]

Idaho947, 2022-12-05:
Hello, any news about this intégration ?

Smanar, 2022-12-06:
Can try  this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "ClimaxTechnology",
  "modelid": "SVGS_00.00.03.08TC",
  "vendor": "Tuya",
  "product": "Vibration sensor SVGS-1ZBS",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_VIBRATION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/vibration"
        }
      ]
    }
  ]
}
```

But I haven't similar device to compare, will be a test.

Idaho947, 2022-12-06:
In phoscon it's stay like this :
[image]

That in pullcontrol :
[image]

Smanar, 2022-12-06:
Ha ?
The poll cluster is not visible on your capture here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6561#issuecomment-1322230331
You have more cluster ATM ?

BTW you have nothing in logs with the flag "IAS" (on deconz/help/debug view) when you try to trigger a vibration ?

In the cluster 0x0500, attribute 0x0002 you can see something change ?

Perhaps you need the bind with the cluster 0x0500

```
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

Idaho947, 2022-12-07:
No I juste have the first notification. It stay at true.

SwoopX, 2022-12-08:
This device could eventually work with configured attribute reporting on the status 0x0002 (if the device supports it). I guess we have like a handful of devices around where that works.

Idaho947, 2022-12-08:
With the new ddf, the senser is not see by phoson.

Smanar, 2022-12-09:
```
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
      "report": [
        {
          "at": "0x0002",
          "dt": "0x19",
          "min": 300,
          "max": 3600,
        }
      ]
    }
```

Idaho947, 2022-12-10:
It doesn't work. It's stay at true more than five minutes

Smanar, 2022-12-10:
Not a surprise for me.
You have made a "hot relaod" or a new inclusion ? because the bind setting can take time on battery powered device.

You third app can't just handle notification.

As workaround I know some users have cheated with changing the type to ZHAPresence (because this kind of device have support for the "auto return" to false.

Smanar, 2022-12-10:
I don't think phoscon is able to manage alarm sensor ?

Idaho947, 2022-12-10:
In phoscon a see in "Vibration detected" in the sensor menu like my picture higher

Smanar, 2022-12-10:
Ha yes right, but you can do something with the detection ?
I never used phoscon for automation so idk if you can use this king of sensor for event.

closed 2023-01-09 by github-actions[bot]