dresden-elektronik/deconz-rest-plugin #6566: Aqara relay LLKZMK11LM (lumi.relay.c2acn01) ghost ON state events from second relay in Home Assistant after updating add-on

Bigtema, 2022-11-23:
## Describe the bug

Problem with device “Aqara Two-way control module Wireless Relay Controller” LLKZMK11LM Model lumi.relay.c2acn01
This device has 2 relays inside. Last summer, after updating deCONZ Add-on from version 6.8.0 [deCONZ 2.10.04 / 3/14/2021 ] to the fresh version on that moment, I begun receive ghost events from second relay. Initial state - first relay is constantly switched ON, and second relay is switched OFF, each 5 minutes, Home Assistant show me that second relay is switched ON, but in real it is OFF, after few seconds HA show me OFF state of second relay. When I check the state of second relay directly though deCONZ, HA immediately toggle to OFF state. Then I made recover deCONZ Add-on to 6.8.0 [deCONZ 2.10.04], after that all went good as should be.
Nowadays I made second attempt of upgrading deCONZ Add-on to version 6.17.0 [deCONZ 2.19.01] and catch the same situation.

I tried to test with adding of new relay, which was not in network before, but it didn’t help. New relay behave as old devices.

In first screenshot you can see 3 old relays and new one. Each even line show second relay of each device. On 2/3 of time line I made downgrade to previous deCONZ, and after that ghost events are gone.
At the bottom there are relay with name “testrel” – it is fresh device. 
The moment of ghost ON event coincides with event in deCONZ log, see that event on second screenshot. 
You can see, that second relay show OFF state, but Home Assistant receive it as ON.
The same issue was described here https://github.com/home-assistant/core/issues/53808

## Expected behavior
No ghost ON event in Home Assistant

## Screenshots
[image]
[image]

## Environment
- Host system: Raspbery pi3, Raspbian buster updated, 
- Running method: Home Assistant 2022.11.3 [Supervised installation in docker] Supervisor 2022.10.2, deCONZ Add-on 6.17.0
- deCONZ version: 2.19.01
- Firmware version: 26660700 
- Device: Conbee II

Mimiix, 2022-12-20:
A dev or someone from Dresden needs to pick it up.

Perhaps @SwoopX  or @Manup ?

akrabio, 2023-01-11:
I am experiencing the same issue. HA and deCONZ are updated to the latest version but I keep on getting ghost updates to state every few minutes for LLKZMK11LM relays

closed 2023-02-09 by github-actions[bot]