dresden-elektronik/deconz-rest-plugin #6571: UseeLink Powerbar SM-SO306-U

BabaIsYou, 2022-11-25:
## Device

- Product name: UseeLink Powerbar SM-SO306-U
[found here ](https://www.amazon.ca/-/fr/gp/product/B0B71Q1CM8/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&th=1)
- Manufacturer: _TZ3000_cfnprab5
- Model identifier: TS011F
- Device type : Please remove all unrelated device types. 
  - Light

[image]
[image]

### Basic
[image]

### Identify
[image]
[image]

### Alarms

### Device Temperature

### Groups
[image]
[image]
[image]

### Scenes
[image]
[image]
[image]
[image]
[image]
[image]
### On/Off
[image]

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
### On/Off cluster for EP 2 to 5
[image]

Here is autogenerated DDF : 
`{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_cfnprab5",
  "modelid": "TS011F",
  "product": "TS011F",
  "sleeper": false,
  "status": "Draft",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x03"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x05"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}`

This device is partially supported as I was waiting to get 5 differents on/off device (1 per 120V plug and the last for USB combined ports) ... and may be a sixth one for the whole powerbar as I read elsewhere.

In Deconz, Phoscon and Domoticz I got 5 on/off plugs but when I tried to put one in Off or On state, all the plugs are turning On or Off. Whatever individual plug I tried to manage.

Where can I locate from where this behavior come from ?

BabaIsYou, 2022-11-25:
Looks like that these type of devices use Tuya Cluster 0xE001 on each EP  for On/Off state.
Thus it's bit new for me and don't know if using this sort of thing in DDF coud help 
`        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"},
          "refresh.interval": 300
        },`
and if so, does the dpip has to be changed according to the EP or does it has to be always 1 ?

BabaIsYou, 2022-11-25:
Replying to myself ;-)
With this kind of DDF it seems to work and that I can address each individual plug 👍 

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_cfnprab5",
  "modelid": "TS011F",
  "product": "TS011F",
  "sleeper": false,
  "status": "Silver",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 84000,
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"},
          "refresh.interval": 5
[... 328 lines cut ...]
          "min": 1,
          "max": 300
        }
      ]
    }
  ]
}

`

But I also got some errors in log : 
`13:01:06:649 emit event/tick: 0xa4c1388ede2f99cc
13:01:06:660 SC tick --> StateCallFunction
13:01:06:667 writeTuyaData, dpid: 0x01, type: 0x10, expr: Item.val == 1 ? 1 : 0;
13:01:06:668 Tuya write expression: Item.val == 1 ? 1 : 0; --> 0
13:01:07:790 emit event/tick: 0xa4c1388ede2f99cc
13:01:07:808 SC tick --> StateCallFunction
13:01:07:812 writeTuyaData, dpid: 0x01, type: 0x10, expr: Item.val == 1 ? 1 : 0;
13:01:07:814 Tuya write expression: Item.val == 1 ? 1 : 0; --> 0
13:01:21:371 SC tick --> StateRead
13:01:21:373 SC state change failed: a4:c1:38:8e:de:2f:99:cc-04
13:01:21:374 SC tick --> StateRead
13:01:21:376 SC state change failed: a4:c1:38:8e:de:2f:99:cc-05
`
Even if state seems to be Ok  when reading json from REST-API. ?!

BabaIsYou, 2022-11-25:
Well, back home (I tried from remote connections) I was less happy … Logically, states seem to be update but physically, nothing happens! No power in any activated plug. Then the errors in logs are real errors and the commands do nothing. ;-/

BabaIsYou, 2022-11-30:
Weel, it seems that this device need to pass thru the "tuya magic sequence" to gain control to individual plug.
This kind of behavior which tends to artificially rebuild a "pseudo-closed world" on an open concept stinks... the powerstrip goes back to seller!

stemer77, 2022-12-23:
Still doesn't work as expected.

BabaIsYou, 2023-01-10:
> Still doesn't work as expected.

And won't natively until [that request](https://forum.phoscon.de/t/including-standard-tuya-magic-spell-sequence-during-joining/2916/8) will be fullfilled.

closed 2023-02-09 by github-actions[bot]

BabaIsYou, 2023-02-23:
Then my next try will certainly be a DDF like this

```
{
	"schema": "devcap1.schema.json",
	"manufacturername": "_TZ3000_cfnprab5",
	"modelid": "TS011F",
	"product": "TS011F",
	"sleeper": false,
	"status": "Silver",
	"subdevices": [{
			"type": "$TYPE_ON_OFF_PLUGIN_UNIT",
			"restapi": "/lights",
			"uuid": [
				"$address.ext",
				"0x01"
			],
			"items": [{
					"name": "attr/id"
				},
				{
					"name": "attr/lastannounced"
				},
				{
					"name": "attr/lastseen"
				},
				{
					"name": "attr/manufacturername"
				},
				{
					"name": "attr/modelid"
				},
				{
					"name": "attr/name"
				},
				{
					"name": "attr/swversion",
					"refresh.interval": 86400,
					"read": {
						"at": "0x0001",
						"cl": "0x0000",
						"ep": 1,
						"fn": "zcl"
					},
					"parse": {
						"at": "0x0001",
						"cl": "0x0000",
						"ep": 1,
						"fn": "zcl",
						"script": "tuya_swversion.js"
					}
				},
				{
					"name": "attr/type"
				},
				{
					"name": "attr/uniqueid"
				},
				{
					"name": "state/alert",
					"default": "none"
				},
				{
					"name": "state/on",
					"parse": {
						"fn": "tuya",
						"dpid": 1,
						"eval": "Item.val = Attr.val;"
					},
					"write": {
[... 375 lines cut ...]
		{
			"bind": "unicast",
			"src.ep": 4,
			"cl": "0x0006",
			"report": [{
				"at": "0x0000",
				"dt": "0x10",
				"min": 1,
				"max": 300
			}]
		},
		{
			"bind": "unicast",
			"src.ep": 5,
			"cl": "0x0006",
			"report": [{
				"at": "0x0000",
				"dt": "0x10",
				"min": 1,
				"max": 300
			}]
		}
	]
}
```

BabaIsYou, 2023-02-26:
Well, seems to work ... as we can see in log 
```
07:56:56:221 70:b3:d5:2b:60:01:75:89/attr/manufacturername expression: Item.val = Attr.val --> _TZ3000_cfnprab5
07:56:56:291 70:b3:d5:2b:60:01:75:89/attr/modelid expression: Item.val = Attr.val --> TS011F
07:56:56:371 sub-device: 70:b3:d5:2b:60:01:75:89-01, has item: attr/id
07:56:56:372 sub-device: 70:b3:d5:2b:60:01:75:89-01, has item: attr/lastannounced
07:56:56:372 sub-device: 70:b3:d5:2b:60:01:75:89-01, has item: attr/lastseen
07:56:56:372 sub-device: 70:b3:d5:2b:60:01:75:89-01, has item: attr/manufacturername
07:56:56:373 sub-device: 70:b3:d5:2b:60:01:75:89-01, has item: attr/modelid
07:56:56:373 sub-device: 70:b3:d5:2b:60:01:75:89-01, has item: attr/name
07:56:56:373 sub-device: 70:b3:d5:2b:60:01:75:89-01, has item: attr/swversion
07:56:56:373 sub-device: 70:b3:d5:2b:60:01:75:89-01, has item: attr/type
07:56:56:374 sub-device: 70:b3:d5:2b:60:01:75:89-01, has item: attr/uniqueid
07:56:56:374 sub-device: 70:b3:d5:2b:60:01:75:89-01, create item: config/configured
07:56:56:374 sub-device: 70:b3:d5:2b:60:01:75:89-01, create item: state/alert
07:56:56:375 sub-device: 70:b3:d5:2b:60:01:75:89-01, create item: state/on
07:56:56:375 sub-device: 70:b3:d5:2b:60:01:75:89-01, has item: state/reachable
07:56:56:461 sub-device: 70:b3:d5:2b:60:01:75:89-02, has item: attr/id
07:56:56:462 sub-device: 70:b3:d5:2b:60:01:75:89-02, has item: attr/lastannounced
07:56:56:462 sub-device: 70:b3:d5:2b:60:01:75:89-02, has item: attr/lastseen
07:56:56:462 sub-device: 70:b3:d5:2b:60:01:75:89-02, has item: attr/manufacturername
07:56:56:463 sub-device: 70:b3:d5:2b:60:01:75:89-02, has item: attr/modelid
07:56:56:463 sub-device: 70:b3:d5:2b:60:01:75:89-02, has item: attr/name
07:56:56:463 sub-device: 70:b3:d5:2b:60:01:75:89-02, has item: attr/swversion
07:56:56:463 sub-device: 70:b3:d5:2b:60:01:75:89-02, has item: attr/type
07:56:56:464 sub-device: 70:b3:d5:2b:60:01:75:89-02, has item: attr/uniqueid
07:56:56:464 sub-device: 70:b3:d5:2b:60:01:75:89-02, create item: state/alert
07:56:56:464 sub-device: 70:b3:d5:2b:60:01:75:89-02, create item: state/on
07:56:56:465 sub-device: 70:b3:d5:2b:60:01:75:89-02, has item: state/reachable
07:56:56:550 sub-device: 70:b3:d5:2b:60:01:75:89-03, has item: attr/id
07:56:56:550 sub-device: 70:b3:d5:2b:60:01:75:89-03, has item: attr/lastannounced
07:56:56:551 sub-device: 70:b3:d5:2b:60:01:75:89-03, has item: attr/lastseen
07:56:56:551 sub-device: 70:b3:d5:2b:60:01:75:89-03, has item: attr/manufacturername
07:56:56:551 sub-device: 70:b3:d5:2b:60:01:75:89-03, has item: attr/modelid
07:56:56:552 sub-device: 70:b3:d5:2b:60:01:75:89-03, has item: attr/name
07:56:56:552 sub-device: 70:b3:d5:2b:60:01:75:89-03, has item: attr/swversion
07:56:56:552 sub-device: 70:b3:d5:2b:60:01:75:89-03, has item: attr/type
07:56:56:552 sub-device: 70:b3:d5:2b:60:01:75:89-03, has item: attr/uniqueid
07:56:56:553 sub-device: 70:b3:d5:2b:60:01:75:89-03, create item: state/alert
07:56:56:553 sub-device: 70:b3:d5:2b:60:01:75:89-03, create item: state/on
07:56:56:553 sub-device: 70:b3:d5:2b:60:01:75:89-03, has item: state/reachable
07:56:56:627 sub-device: 70:b3:d5:2b:60:01:75:89-04, has item: attr/id
07:56:56:628 sub-device: 70:b3:d5:2b:60:01:75:89-04, has item: attr/lastannounced
07:56:56:628 sub-device: 70:b3:d5:2b:60:01:75:89-04, has item: attr/lastseen
07:56:56:628 sub-device: 70:b3:d5:2b:60:01:75:89-04, has item: attr/manufacturername
07:56:56:629 sub-device: 70:b3:d5:2b:60:01:75:89-04, has item: attr/modelid
07:56:56:629 sub-device: 70:b3:d5:2b:60:01:75:89-04, has item: attr/name
07:56:56:629 sub-device: 70:b3:d5:2b:60:01:75:89-04, has item: attr/swversion
07:56:56:630 sub-device: 70:b3:d5:2b:60:01:75:89-04, has item: attr/type
07:56:56:630 sub-device: 70:b3:d5:2b:60:01:75:89-04, has item: attr/uniqueid
07:56:56:630 sub-device: 70:b3:d5:2b:60:01:75:89-04, create item: state/alert
07:56:56:630 sub-device: 70:b3:d5:2b:60:01:75:89-04, create item: state/on
07:56:56:631 sub-device: 70:b3:d5:2b:60:01:75:89-04, has item: state/reachable
07:56:56:708 sub-device: 70:b3:d5:2b:60:01:75:89-05, has item: attr/id
07:56:56:708 sub-device: 70:b3:d5:2b:60:01:75:89-05, has item: attr/lastannounced
07:56:56:709 sub-device: 70:b3:d5:2b:60:01:75:89-05, has item: attr/lastseen
07:56:56:709 sub-device: 70:b3:d5:2b:60:01:75:89-05, has item: attr/manufacturername
07:56:56:709 sub-device: 70:b3:d5:2b:60:01:75:89-05, has item: attr/modelid
07:56:56:710 sub-device: 70:b3:d5:2b:60:01:75:89-05, has item: attr/name
07:56:56:710 sub-device: 70:b3:d5:2b:60:01:75:89-05, has item: attr/swversion
07:56:56:710 sub-device: 70:b3:d5:2b:60:01:75:89-05, has item: attr/type
07:56:56:710 sub-device: 70:b3:d5:2b:60:01:75:89-05, has item: attr/uniqueid
07:56:56:711 sub-device: 70:b3:d5:2b:60:01:75:89-05, create item: state/alert
07:56:56:711 sub-device: 70:b3:d5:2b:60:01:75:89-05, create item: state/on
07:56:56:711 sub-device: 70:b3:d5:2b:60:01:75:89-05, has item: state/reachable
07:56:56:712 DEV found DDF for 0x70B3D52B60017589, path: /home/pi/.local/share/dresden-elektronik/deCONZ/devices/_TZ3000_cfnprab5_Powerbar.json
...
07:56:57:202 70:b3:d5:2b:60:01:75:89-01/attr/swversion expression: Item.val = Attr.val --> 202+TZZSW25SL102
07:56:57:233 70:b3:d5:2b:60:01:75:89-02/attr/swversion expression: Item.val = Attr.val --> 202+TZZSW25SL102
07:56:57:267 70:b3:d5:2b:60:01:75:89-03/attr/swversion expression: Item.val = Attr.val --> 202+TZZSW25SL102
[... cut ...]
07:56:57:771 0x70B3D52B60017589 missing cluster 0xEF00, frame control 0x00000018
07:56:57:779 DB UPDATE device_descriptors SET data = x'010401000101070000060003000400050001e000ef00', timestamp = 1677416217 WHERE device_id = (SELECT id FROM devices WHERE mac = '70:b3:d5:2b:60:01:75:89') AND endpoint = 1 AND type = 4
```

Just the `missing cluster 0xEF00` seems strange to me as my DDF doesn't make any reference to that cluster, and the cluster is present. But that works using GUI. But not thru REST-API, then my DFF is certainly bad !

BabaIsYou, 2023-02-26:
Well, @Smanar suggested to delete the tuya reference in `state` and it works ! Thx !
Here is the DDF : 
```
{
   "schema":"devcap1.schema.json",
   "manufacturername":"_TZ3000_cfnprab5",
   "modelid":"TS011F",
   "product":"TS011F",
   "sleeper":false,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_ON_OFF_PLUGIN_UNIT",
         "restapi":"/lights",
         "uuid":[
            "$address.ext",
            "0x01"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion",
               "refresh.interval":86400,
               "read":{
                  "at":"0x0001",
                  "cl":"0x0000",
                  "ep":1,
                  "fn":"zcl"
               },
               "parse":{
                  "at":"0x0001",
                  "cl":"0x0000",
                  "ep":1,
                  "fn":"zcl",
                  "script":"tuya_swversion.js"
               }
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"state/alert",
               "default":"none"
            },
            {
               "name":"state/on"
            },
            {
               "name":"state/reachable"
            },
[... 317 lines cut ...]
         "report":[
            {
               "at":"0x0000",
               "dt":"0x10",
               "min":1,
               "max":300
            }
         ]
      },
      {
         "bind":"unicast",
         "src.ep":5,
         "cl":"0x0006",
         "report":[
            {
               "at":"0x0000",
               "dt":"0x10",
               "min":1,
               "max":300
            }
         ]
      }
   ]
}
```

PR #6865 (2023-04-19): Update _TZ3000_TS011F_PowerStrip to include config/tuya_unlock
To use new attribute `config/tuya_unlock` instead of deturning `config/configured`
Will definitely close #6571 and fix #6225