dresden-elektronik/deconz-rest-plugin #6574: Namron Panel Heater

phuongpham, 2022-11-26:
I have Namron Panel Heater 1000W White. I can't at this device to phoscon.
Is this device not supported?

Product name: Namron Panel Heater (1000W White)
Manufacturer: Namron AS
Model identifier: 5401395
Device type : 
Thermostat
Switch

## Screenshots

[image]

### Basic

[image]

### Identify

[image]

### Groups

[image]

### Scenes

[image]

### Thermostat

[image]

### Thermostat user interface

[image]

### Simple Metering

[image]

### Electrical Measurement 
[image]

Smanar, 2022-11-27:
Hello can you try this DDF ?

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "NAMRON  AS",
  "modelid": "5401395",
  "vendor": "Namron",
  "product": "Panel Heater 1000W White",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0201",
          "0x0204"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "default": 0
        },
        {
          "name": "config/mode"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/locked",
          "refresh.interval": 3660,
          "parse": {
            "at": "0x0001",
            "cl": "0x0204",
            "ep": 1,
            "eval": "Item.val = Attr.val;",
            "fn": "zcl"
          }
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0B04"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power"
        },
        {
          "name": "state/voltage"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0702"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "default": 0
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "default": 0
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0204",
      "report": [
        {
          "at": "0x0001",
          "dt": "0x30",
          "min": 60,
          "max": 3600
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0201",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 30,
          "max": 360,
          "change": "0x0000000A"
        }
      ]
    }
  ]
}
```

And tell me what is missing ?

phuongpham, 2022-11-27:
@Smanar great!
I think this one is similar to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6053

Most of the thing I managed to control/view.

I'm not sure if it's supported. 
But I would like to be able to control/view the relay state.
Not sure if the ON/OFF cluster is the correct one.
But when I try this I get follow error: status 0xc3

[image]

Smanar, 2022-11-27:
>I think this one is similar to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6053

Ha yes right, have forget the previous one, but not exaclty same device, it's possible to improve them. For exemple I have tried to implement the "config/locked" on this one.

And how have you selected the on/off cluster ? I can't see it on the capture ?

From your documentation, for the "valve state" (relay state"on this device) it's probably the attribute 0x0029

```
HVAC relay state/ ThermostatRunningState
Indicates the relay on/off status, here only supports bit0( Heat State)
```
And it's  native, I have removed it from the DDF but you can just add

```
        {
          "name": "state/on"
        }
```

On the ZHAThermostat.

It can need bind/report, just make try as it, if it don't work, read the value in deconz, if read the value in deconz update the value in the API, it mean you need a bind/report.

phuongpham, 2022-11-27:
@Smanar i tried to add the DDF file but still the same.

In deconz I get this error/message

[image]

Smanar, 2022-11-28:
You have a "hot relaod" ? or have re-included the device ?
The device is still not reconised ?

I don't found the part you are using on the GUI, but this attribute is read only, you can't force the relay.
To force it you can force the device using "config/mode" for exemple, but not the relay.

phuongpham, 2022-11-28:
> You have made a "hot relaod" using the editor ? or have re-included the device after adding the DDF (need too restart deconz, long story but deconz scan new DDF only at start) ? The device is still not reconised ?
> 
> I don't found the part you are using on the GUI, but this attribute is read only, you can't force the relay. To force it you can force the device using "config/mode" for exemple, but not the relay.

I just changed the file and restart the deconz app

Smanar, 2022-11-29:
>I just changed the file and restart the deconz app

You can do that, but the next time ^^.
For the moment the device is not reconized, so not in the API yet, so you need to re-include it, it will open the API and the DDF to add it.

The next time, except ifwe need to reconfigure the device, restarting deconz will be enought.

If it work the node name will go from "0xXXXX" to a real name in the GUI.

BabaIsYou, 2022-11-29:
> Here is the documentation for the product https://www.elektroimportoren.no/docs/lib/5401395-Manual-18.pdf

In the documentation I could see that there is a **Display Auto Off** feature that may can be controlled using the same kind of tweak I used on Sinopé thermostat. Something like this in DDF (in the Thermostat part off the DDF).
Not quite sure of the data type because your screenshots doesn't show attribute 0x1001 in Thermostat Cluster, nor the default value but could be ajusted.
`       {
          "name": "config/ledindication",
          "description": "Use it for Display Auto Off configuration.\nfalse = Deactivated\ntrue = Activated",
          "refresh.interval": 360
          "read": {
            "at": "0x1001",
            "cl": "0x0201",
            "ep": 1,
            "fn": "zcl"
          },
          "write": {
            "at": "0x1001",
            "cl": "0x0201",
            "dt": "0x30",
            "ep": 1,
            "eval": "Item.val",
            "fn": "zcl"
          },
          "parse": {
            "at": "0x1001",
            "cl": "0x0201",
            "ep": 1,
            "eval": "Item.val = Attr.val;",
            "fn": "zcl"
          },
          "default": true
        },``

closed 2022-12-28 by github-actions[bot]

phuongpham, 2023-01-06:
> {
>           "name": "state/on"
>         }

@Smanar  Could you please provide me new DDF with this included? I haven't been working on DDF before so I'm not sure which line I should add     
```
 {
          "name": "state/on"
        }
```

Smanar, 2023-01-07:
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "NAMRON AS",
  "modelid": "5401395",
  "vendor": "Namron",
  "product": "Panel Heater 1000W White",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0201",
          "0x0204"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "default": 0
        },
        {
          "name": "config/mode"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/locked",
          "refresh.interval": 3660,
          "parse": {
            "at": "0x0001",
            "cl": "0x0204",
            "ep": 1,
            "eval": "Item.val = Attr.val;",
            "fn": "zcl"
          }
        },
        {
          "name": "state/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0B04"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power"
        },
        {
          "name": "state/voltage"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0702"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "default": 0
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "default": 0
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0204",
      "report": [
        {
          "at": "0x0001",
          "dt": "0x30",
          "min": 60,
          "max": 3600
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0201",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 30,
          "max": 360,
          "change": "0x0000000A"
        }
      ]
    }
  ]
}
```

phuongpham, 2023-01-07:
@Smanar Great!
I can see the status now the GUI.
I have integrate deconz with openHAB.
Currently I can't see that this channel is available in openHAB. 
Maybe the deconz binding also need to be updated to get this support?

[image]

Smanar, 2023-01-07:
The DDF have no impact on the GUI.
It have an impact only on the API, used by openHAB for exemple.

You need to restart deconz forthe field be added in the json.
You can check it direclty in the API using for exemple phoscon/help/api Information/sensor

If the field is visible in the API and not in third app, it's probably because the app have created the entry before the field was added. you can resynchronise it but the procedure depend of third app.

svinz, 2023-02-16:
Any update on this?

Smanar, 2023-02-16:
Something is missing ?

phuongpham, 2023-03-09:
@Smanar  when will this DDF be a part of the build?
I just noticed that the device stop working. The reason for this was the deconz was updated and the DDF/json file was removed. I had to recreate the file and the item was starting to work again.

Smanar, 2023-03-09:
Oups sorry, noone have asked for I make a PR so nothing was done.
PR just made here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6792

phuongpham, 2023-03-09:
> Oups sorry, noone have asked for I make a PR so nothing was done. PR just made here #6792

Great! Thanks for your help