dresden-elektronik/deconz-rest-plugin #6576: Sonoff Disconnect

Mephisto090, 2022-11-27:
## Describe the bug

hello,

I've had a lot to do lately with keeping the sensors "alive". I have to reconnect the sensors every day. I press on Phoscon on reconnect and press the button on the sensor. So deconz gets a sign of life and everything works normally.

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

PhosconApp:
[image]

Jarvis (iobroker)
[image]

## Environment

 - Host system: Windows-PC
 - Running method: Windows
 - Firmware version: 26780700
 - deCONZ version: 2.19.00 / 19.9.2022
 - Device: ConBee II
 - Do you use an USB extension cable: Yes (2m)
 - Is there any other USB or serial devices connected to the host system? If so: Which?: USB Hard Drive

## deCONZ Logs

No debug output because none of the devices are currently connected

## Additional context

Mimiix, 2022-11-27:
Hi

As this is not a issue as described in #5113 i am closing this issue.

Feel free to open a topic in the forums.

Kind regards,
Dennis

closed 2022-11-27 by Mimiix

lupa18, 2023-06-04:
@Mephisto090 did you get them working?

Mimiix, 2023-06-04:
> @Mephisto090 did you get them working? (same here). And I am editing some articles here:
> 
> * https://zigbee.blakadder.com/Sonoff_SNZB-01.html
> * will try to edit the others
> 
> and prevent users to get trapped with those devices.

This is not the place to ask.

I have sonoff that I'm happy with. So basing it on this guy with a issue is plain wrong(especially when his issue has nothing to do with the sonoff).