dresden-elektronik/deconz-rest-plugin #6582: Tuya TS0201 _tz3000_xr3htd96

zigomatichub, 2022-11-29:
## Device

- Product name: Temperature & Humidity Sensor
- Manufacturer: _tz3000_xr3htd96
- Model identifier: TS0201
- Device type : 
  - Sensor

## Screenshots

[image]
[image]
[image]

### Basic

### Identify
[image]

### Alarms

### Device Temperature
[image]

[image]
### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
2x AAA batteries
[image]

Smanar, 2022-11-29:
Hello this device is already in PR list https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6244

zigomatichub, 2022-11-29:
Ok sorry, I searched on _tz3000_xr3htd96 on issues without received any result. Next tume, I will search on both

closed 2022-11-29 by zigomatichub

Smanar, 2022-11-29:
Don't worry, realy need a PR merge, too much devices in waiting list ...