dresden-elektronik/deconz-rest-plugin #6584: Danfoss Ally Setpoint Command not working with DDF

luka6000, 2022-11-30:
## Describe the bug

My Danfoss Allys work kind of strange this heating season. I think this is connected with device DDF.
It takes about 3 hours for the valve to fully open
[image]

This issue was already resolved with this fix
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3927

But is it still valid fix with DDF device description?

## Steps to reproduce the behavior

Set heating to low value
wait for the vale to close
set heating to high value
the valve will open but very slowly

## Expected behavior

like with this fix https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3927

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26720700
 - deCONZ version: 2.19.1
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? ESP32

## deCONZ Logs

## Additional context

SwoopX, 2022-11-30:
This still works as before so there hasn't been any change in this regard. All my TRVs work as expected in this regard. I only observe something like this when they operate on schedules.

luka6000, 2022-11-30:
ok, I have schedules enabled but this setpoint was set manually. I’ll check with schedules disabled. Thanks for confirmation!

luka6000, 2022-11-30:
@SwoopX do you have 0x4016 Radiator Covered = true?

SwoopX, 2022-11-30:
No, it's false on all radiators here.

closed 2022-12-29 by github-actions[bot]

drvenabili, 2025-01-26:
@luka6000  Sorry for this necropost. How do you make sure that the TRVs remain with `radiator_covered = True`? Mine seem to reset to False whenever. 

I have tried to edit your "external sensor temp" blueprint to not only update the temperature but in the same call force radiator_covered = True (https://gist.github.com/drvenabili/bd40e4cfe058d7d97905c62d845e8afe), but it's not working -- no error, simply only the external temp is updated. I imagine it's because the PUT on `http://192.168.0.2/api/1BF6C2B089/sensors/<ally_thermostat_ID>/config/` with 
```JSON
{
    "radiatorcovered": true
} 
```
doesn't work ("parameter radiatorcovered not available").

I am bit lost as I can update other fields with the same command + I can do it via VNC, but not via the REST API. I would appreciate any help you might have.