dresden-elektronik/deconz-rest-plugin #6590: Deconz, RestApi and VNC hangs frequently for 30-60 seconds

LukasA83, 2022-12-04:
System:
Raspi 4b
Deconz docker 2.19.03, same issue with, at least, .02

Frequently the network lags/hangs being unresponsive and therewith not receiving updates. I  the VNC screen all nodes stop to blink blue for this period while the screen stays responsive. I enabled zhe logging and tried to copy a snippet:

`16:39:22:667 ZCL attribute report 0x00158D000362C576 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
16:39:26:937 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
16:39:26:938 Skip automatic channel change, TODO warn user
16:39:36:937 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
16:39:36:938 Skip automatic channel change, TODO warn user
16:39:47:602 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
16:39:47:603 Skip automatic channel change, TODO warn user
16:39:48:437 Skip idle timer callback, too early: elapsed 830 msec
16:39:56:937 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
16:39:56:937 Skip automatic channel change, TODO warn user
16:40:04:520 ZCL attribute report 0x00158D000342F3F7 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 `

Any idea?

Mimiix, 2022-12-04:
Hi

As this issue is not as described in #5113 i am closing this issue.

Feel free to open a topic on the forums 

Kind regards,
Dennis

closed 2022-12-04 by Mimiix

LukasA83, 2022-12-04:
It is clearly a bug because it was working before, so why closing it?

Smanar, 2022-12-04:
You haven't followed the procedure.

>16:39:26:937 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration

Take a look here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6515#issuecomment-1312781817

LukasA83, 2022-12-04:
I looked at it and corrected the panid. Still, same issue:
`17:31:56:697 ZCL attribute report 0x00158D000362C576 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
17:32:22:470 Skip idle timer callback, too early: elapsed 926 msec
17:32:24:436 Skip idle timer callback, too early: elapsed 873 msec
17:32:45:126 ZCL attribute report 0x00158D000362C576 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 `

ebaauw, 2022-12-04:
I think this is a duplicate of https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6588.

LukasA83, 2022-12-04:
Thanks for pointing this out, I have seen it. For me it is not involving any commands, but can confirm it works without any issue with 2.18.02.

Mimiix, 2022-12-04:
It's not a bug as the log indicates clearly there's a configuration issue as Smanar points out.

LukasA83, 2022-12-04:
This is already gone, I corrected it, issue remains.

Mimiix, 2022-12-04:
In that case it's a duplicate of the issue Erik mentioned.

I am not reopening this one in any case as the template has been ignored . Please use the forums or post in the issue Erik mentions.

Locking this one up as there's no point in continuing here.

Mimiix, 2022-12-04:
Personal advice: It really is in your best entrance to post on the forums and get help there. The GitHub is simply for bugs and I believe this might be some configuration issue.