dresden-elektronik/deconz-rest-plugin #6591: Incorrect light type for LEDVANCE A19 RGBW

Thomas-Vos, 2022-12-04:
## Describe the bug

A user reported that their LEDVANCE A19 RGBW has the wrong light type. It is reported as a `Color temperature light` but it also supports setting a color. So it should be `Extended color light` instead.

https://community.hueessentials.com/t/ledvance-a19-rgbw-bulb-no-color-controls/4579

## Steps to reproduce the behavior
N/A

## Expected behavior
Extended color light as light type

## Screenshots
[image]
https://community.hueessentials.com/t/ledvance-a19-rgbw-bulb-no-color-controls/4579

## Environment
N/A

## deCONZ Logs
N/A

## Additional context

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3721
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3365

Mimiix, 2022-12-04:
Hi Thomas

Can you (or ask the user to) provide the rest api response ?

Kind regards,

Thomas-Vos, 2022-12-04:
@Mimiix already discussed the issue with the devs on Discord, and I asked the user to provide the DDF so the bug can be fixed. It seems deCONZ is overriding the light type incorrectly here: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/5d136aed7d2e79df663ab3f907954b6f3e6662bf/de_web_plugin.cpp#L3260

I will make a PR to fix this code, and once the user reports the DDF it can be fixed in a DDF as well.

Smanar, 2022-12-04:
Have found capture here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3365

Thomas-Vos, 2022-12-04:
Created a PR to fix the legacy code here: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6592

Ltek, 2022-12-04:
DDF...

{
  "schema": "devcap1.schema.json",
  "manufacturername": "LEDVANCE",
  "modelid": "A19 RGBW",
  "product": "A19 RGBW",
  "sleeper": false,
  "status": "Draft",
  "subdevices": [
    {
      "type": "$TYPE_COLOR_TEMPERATURE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/colorcapabilities",
          "description": "The supported color modes as bitmap.",
          "refresh.interval": 84000
        },
        {
          "name": "config/ctmax",
          "description": "Maximum supported color temperature.",
          "refresh.interval": 84000
        },
        {
          "name": "config/ctmin",
          "description": "Minimum supported color temperature.",
          "refresh.interval": 84000
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/bri",
          "description": "The current brightness.",
          "refresh.interval": 5
        },
        {
          "name": "state/colormode",
          "description": "The currently active color mode.",
          "refresh.interval": 300
        },
        {
          "name": "state/ct",
          "description": "The current Mired color temperature of the light. Where Mired is 1000000 / color temperature (in kelvins).",
          "refresh.interval": 5
        },
        {
          "name": "state/effect",
          "description": "The currently active effect.",
          "default": "none"
        },
        {
          "name": "state/hue",
          "description": "The current color hue."
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        },
        {
          "name": "state/sat",
          "description": "The current color saturation.",
          "refresh.interval": 5
        },
        {
          "name": "state/x",
          "description": "The current color x coordinate.",
          "refresh.interval": 5
        },
        {
          "name": "state/y",
          "description": "The current color y coordinate."
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0008",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0300",
      "report": [
        {
          "at": "0x0007",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0003",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        },
        {
          "at": "0x0004",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        },
        {
          "at": "0x0008",
          "dt": "0x30",
          "min": 1,
          "max": 300
        }
      ]
    }
  ]
}

Thomas-Vos, 2022-12-04:
Thanks a lot for the DDF!

The unknown in Hue Essentials is unrelated to this issue and will be fixed in the next version, it's just a missing icon.

manup, 2023-04-01:
I'd propose to add @Smanar DDF. Currently we don't have any LEDVANCE lights available as DDF but a bunch of C++ workarounds predating DDF. So it would be cool to transition these lights to DDF over time and remove the C++ parts after that.

ebaauw, 2023-04-01:
Can we please, please create new DDFs first-time-right?
- `config/colorcapabilities` of `16` is wrong.  If the legacy code is correct, it should be `0x19`.  Are we sure this light doesn't support enhanced hue/saturation nor color loop?  Probably, as these are pre-Zigbee-3 ZHA lights, without the ZLL extensions (hence the wrong type, as _Enhanced Color Light_ is a ZLL extension).
- `config/colorcapabilities`, `config/ctmax` and `config/ctmin` have been deprecated - please use `cap/color/capabilities`, `cap/color/ct/max` and `cap/color/ct/min` instead.
- Please add the `cap/color/xy` capabilities.  I'm not sure if LEDVANCE exposes these, nor whether they use the _ColorPoint_ ZIgbee attribute set (like Hue), or the _Primary_ set (like IKEA).  If not, hard-code them (like MLI).  According to my notes, LEDVANCE use: `{  r: [0.6972, 0.3027], g: [0.1737, 0.6991], b: [0.1227, 0.0959] }`.  Multiply these API values by 0xFFFF to get the Zigbee values.
- Please double-check whether they support _Off with Effect_, _Move to Level (with On/Off)_, _Trigger Effect_, and, if so, include the corresponding `cap` items.
- Please double check whether they update the _Current X_ and _Current Y_ values when setting color temperature, and, if so, include `cap/color/ct/computes_xy`.
- Please double check whether they support _Execute_If_Off_ and if so, include the corresponding `config` items.
- I don't suppose they support the _Startup_ attributes, as OSRAM uses a specific command to set the startup state.  Not sure if we ever exposed that through the API.

For examples see
- Hue: `devices/philips/light_zb3_C_gradient.json`
- MLI: `devices/mli/zbt-extendedcolor.json`
- IKEA: `devices/ikea/cws_light.json`