dresden-elektronik/deconz-rest-plugin #6594: Nous Smart ZigBee Socket A1Z (_TZ3000_2putqrmw)

nWRqenkF, 2022-12-05:
## Device

- Product name: Nous Smart ZigBee Socket A1Z
- Manufacturer: _TZ3000_2putqrmw
- Model identifier: TS011F
- Device type : 
  - Light
  - Sensor

## Screenshots
[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]

### Scenes
[image]
[image]
[image]
[image]

### On/Off
[image]
[image]

### Simple Metering
[image]
[image]

### Other clusters that are not mentioned above
### Electrical Measurement
[image]
[image]

### OTAU
[image]

nWRqenkF, 2022-12-05:
I'm aware of the issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5917 but the alphanumeric part of the manufacturer part is different (here "_TZ3000_2putqrmw" vs "_TZ3000_ksw8qtmt" in issue 5917).

Anyhow: the device is recognized in phoscon, but in openhab3 via deconz-REST-API only the On/Off-Channel is available, the electric measurement is missing.

nWRqenkF, 2022-12-05:
Current DDF from deCONZ application:

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_2putqrmw",
  "modelid": "TS011F",
  "product": "TS011F",
  "sleeper": false,
  "status": "Draft",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

Smanar, 2022-12-06:
Can test if it's just a clone.
You already have the DDF for the Nous A1z here https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_ksw8qtmt_plug_nous_a1z.json

You just need to edit 

```
  "manufacturername": "_TZ3000_2putqrmw",
  "modelid": "TS011F",
```

and replace them with 

```
  "manufacturername": ["_TZ3000_2putqrmw", "_TZ3000_2putqrmw"]
  "modelid": ["TS011F", "TS011F"],
```

Them you can restart deconz and re-include your device.

nWRqenkF, 2022-12-07:
Thanks worked for me!

Minor correction (missing comma in 1st line):
````
"manufacturername": ["_TZ3000_2putqrmw", "_TZ3000_2putqrmw"],
"modelid": ["TS011F", "TS011F"],
````

Smanar, 2022-12-07:
It's the deconz working mode ^^.
Look at the uniqueID, it's composed by the MAC adress - the endpoint - The cluster.

Deconz make a part for "light" + a part for "sensor"
And you have 1 sensor by used cluster.

So it depend of device, some device have consumption and power on the same cluster, not this one.

SvenMartin81, 2022-12-14:
Hi, i have the Same device with the Same situation. Only the Switch is there.

Smanar, 2022-12-14:
The switch ? Not a plug ?

You have too  the _TZ3000_2putqrmw ? You can use this DDF https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6589

SvenMartin81, 2022-12-15:
Hi,

Yes i will Check this. Can i Upload from Deconz with any doubt? Or Simi Need my Special Write access on and Folders?

Mit freundlichen Grüßen

Sven Martin
Mühlenring 63
23923 Selmsdorf

Tel.: +49 (0)172 5180261

SvenMartin81, 2022-12-15:
Do you know when this will be integrales in the next Deconz Update?

Or can I add a new file in the Tuya Finder with These changes and the System will recognice this.

In the actuall Deconz-2.19.03-qt5.deb was this Not changed.

Please, leg ne know.

Mit freundlichen Grüßen

Sven Martin
Mühlenring 63
23923 Selmsdorf

Tel.: +49 (0)172 5180261

SvenMartin81, 2022-12-15:
> What is your OS ?
>
> On normal one
>
> All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.
>
> On HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839
>
> I hope we will have the file officialy in next version.

Hi,
I am Running raspi on a Pi4B with Raspbee on it.
But i cannot overwrite the files because I have No write Access with the User „Pi“.
What can i do?

Mit freundlichen Grüßen

Sven Martin
Mühlenring 63
23923 Selmsdorf

Tel.: +49 (0)172 5180261

SvenMartin81, 2022-12-16:
Hi,

I got it but the only missing Sensor in Home Assistant is the voltage Sensor. The „power“ and „consumption“ Sensor is Running fine.

What can i so to have the „voltage“ Sensor?

Mit freundlichen Grüßen

Sven Martin
Mühlenring 63
23923 Selmsdorf

Tel.: +49 (0)172 5180261

Mimiix, 2022-12-16:
@SvenMartin81  Voltage is a attribute of the Power entity in HA. You need to create a template sensor for taht to become a seperate sensor.

Alex-joomla, 2022-12-17:
I' ve managed to edit the ddf and so I'm able to display power and consumption. Consumption is running fine but power is displayed as 0 as in deconz. But power is a calculated value of current and voltage, should it been calculated by the device or by deconz?

[image]
[image]

Alex-joomla, 2022-12-17:
> The „power“ and „consumption“ Sensor is Running fine. 
> […](#)

Really, your power sensor is running fine? Mine shows nothing

Smanar, 2022-12-18:
Perhaps you don't have enought consumption ?
But if the value is 0 in deconz, you will have 0 in the API, the value is gived by the device.

Alex-joomla, 2022-12-19:
> Perhaps you don't have enought consumption ?
> But if the value is 0 in deconz, you will have 0 in the API, the value is gived by the device.

Thats exactly the case. Power draw was too low, had fairy lights in it. Tested it with a water boiler and it worked.

Although I cannot understand it. Current is measured, why doesn't it calculate it to power? Consumption rises anyway.

karsten-eger, 2022-12-20:
hmm, got two of those as well and they are detected just fine as a plug. All other parts are missing. I upgraded deconz and phoscon and edited the files as mentioned in #6589 but it's still detected as _TZ3000_2putqrmw and not nous at all.

restarted deconz and the raspi several times, no change.
there are no files in the users local folder either.

Smanar, 2022-12-20:
>Although I cannot understand it. Current is measured, why doesn't it calculate it to power? Consumption rises anyway.

Need to ask that to tuya devs ^^

>but it's still detected as _TZ3000_2putqrmw and not nous at all.

It is and will stay a _TZ3000_2putqrmw , the only difference is with the DDF you will have 2 more sensors

karsten-eger, 2022-12-20:
> It is and will stay a _TZ3000_2putqrmw , the only difference is with the DDF you will have 2 more sensors
I thought the device manufacturer is named nous after the change in the phoscon ui

But anyways, no, no new sensors detected. Only the plugs so far.

Smanar, 2022-12-20:
You have the GUI ? you can check if the DDF is used by deconz with making a right clic and "edit DDF", it will open the editor and you will see the used DDF.

After adding the DDF you need to re-include the device, for new entries will be created.

Smanar, 2022-12-20:
You can run deconz with command line with debug enabled, but will be a little overkill just for that.

You have edited the old file ? perhaps  a typo inside ? You are not using docker ?

karsten-eger, 2022-12-22:
Ah they are reachable via the API but they are not listed as sensors in the sensor list.
`{
    "config": {
        "on": true,
        "reachable": true
    },
    "etag": "[...]",
    "lastannounced": null,
    "lastseen": "2022-12-22T14:08Z",
    "manufacturername": "_TZ3000_2putqrmw",
    "modelid": "TS011F",
    "name": "Consumption 35",
    "state": {
        "consumption": 0,
        "lastupdated": "2022-12-22T14:04:56.880"
    },
    "swversion": "1.0.13",
    "type": "ZHAConsumption",
    "uniqueid": "a4:c1:38:0d:7e:b9:71:38-01-0702"
}`

The rest is not a deconz problem I guess.

Smanar, 2022-12-22:
But this JSON is from the sensors list ?
You mean in the phoscon sensors list ?

karsten-eger, 2022-12-22:
yep, Phoscon Webapp API client sensor information. Each of those two plugs exposes two sensors - consumtion and power respectively

Smanar, 2022-12-23:
So it's working now ?

karsten-eger, 2022-12-23:
I guess this counts as working regarding deconz. Thought it will use the given manufacturer name though.

Now it's a phoscon issue.

Thx for your help and patience. 😄

bankhans, 2022-12-29:
>   "manufacturername": ["_TZ3000_2putqrmw", "_TZ3000_2putqrmw"]
>   "modelid": ["TS011F", "TS011F"],

Thank You,
i can see now power and consumption in openhab.
[image]

RezzZ, 2022-12-30:
@karsten-eger did you just press the plug button for 10 seconds for the new sensors to show up while looking for new lights in phoscon app?
I'm trying to get the plugs working with power consumption but API and homeassistant only show on/off switch for me.
I'm running deconz in docker and I updated the file inside my docker container to reflect the recommended changes..

Kenny-BC, 2023-01-04:
Hey Guys,
i can add my Nous 2putqrmw and after i edit the ddf and save it under:
/home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices
Now i´ve the consumption and power values in homeassistant. 

I don´t know but when i open the phoscon gui sometimes there are 3 plugins online and sometimes 1 and 5 minutes later 5.

Can someone explain why they hopping? Normally they run as repeater and at the moment i´ve 68 devices listed in deconz.

Mimiix, 2023-01-04:
> Hey Guys, i can add my Nous 2putqrmw and after i edit the ddf and save it under: /home//.local/share/dresden-elektronik/deCONZ/devices Now i´ve the consumption and power values in homeassistant.
> 
> I don´t know but when i open the phoscon gui sometimes there are 3 plugins online and sometimes 1 and 5 minutes later 5.
> 
> Can someone explain why they hopping? Normally they run as repeater and at the moment i´ve 68 devices listed in deconz.

The above is unrelated to this issue or device(interference issues). Please ask on Discord or Forums.

closed 2023-02-02 by github-actions[bot]