dresden-elektronik/deconz-rest-plugin #6595: Add DDF for Xiaomi Aqara Smart Pet Feeder C1 (PETC1-M01/aqara.feeder.acn001)

lboue, 2022-12-06:
## Xiaomi Aqara Aqara Smart Pet Feeder C1

- Product name: Smart Pet Feeder C1
- Manufacturer: Xiaomi Aqara
- Model identifier: aqara.feeder.acn001
- Device type : Pet Feeder

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

rdbtn-no, 2023-01-04:
Would like support for this device as well.

frnsth, 2023-05-10:
I'd appreciate support for this device as well. Thanks in advance.

Idaho947, 2023-05-21:
Hello,

Any news about this intégration ?

Mimiix, 2023-05-21:
Happy to reopen but without dev interest it's not proceeding.

@Smanar or @BabaIsYou anything for you?

Smanar, 2023-05-21:
The problem is deconz is not made at all for this kind of device.
We can make a on/off but after .....

BabaIsYou, 2023-05-21:
> but without dev interest it's not proceeding.

AS @Smanar mentionned, It's more a question of feasibility than interest in this case. 

We have already tried to integrate some dedicated devices (valve, IR transmitter, air quality sensor) and the discussions that followed are still ongoing or the integration is still not functional.

To report some specific information like portios_weight, weight_per_day, serving_size ... we need to finish integration of something like generic ZHAMeasurement ...

Piksimou, 2023-05-26:
> The problem is deconz is not made at all for this kind of device. We can make a on/off, a "mode" setting , but after .....

I understand but on/off or "mode" setting, It's already good. the rest can be managed with scenarios on the home automation box.

Smanar, 2023-05-26:
BTW this device don't have 0x0004 attribute on the basic cluster 0x0000 ? (the manufacture name).
It's grayed on your capture, it's normal, or it have failed when you have made the capture ?

Piksimou, 2023-05-26:
I don't knwon. It's Iboue who made capture. I think i have failed when he make capture.

Piksimou, 2023-05-26:
I try tomorrow and i tell you.

Piksimou, 2023-05-29:
> No, I think it's from the device
> 
> > warn 2022-08-16 10:49:37: Received message from unsupported device with Zigbee model 'aqara.feeder.acn001' and manufacturer name 'undefined'
> 
> From Z2M. But the problem is deconz use this field to reconise the device ....
> 
> If you make a right clic on the node and "edit DDF" what is there in the field "manufacture name" ?

I try edit ddf and there are nothing in field "manufacure name".
 0x0004 attribute on the basic cluster 0x0000 ? (the manufacture name), he is gray also.

But in my network zigbee, device work fine in router.

Smanar, 2023-05-30:
Yeah but the zigbee working mode have nothing to see with the API working mode.

From my memory deconz use the model ID AND the manufacture name to reconise device.

```
    if (ddf.manufacturerNames.last().isEmpty() || ddf.modelIds.isEmpty() || ddf.modelIds.front().isEmpty())
    {
        return false;
    }
```

So I don't think it's possible to do something for it.

closed 2023-06-28 by github-actions[bot]

meetyourlaser, 2023-09-30:
Do I understand correctly: we will never see support for the Xiaomi Aqara Smart Pet Feeder C1 in DeCONZ? :'-(

Smanar, 2023-09-30:
Not if the manufacture name is empty.