dresden-elektronik/deconz-rest-plugin #6597: Tuya Zigbee Door Sensor ZG-102Z

ThorstenDtk, 2022-12-06:
## Device

- Product name: Zigbee Door Sensor ZG-102Z
- Manufacturer: _TZE200_pay2byax
- Model identifier: TS0601
- Device type : Sensor

## Screenshots

### Node
[image]

### Node Info
[image]

### Power Configuration
[image]

### IAS Zone

[image]

### Basic

[image]

### OTAU

[image]

I read a couple of posts like https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5331, created and loaded the following DDF:
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_pay2byax",
  "modelid": "TS0601",
  "product": "ZG-102Z",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 3700,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0002",
            "cl": "0x0500",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2",
            "fn": "zcl"
          }
        },
        {
          "name": "config/checkin",
          "public": false,
          "awake": true,
          "refresh.interval": 3600,
          "static": 3600
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/open",
          "awake": true,
          "parse": {
            "at": "0x0002",
            "cl": "0x0500",
            "ep": 1,
            "eval": "Item.val = (Attr.val & 0x3) != 0"
          }
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 43200,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

The device does not appear in Phosco. The object is visible in iOBroker (I'm using the DeCONZ adapter):

[image]

But the values like open or battery are never updated.

Unfortunately I already bought 15 of these sensors (my bad, I thought that they work with DeCONZ). So I would really like to make them work. Any help would be appreciated!

ThorstenDtk, 2022-12-07:
When I move the magnet away from the sensor (open) I see the following log entries:
```
22:50:53:649 Tuya debug 4 : Address 0xA4C138C533461230 Payload 00bd0101000101
22:50:53:651 Tuya debug 5 : Status: 0 Transid: 189 Dp: 257 (0x01,0x01) Fn: 0 Data 1
22:50:56:669 Tuya debug 4 : Address 0xA4C138C533461230 Payload 00be6502000400000006
22:50:56:671 Tuya debug 5 : Status: 0 Transid: 190 Dp: 613 (0x02,0x65) Fn: 0 Data 6
```

And when I put the magnet next to the sensor (close):

```
22:51:41:589 Tuya debug 4 : Address 0xA4C138C533461230 Payload 00c00101000100
22:51:41:591 Tuya debug 5 : Status: 0 Transid: 192 Dp: 257 (0x01,0x01) Fn: 0 Data 0
22:51:44:640 Tuya debug 4 : Address 0xA4C138C533461230 Payload 00c16502000400000005
22:51:44:641 Tuya debug 5 : Status: 0 Transid: 193 Dp: 613 (0x02,0x65) Fn: 0 Data 5
22:51:46:629 Tuya debug 4 : Address 0xA4C138C533461230 Payload 00c20101000100
22:51:46:630 Tuya debug 5 : Status: 0 Transid: 194 Dp: 257 (0x01,0x01) Fn: 0 Data 0

```

ThorstenDtk, 2022-12-08:
> Lol, not your first tuya device ? Seriously users like you are a pleasure.

Thank you! Well, I'm working in IT support for about 25 years now. And I wouldn't ask for help if I didn't do any research before. But you're wrong, it's my first Tuya device. All others (Tradfri, GLEDOPTO, ...) worked out of the box. The point is that  I couldn't find any 'how to create DDF for new devices'.  And figuring out how to do that would most likely require a lot of time - because I assume you need to know the protocol being used, etc. I consider myself expert in Exchange, SPF/DKIM/DMARC, PS scripting, etc. And I have no ambition to leave the 'user level' of Zigbee/DeCONZ knowledge. 

I just tested the DDF, and it works like a charm. You made my day!

[image]

I paired a 2nd sensor and it's working too.

Thank you very much for helping so quickly! If you ever travel to Freising (close to Munich airport, we have the world's oldest continuously operating brewery here) please let me know, I owe you a beer!

And no, I won't ask how to make the devices appear in PhosCON. At least not until I made sure that I won't find the answer here in the forum.

Smanar, 2022-12-09:
We was in germany the last month, too late ^^

But it's not complicated when users give all usefull informations even you haven't asked them yet, and already made 90% of the work.

Wana make a PR to have it officialy ? or do you want I make it ?

Smanar, 2022-12-10:
PR done here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6606

closed 2023-01-09 by github-actions[bot]