dresden-elektronik/deconz-rest-plugin #6598: CoRui Tuya Door Window Sensor // TZ3000_au1rjicn // TS0203

Burn-In, 2022-12-07:
## Device

- Product name: CoRui Tuya Door Window Sensor
- Manufacturer: TZ3000_au1rjicn
- Model identifier: TS0203
- Device type : Sensor
  
## Screenshots

[image]
[image]

### Power Configuration
[image]

Smanar, 2022-12-10:
You have a little typo to edit before testing the DDF

`"modelid": ["TS0203", "TS0203",, "TS0203"],`

BabaIsYou, 2022-12-10:
My bad … sorry

Burn-In, 2022-12-13:
I don't know how to use a DDF for the device. Because I'm using the HA-Deconz-Addon i have no external access to the deconz folder where the *.json is saved.

I edited the current DDF with the editor but without could adding bindings because i cannot control the windows in the in the Addons Deconz interface. So, drag-n-drop of bindings is not possible.

After using that edited one, the device will be found in Phoscon but without status updates.
[image]

BabaIsYou, 2022-12-13:
Good beginning ;-)

You also miss the battery parsing that tou can also ass using DDF editor.

BTW without bindings there is chance it won’t work either.

i’m not using HA, but I saw some threads in HA forum that could help to access devices json folder, like that [one](https://community.home-assistant.io/t/how-to-install-a-new-deconz-ddf-file/390455) 

Hope this help ;-)

Smanar, 2022-12-13:
Yep, I think too you can do all the stuff with the editor, you have understant how it work.

Bonus:
Drag and drop on HA https://forum.phoscon.de/t/ddf-editor-how-to-drag-bindings/1351
To manage DDF you have too this link https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

Short story, the persistent file for DDF is hidden and is `/data/.local/share/dresden-elektronik/deCONZ/devices/`

Burn-In, 2022-12-15:
> Bonus: Drag and drop on HA https://forum.phoscon.de/t/ddf-editor-how-to-drag-bindings/1351 

Sure i had read this already, but in the Deconz GUI, when started from inside the HA addon, the DDF Editor screen is not floating. It is always fullscreen over the device screen. So, the only existing method to add bindings by drag-and-drop is not possible.

[image]

> To manage DDF you have too this link https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

It seems to be there no working simple method to enter an addon container from HA OS through the Terminal or Editor.

Smanar, 2022-12-15:
> Undocking the “Cluster Info-Tab” into a floating panel, as shown, using the icons in the top right corner of the Tab works when using the noVNC view in home assistant using the deconz plugin/addon.

>It seems to be there no working simple method to enter an addon container from HA OS through the Terminal or Editor.

```
docker container ls
docker exec -it {id} /bin/bash
cd /data/.local/share/dresden-elektronik/deCONZ/devices/
```

Burn-In, 2022-12-16:
Regarding “Cluster Info-Tab”, that's what I was thinking and looking for the whole time. Unfortunately, all windows don't have control icons on the right, running always fullscreen. Refer to the picture above.

`ssh root@{localIP} -p 22222` returns only `Connection refused`

Smanar, 2022-12-16:
You have put the good local ip ?
Perhaps HA need an addon to use ssh ?

Burn-In, 2022-12-16:
The IP is correct but I don't know if the port is so.

Burn-In, 2022-12-16:
I also connected with the noVPN addon to Deconz GUI, it looks the same. It seems the title bars are not cut on the right but they don't have icons there at all.

Smanar, 2022-12-17:
> Ssh to the hassio on Raspberry. You need to use a ADDON or something for this
> Either SSH & Web Terminal or some similar plugin, or just ssh root@{ip_address} -p 22222

Can try one of the 2 plugin ?

closed 2023-01-16 by github-actions[bot]

Xyaren, 2023-01-31:
I now also am an owner of said device.
I can confirm the DDL from above works (after fixing the dublicate `,` in the modelids).

@Smanar do you need me to do any testing ?

BabaIsYou, 2023-01-31:
Good News. Does your DDF also includes         {
          "name": "config/pending"
        },
        {
          "name": "config/enrolled",
          "public": false
        },

could you paste it and/or create a PR if already complete ?

BabaIsYou, 2023-02-01:
> What do I need to check ?

If device is recognized with a name and not an hexdecimal denomination and that you can check that it returns information in HA as you expected them (open/close/battery ...) then it's fine.

Checkdat, 2025-02-09:
Hi,
I have paired some TZ3000 (_TZ3000_n2egfsli) with the modelid SNZB-04 (change in DDF-file)

Smanar, 2025-02-10:
But there is already something for the "_TZ3000_n2egfsli", here https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0203_door_sensor.json

Checkdat, 2025-02-11:
> Ha ?
> So you mean the maufacture name is "_TZ3000_n2egfsli but the model id not "TS0203" but "SNZB-04" ?
> So yes impossible to guess it, and honestly, i didn't think it was possible, it mean Tuya sell some hardware, so with the same manufacture name, to other brands, and thoses one just change the model id  ?
> And if it's that, you are right, better to edit the \sonoff\snzb-04_open_close_sensor.json

Hi,I bought these from AliExpress:and just editedt the TZ3000 DDF with Model-ID part with SNZB-04Why is se Sonoff DDF the smarter choice?Cause the Manufacture is _TZ3000_n2egfsliAnd now it works very fine for me.Mit freundlichen Grüßen Marco Braunbeck

Smanar, 2025-02-11:
No problem, you are right, have found another topic with the same discover > https://github.com/JohanBendz/com.tuya.zigbee/issues/998
And have check the link, nothing about "Sonoff", or about "SNZB-04", I think they have a problem here ^^.

So do you want to make a PR to submit the device (explain the model id else others devs will react like me) or Want I make it ?

Checkdat, 2025-02-11:
Please go for it !Think your Post will have a biger benefit for the Cumounity ;-).Im glad when i my telling has helped Mit freundlichen Grüßen Marco Braunbeck

Smanar, 2025-02-12:
Thx, done > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8121