dresden-elektronik/deconz-rest-plugin #6600: Aqara light sensor t1 (GZCGQ11LM)

Ragnar1985, 2022-12-07:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Light
  - Lock
  - Remote
  - Sensor
  - Siren
  - Switch
  - Thermostat
  - Other: 

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2022-12-07:
Hi,

There's missing a lot of information. Please add correct screenshots.

Ragnar1985, 2022-12-07:
Is that enough information?

SwoopX, 2022-12-08:
I'd expect the device to work if you extend your local `xiaomi_gzcgq01lm_light_sensor.json` with lumi manufacturer code and your model ID at the top of the file. You might wanna give that a go.

Ragnar1985, 2022-12-09:
I'm a newbie, how do I do that?

SwoopX, 2022-12-10:
Find the file mentioned above on your system, open it with an editor, find the respective lines and replace them with the below. Then save, restart deconz and reset and re-pair your light sensor.

```
  "manufacturername": ["$MF_XIAOMI", "$MF_LUMI", "$MF_LUMI"],
  "modelid": ["lumi.sen_ill.mgl01", "lumi.sen_ill.mgl01", "lumi.sen_ill.agl01"],
```

Ragnar1985, 2022-12-11:
It works with this lines
"manufacturername": ["$MF_XIAOMI", "$MF_LUMI", "$MF_LUMI"],
  "modelid": ["lumi.sen_ill.mgl01", "lumi.sen_ill.mgl01", "lumi.sen_ill.agl01"],

Mimiix, 2022-12-11:
Please wait until its merged

closed 2023-01-07 by manup