dresden-elektronik/deconz-rest-plugin #6601: Jung wall switch stops working within 12 hours

RezzZ, 2022-12-07:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi3b+
 - Running method: raspbian docker deconzcommunity/deconz:stable
 - Firmware version: (26680700)
 - deCONZ version: (2.19.01 / 19/09/2022)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Mimiix, 2022-12-07:
The logs are in Deconz, not Phoscon.

I am tempting to close this as it is not compliant with #5113.

Can you please update to 2.19.3 to rule another issue out?

RezzZ, 2022-12-07:
Hi @Mimiix , just updated to 2.19.3
I will re-pair the switch and will update this thread tomorrow. If you can direct me into the position for the logs, in the meantime that will me much apreciated.

RezzZ, 2022-12-08:
also after upgrade the wall switch stopped communicating deconz instance

Mimiix, 2022-12-08:
Can you share some logs?

APS, APS L2, ERROR, ERROR L2, INFO,INFO_L2.

RezzZ, 2022-12-08:
I will try to fetch them. unfortunately I don't have the debug view in my deconz gui so looing for a different way to extract them

RezzZ, 2022-12-08:
[image]

RezzZ, 2022-12-08:
here one minute of the logs you mentioned. I pressed buttons from the jung switch around 11:18:40
```11:18:18:395 Set sensor check interval to 100 milliseconds
11:18:18:405 rule event /config/localtime: 11:18:17.425 -> 11:18:18.396 (971ms)
11:18:18:739 poll node 00:0b:57:ff:fe:d4:bc:ab-01
11:18:18:740 Poll light node Kitchen Dimmable 1
11:18:18:791 Poll APS request to 0x000B57FFFED4BCAB cluster: 0x0006 dropped, values are fresh enough
11:18:19:411 rule event /config/localtime: 11:18:18.396 -> 11:18:19.395 (999ms)
11:18:19:820 poll node 00:0d:6f:ff:fe:2c:92:6e-01
11:18:19:821 Poll light node Range extender Tower
11:18:19:911 APS-DATA.request id: 55, addrmode: 0x03, addr: 0x0017880100e13833, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
11:18:19:913 	asdu (length: 2): 5904
11:18:19:947 APS-DATA.confirm id: 55, status: 0x00 SUCCESS
11:18:19:949 APS-DATA.confirm request id: 55 -> confirmed, timeout 0
11:18:20:027 APS-DATA.indication srcAddr: 0xfff7, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -51
11:18:20:029 	asdu: 5900140402984005ffff2e210051fe950c0188170085693902ff4b984005ffff2e21005bec9e0c01881700f3093902ff53
11:18:20:030 APS-DATA.indication request id: 55 -> finished
11:18:20:031 APS-DATA.request id: 55 erase from queue
11:18:20:396 rule event /config/localtime: 11:18:19.395 -> 11:18:20.395 (1000ms)
11:18:20:742 poll node 00:17:88:01:0c:95:fe:51-0b
11:18:20:743 Poll light node Living Duo2.2
11:18:20:803 read attributes of 0x001788010C95FE51 cluster: 0x0006: [ 
11:18:20:805 0x0000 
11:18:20:806 ]
11:18:20:807 add task 8922 type 19 to 0x001788010C95FE51 cluster 0x0006 req.id 59
11:18:20:808 Poll APS request 59 to 0x001788010C95FE51 cluster: 0x0006
11:18:20:810 APS-DATA.request id: 59, addrmode: 0x03, addr: 0x001788010c95fe51, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 0 len: 5 tx.options 0x00
11:18:20:811 	asdu (length: 5): 10fb000000
11:18:20:866 Poll APS confirm 59 status: 0x00
11:18:20:868 Erase task req-id: 59, type: 19 zcl seqno: 251 send time 0, profileId: 0x0104, clusterId: 0x0006
11:18:20:869 APS-DATA.confirm id: 59, status: 0x00 SUCCESS
11:18:20:870 APS-DATA.confirm request id: 59 -> erase from queue
11:18:20:872 aps request id: 59 finished, erase from queue
11:18:20:897 APS-DATA.indication srcAddr: 0x6985, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -51
11:18:20:898 	asdu: 18fb010000001000
11:18:20:899 Node data 0x001788010c95fe51 profileId: 0x0104, clusterId: 0x0006
11:18:20:900 0x001788010C95FE51: update ZCL value 0x0B/0x0006/0x0000 after 0 s
11:18:21:395 poll node 00:17:88:01:02:e0:c5:01-0b
11:18:21:396 Poll light node Bedroom Ceiling
11:18:21:396 Idle timer triggered
11:18:21:412 rule event /config/localtime: 11:18:20.395 -> 11:18:21.395 (1000ms)
11:18:21:416 read attributes of 0x0017880102E0C501 cluster: 0x0006: [ 
11:18:21:418 0x0000 
11:18:21:419 ]
11:18:21:420 add task 8926 type 19 to 0x0017880102E0C501 cluster 0x0006 req.id 64
11:18:21:421 Poll APS request 64 to 0x0017880102E0C501 cluster: 0x0006
11:18:21:495 APS-DATA.request id: 64, addrmode: 0x03, addr: 0x0017880102e0c501, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 0 len: 5 tx.options 0x00
11:18:21:496 	asdu (length: 5): 10fc000000
11:18:21:525 Poll APS confirm 64 status: 0x00
11:18:21:526 Erase task req-id: 64, type: 19 zcl seqno: 252 send time 0, profileId: 0x0104, clusterId: 0x0006
11:18:21:527 APS-DATA.confirm id: 64, status: 0x00 SUCCESS
11:18:21:527 APS-DATA.confirm request id: 64 -> erase from queue
11:18:21:546 APS-DATA.indication srcAddr: 0x896a, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 247, rssi: -63
11:18:21:547 	asdu: 18fc010000001000
11:18:21:548 APS-DATA.request id: 64 erase from queue
11:18:21:549 Node data 0x0017880102e0c501 profileId: 0x0104, clusterId: 0x0006
11:18:21:549 0x0017880102E0C501: update ZCL value 0x0B/0x0006/0x0000 after 0 s
11:18:21:551 Websocket 10.105.1.5:37202 send message: {"attr":{"colorcapabilities":16,"ctmax":454,"ctmin":153,"id":"12","lastannounced":"2022-08-22T18:29:52Z","lastseen":"2022-12-08T10:18Z","manufacturername":"Philips","modelid":"LTC001","name":"Bedroom Ceiling","swversion":"1.50.2_r30933","type":"Color temperature light","uniqueid":"00:17:88:01:02:e0:c5:01-0b"},"e":"changed","id":"12","r":"lights","t":"event","uniqueid":"00:17:88:01:02:e0:c5:01-0b"} (ret = 48420824)
11:18:21:552 Websocket 10.105.1.5:42256 send message: {"attr":{"colorcapabilities":16,"ctmax":454,"ctmin":153,"id":"12","lastannounced":"2022-08-22T18:29:52Z","lastseen":"2022-12-08T10:18Z","manufacturername":"Philips","modelid":"LTC001","name":"Bedroom Ceiling","swversion":"1.50.2_r30933","type":"Color temperature light","uniqueid":"00:17:88:01:02:e0:c5:01-0b"},"e":"changed","id":"12","r":"lights","t":"event","uniqueid":"00:17:88:01:02:e0:c5:01-0b"} (ret = 48420824)
11:18:22:395 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
11:18:22:443 rule event /config/localtime: 11:18:21.395 -> 11:18:22.398 (1003ms)
11:18:22:501 poll node 00:17:88:01:02:5c:7e:39-0b
11:18:22:502 Poll light node Dining 3
11:18:22:549 read attributes of 0x00178801025C7E39 cluster: 0x0006: [ 
11:18:22:550 0x0000 
11:18:22:550 ]
11:18:22:551 add task 8931 type 19 to 0x00178801025C7E39 cluster 0x0006 req.id 70
11:18:22:551 Poll APS request 70 to 0x00178801025C7E39 cluster: 0x0006
[... cut ...]
11:18:59:528 0x0000 
11:18:59:529 ]
11:18:59:529 add task 9091 type 19 to 0x0017880100D6AF65 cluster 0x0006 req.id 73
11:18:59:529 Poll APS request 73 to 0x0017880100D6AF65 cluster: 0x0006
11:18:59:623 Poll APS confirm 73 status: 0x00
11:18:59:625 Erase task req-id: 73, type: 19 zcl seqno: 21 send time 0, profileId: 0x0104, clusterId: 0x0006
11:18:59:673 Node data 0x0017880100d6af65 profileId: 0x0104, clusterId: 0x0006
11:18:59:675 0x0017880100D6AF65: update ZCL value 0x0B/0x0006/0x0000 after 0 s```

Mimiix, 2022-12-08:
Can you show me a screenshot of deCONZ?

RezzZ, 2022-12-08:
here a screen, are you interested in a specific part?
[image]

Mimiix, 2022-12-08:
I'm wondering if there's any bad connections. To which one is it connection when it's online?

Mimiix, 2022-12-08:
> hmm no lines, but the device works.. do I need to refresh the network? btw, I really appreciate your support! [image] update1, I new line appeared: [image] it's connected to a blitzwolf smart power plug (even though the conbeeII controller is less then 2 meters away, other side of the wall in the metercloset) update2, hmm the blitzwolf plug that is connected is not even in the same room, its upstairs in a bedroom..

That's probably the reason it drops. But it picks itself, so it seems that it finds this better.

I'd really like to close this issue here, as this is not a bug. Would you be able to open a topic on the forums?

Mimiix, 2022-12-08:
No worries :) Feel free to tag/ping me on Forums or perhaps Discord.

closed 2022-12-08 by Mimiix