dresden-elektronik/deconz-rest-plugin #6603: Device name

tobiraak, 2022-12-09:
## Device

- Product name: Tuya Zigbee Temp & Humidity Sensor
- Manufacturer: _TZE200_yjjdcqsq
- Model identifier: TS0601
- Device type : 
  - Sensor

## Screenshots

[image]

[image]

No other clusters available sadly. 

I am using home assistant, trying to connect the temp sensors via phoscon/deconz but it is not connecting and not showing up in the ui.

BabaIsYou, 2022-12-09:
A bit surprised that you don't have a Tuya specific cluster (0XEF00) if you don't have Temp (0x402) or Humidity (0x405) standard clusters. Are you sure that joining process for that device was correct ?

After rejoining, and if you have Tuya cluster then you could perhapds try this DDF file 

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_yjjdcqsq",
  "modelid": "TS0601",
  "product": "ZTH02",
  "sleeper": true,
  "status": "Silver",
  "subdevices": [
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = 10 * Attr.val;" },
          "read": {"fn": "tuya"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = 10 * Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}
```

tobiraak, 2022-12-10:
My connecting procedure was this: Make the Temp Sensor go into pairing mode -> open up phoscon webapp through Home Assistant and go into pairing mode as well -> since they didn't show up there I went into the deconz app (also via Home Assistant), and here they showed up but without any cluster.

If you have any other procedure that I can try please let me know. I will try the DDF file even though I have no clusters.

BabaIsYou, 2022-12-10:
Your process seems correct but sometimes devices won’t complete joining process … but can be completed by rejoining it without deleted it first.

Does the DDF of any help (not for connection anyway but for attributes)

tobiraak, 2022-12-11:
I tried to reconnect again, they won't connect. I deleted the nodes and re-added them but still the same. Also: I am unable to edit the DDF. I have to rightclick the node and "edit DDF" right? No chance to do so unfortunately. Do you have any other idea? I have two of them so I can confirm that they are malfunctioning.

BabaIsYou, 2022-12-12:
Just to be sure : you deleted it and it won’t pair anymore ?

tobiraak, 2022-12-12:
They paired again but still the same problem line in the beginning. But I tried your steps first that you provided me in the first place.

BabaIsYou, 2022-12-12:
> Also: I am unable to edit the DDF. I have to rightclick the node and "edit DDF" right? No chance to do so unfortunately

I'm mot familiar with HA but think you van fond a way to add DDF file with that thread : [https://community.home-assistant.io/t/how-to-install-a-new-deconz-ddf-file/390455](url)

nanostrukturtechniker, 2022-12-29:
For me the sensor came with a pretty empty battery, maybe this is the reason for the incomplete pairing?

Here are all clusters for me:
[image]

[image]

[image]

[image]

[image]

[image]

[image]

Smanar, 2022-12-29:
>For me the sensor came with a pretty empty battery, maybe this is the reason for the incomplete pairing?

No it's just the device haven't send value.
To have them press the "read" button, but you need to wake up the device in same time (generaly using short press on the reset button)

tobiraak, 2022-12-29:
The device is still not able to connect even with all your provided troubleshooting. How ever I tried the Zigbee Plugin that works like a charm. Every Zigbee Device that I have is recognized even the temperature sensors. I just needed to set the DeconZ Bridge in the Zigbee Plugin Settings.

I don't know why the DeconZ/Phoscon Plugin is not working but for everyone with the same issues I can recommend switching from the DeconZ Plugin to the Zigbee Plugin if you are using Home Assistant. 
I will close this issue since I got it working with a different approach.

PR #6986 (2023-05-21): DDF add ZTH01 & ZTH02 temp and humidity sensor
Fix #6603
commit 1136ffd4 (2023-05-21): DDF add ZTH01 & ZTH02 temp and humidity sensor (#6986)

Fix #6603
closed 2022-12-29 by tobiraak