dresden-elektronik/deconz-rest-plugin #6604: Tuya temperature & humidity sensor ZG227Z tze200_a8sdabtg TS0601

ThorstenDtk, 2022-12-09:
[image]

## Device

- Product name: Tuya temperature & humidity sensor
- Manufacturer: tze200_a8sdabtg 
- Model identifier: TS0601
- Device type : Sensor

## Screenshots

[image]

### Basic
[image]

### Temperature Measurement
[image]

### Relative Humidity Measurement
[image]

### Basic
[image]

### OTAU
[image]

I slightly modified the DDF from https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6063 (manufacturer ID, path):
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_a8sdabtg",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "Temperature and humidity sensor",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/_TZE200_a8sdabtg.json",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = 10 * Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = 10 * Attr.val;" },
          "read": {"fn": "tuya"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```
It's almost working, it's just the battery which doesn't get updated:

[image]

I turned on logging in the GUI, but there are no log entries starting with 'Tuya debug 4/5'. Not even for the Tuya door contacts. 
Which log level is required, and which log entries are relevant?

Don't know if it's the device which periodically sends data or when a value changes or if it can be polled. 

From PhosCON:
```

    "36": {
        "config": {
            "battery": 0,
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "etag": "4c1e875cac0b50dc8ca8eb93740d2256",
        "lastannounced": null,
        "lastseen": "2022-12-09T20:56Z",
        "manufacturername": "_TZE200_a8sdabtg",
        "modelid": "TS0601",
        "name": "Humidity 34",
        "state": {
            "humidity": 780,
            "lastupdated": "2022-12-09T20:57:13.535"
        },
        "type": "ZHAHumidity",
        "uniqueid": "a4:c1:38:35:7b:18:6b:9d-01-0405"
    }
```

BabaIsYou, 2022-12-10:
Hello,
You didn't provide complete screenshot for cluster 0x001 then not easy to see if it's used or if you have to use Tuya cluster (like temp & humidity).

Perhaps you can try this for battery (as suggested in #6535)   : 

```
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
```

And BTW this for swversion

```
        {
          "name": "attr/swversion",
          "refresh.interval": 84000,
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl",
            "script": "tuya_swversion.js"
          }
        },
```

Smanar, 2022-12-10:
No, you are right, this device is a clone of this one https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6543

ThorstenDtk, 2022-12-10:
I took the ddf from #6543, edited manufaturername and modelid, deleted the node and added it back. This is the DDF I used:

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_znbl8dj5", "_TZE200_qoy0ekbd", "_TZE200_a8sdabtg"],
  "modelid": ["TS0601", "TS0601", "TS0601"],
  "vendor": "Tuya",
  "product": "Temperature Humidity Sensor TS0601",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = (100 * Attr.val) + R.item('config/offset').val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = (10 * Attr.val) + R.item('config/offset').val;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}
```

Battery still isn't working, but now the humidity looks ok:

[image]

[image]

Do I just need to wait longer to get a value for battery?

BabaIsYou, 2022-12-10:
You won’t see any value in GUI because this device seems to use only Tuya cluster for humidity (that why we use Tuya function in DDF). Then, if it works you’ll see the correct value in phoscon

Smanar, 2022-12-10:
Yep, have checked again, I m sure it's the dpid 4.
Can need to wait a little.

ThorstenDtk, 2022-12-10:
> Yep, have checked again, I m sure it's the dpid 4. Can need to wait a little.

You're right, it's working. Thanks a lot!

ThorstenDtk, 2022-12-10:
> You won’t see any value in GUI because this device seems to use only Tuya cluster for humidity (that why we use Tuya function in DDF). Then, if it works you’ll see the correct value in phoscon

Are you talking about DeCONZ? I can see it there:

[image]

Smanar, 2022-12-10:
Hu ?
Lol, idk how it can be visible on deconz neither ...

I will ask the user to add your manufacture name on his PR

quikfast, 2022-12-16:
Sorry I'm a noob on this, I have these sensor also but do not what to do exaclty, i do not see it in the Phoscon but in Deconz, and Deconz is not something what I used daily from Home Assistant.
[image]

I need to edit the DDF? 
When I'm in this I see the following:

[image]
How can I edit the DDF and copy the text from above?

Smanar, 2022-12-16:
I think it will be faster to create a DDF in the hidden persistent folder` /data/.local/share/dresden-elektronik/deCONZ/devices/`

https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

quikfast, 2022-12-16:
How can I create this? I can save it from deconz this folder, but how can I edit the JSON file from SSH?

ThorstenDtk, 2022-12-16:
In SSH you can use pico (sudo pico <filename>), copy in from the clipboard and then save the file.

quikfast, 2022-12-16:
@ThorstenDtk  sorry mate, still not clear for me. 

I have created a JSON file with the name TEMP.json. This file is located on my Home Assistant in the /config/ folder. 
It contains the following code:

`{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_znbl8dj5", "_TZE200_qoy0ekbd", "_TZE200_a8sdabtg"],
  "modelid": ["TS0601", "TS0601", "TS0601"],
  "vendor": "Tuya",
  "product": "Temperature Humidity Sensor TS0601",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = (100 * Attr.val) + R.item('config/offset').val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = (10 * Attr.val) + R.item('config/offset').val;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}`

Now I need to copy this TEMP.JSON file from /config/ to the Deconz folder, but do not know how. 
Even when it is copied, I do not know what to do, open DDF and reload?

quikfast, 2022-12-17:
Ok I was able to edit a file now with Nano and save it this to the data folder of the deconz, however I open the file and can see every thing, I click on hot reload and save.

[image]

But when I edit again the DDF of the object, it shows again empty:

[image]

ThorstenDtk, 2022-12-17:
I'm not an expert. I also saw strange things when editing the DDF through the 'edit DDF' button. So at the end I just created  & saved the DDF with pico to the right folder. And after repairing the device(s) it worked.

Smanar, 2022-12-17:
If the DDF is in the good folder, just just need to restart deconz.
It will be used automatically, don't need to load it.

What is the file path ?

on normal OS

> All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

On HA
`/data/.local/share/dresden-elektronik/deCONZ/devices/`

quikfast, 2022-12-17:
Ok So I did modify it from the folder: /data/.local/share/dresden-elektronik/deCONZ/devices/
I opened the object in deconz -> Right click -> Edit DDF -> Open file -> /data/.local/share/dresden-elektronik/deCONZ/devices/filname.JSON file selected and all is loaded.

[image]

Clicked on save and after that Hot reloaded. Rebooted HA and restarted Deconz but still not visible. What is the correct step exactly...

quikfast, 2022-12-17:
Removed the JSON file from command line and worked again, also after a new try it worked now, the sensor is not visible in phoscon but in HA it is.

However, the battery shows up as 0% and also do not get any temperature..

[image]
[image]

quikfast, 2022-12-17:
The device is visible but I removed the entities, my fault. Does someone know how I can remove the whole device? 
Because I removed it from Deconz, restarted HA but still visible. Device is now not working...

quikfast, 2022-12-17:
I can see the device from the api:

                    {
    "config": {
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "etag": "b08a50ae0e3ac2136d3ee225089ba01c",
    "lastannounced": null,
    "lastseen": "2022-12-17T14:05Z",
    "manufacturername": "_TZE200_a8sdabtg",
    "modelid": "TS0601",
    "name": "Temperature 8",
    "state": {
        "lastupdated": "none",
        "temperature": null
    },
    "type": "ZHATemperature",
    "uniqueid": "a4:c1:38:0c:bc:68:79:80-ff-0402"
}

I removed it from HA, my fault and need to repair it, however I'm not able to remove the device in Deconz, I can remove it but its still visble in the API.

Smanar, 2022-12-17:
>"uniqueid": "a4:c1:38:0c:bc:68:79:80-ff-0402"

wich one deconz version have you ?

You have a "bad" entry, with "FF" endpoint, you need to have "01".

Can you remove it using phoscon ?
Then in deconz with "del" key ?

>I opened the object in deconz -> Right click -> Edit DDF

Just with making that, you need to see the DDF, you don't need to use "load"

quikfast, 2022-12-17:
> 

I would like to remove the device and start over... right click delete node works, however in the api the device is still visible..
[image]

So it must be: "uniqueid": "a4:c1:38:0c:bc:68:79:80-01-0402"?

quikfast, 2022-12-17:
Don't aks me why and how, but it works finally!! 
It has also a unique ID right now.

Other question, if I would like to order one more, will there be any conflict?

Smanar, 2022-12-17:
>Other question, if I would like to order one more, will there be any conflict?

No ^^, you have done the harder, one time the DDF is in the good folder, it will be used for all new inclusion.
And if you have used an "user" folder, it will be prioritary on the deonz one, so it will be not removed on next deconz update.

I will check if there is a PR for this device.

quikfast, 2022-12-18:
Thanks! I do see only that the sensor isn't updating anymore... 

                    {
    "config": {
        "battery": 100,
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "etag": "9c7a9feb4adf29107aa15ed98814d963",
    "lastannounced": "2022-12-17T16:43:58Z",
    "lastseen": "2022-12-17T17:16Z",
    "manufacturername": "_TZE200_a8sdabtg",
    "modelid": "TS0601",
    "name": "Temperature 4",
    "state": {
        "lastupdated": "2022-12-17T17:12:58.041",
        "temperature": 1910
    },
    "type": "ZHATemperature",
    "uniqueid": "a4:c1:38:0c:bc:68:79:80-01-0402"

Last was 13 hours go...

Smanar, 2022-12-18:
Yes, it's a a tuya device, so you can't have bind/report issue.

The connexion ? can you make a try close to a router/coordinator ? But in this situation you will have a bad lastseen too.

Same issue for humidity ?

quikfast, 2022-12-18:
Yeah I started over, but after a reset the device is not detected anymore by deconz. So I think I will stop it for now and will look further.

closed 2023-01-16 by github-actions[bot]