dresden-elektronik/deconz-rest-plugin #6610: Connecte Zigbee Smart Plug

corvy, 2022-12-14:
## Device

- Product name: Connecte Smart Socket 16A 
- Manufacturer: _TZ300_5ity3zyu
- Model identifier: TS0121
- Device type : 
  - Other: smart socket

[image]

## Screenshots

[image]

### Basic
[image]

[image]

### Identify
[image]

### Scenes
[image]
[image]
[image]
[image]

### Groups
[image]
[image]

### On Off 
[image]
[image]

### Simple Metering
[image]

### Electrical Measurement
This is the one I am most interested in. I want the consumption (active power) data to be available in Home Assistant as a sensor. 
[image]

### Tuya Specific
[image]

### OTAU
[image]

This device can be added without a problem. Seems to be an OEM Tuya device. Registers as a light-switch in Deconz. My #1 wish is to get the consumption data into HA to do some automation. 

Thank you!

BabaIsYou, 2023-01-09:
Could you begin to try with that DDF ? May be with some chance it won't exclusively use Tuya specific cluster ;-)

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ300_5ity3zyu",
  "modelid": "TS0121",
  "product": "TS0121",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SMART_PLUG",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0006",
            "cl": "0x0000",
            "ep": 0,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0006",
            "cl": "0x0000",
            "ep": 255,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
[... 215 lines cut ...]
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 30,
          "max": 360,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 30,
          "max": 360
        }
      ]
    }
  ]
}
```

closed 2023-02-08 by github-actions[bot]

corvy, 2023-02-08:
Sorry for not responding here. My Zigbee network just crashed totally so I have migrated to Zigbee2MQTT. Thanks for your efforts  @BabaIsYou. Good luck on the project going forward. Maybe we will meet again ;)