dresden-elektronik/deconz-rest-plugin #6611: Xiaomi Devices weird behavior after Deconz Restart

marcor78, 2022-12-15:
## Describe the bug

Hello!
I recently migrated from marthoc/deconz to the community deconz image and I'm experiencing an odd behaviour from all my xiaomi devices. (I already posted this issue in the docker community but it seems this is not related to the docker image).
My network is made with 17 xiaomi devices (multisensors, motion sensors and contact sensors) and a bunch of Ikea Tradfri devices (remotes, motion sensors, lights).
I use to reboot deconz every night on a cronjob and after every reboot - in the UI - ikea tradfri devices are immediately available and with their correct state, while xiaomi devices are all unavailable until their state changes (exmotion detected, temperature changes, etc). After the first xiaomi device is available again in the UI, slowly all other devices become available, but with a wrong state: all binary_sensors (ex. contact, motion) appear as "on" even if they're really off (door closed appear as open in Deconz UI).

## Steps to reproduce the behavior

 In practice:

- 3am: deconz restarts on cronjob
- 4am: homeassistant restarts on cronjob
- 7am: I wake up and all the motion sensors are "on" and my front door appears as open, with a heart attack included

I tried everything:

- unpair-repair devices
- changed batteries
- reset devices through VNC
- repull docker image

Result: none of this changed the behavior

## Expected behavior

Xiaomi Devices should be available soon after Deconz restart and should correctly report current phisical state

## Screenshots

[image]

## Environment

 - Host system: NUC (NUC6CAYH) with Proxmox VE 7.3-3
 - Running method: Community Docker container running in VM
 - Firmware version: 264A0700
 - deCONZ version: 2.19.03
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? a couple of ESP32 boards, an UPS (Tecnoware UPS ERA PLUS 1100) and 2 HD storage (WD and Seagate)

## deCONZ Logs

## Additional context

Mimiix, 2022-12-15:
Hi,

As this is not a bug report as described in #5113, i am closing this issue.

Please open a topic on the Forums (https://forum.phoscon.de/).

Kind regareds,

closed 2022-12-15 by Mimiix

pergolafabio, 2023-01-24:
hey @marcor78 , is it still an issue for you? how did you resolve? seeing same thing now since updating to 2.20.1

marcor78, 2023-01-24:
> hey @marcor78 , is it still an issue for you? how did you resolve? seeing same thing now since updating to 2.20.1

Hello, first I migrated back to marthoc/deconz and things got a little better, so it was probably something related to the docker container (at least this is my belief). Anyway, system was not fully reliable anymore and I decided to migrate everything to zigbee2mqtt with a new dongle (sonoff)

pergolafabio, 2023-01-24:
ah, ok, thats not an option for me ...