dresden-elektronik/deconz-rest-plugin #6614: Tuya temperature senor

quikfast, 2022-12-16:
I have a Tuya temperature sensor for testing, however I'm not able to connect this to Home Assistant/Deconz.

## Describe the bug
device visible as manufacturer and model identifier as unknown, visible from deconz but not detected in Phoscon

[image]

## Environment
Home Assistant/Deconz

 Raspberry Pi 4

Mimiix, 2022-12-16:
Hi,

this is not a bug, device is simply not supported.

Please file a device request if none is already in place. Make sure to carefully read the template to avoid issue closing. All information is in the template.

Closing this issue.

Kind regards,

Mimiix, 2022-12-16:
Reading you already found  the request that was opened by another user.

closed 2022-12-16 by Mimiix