dresden-elektronik/deconz-rest-plugin #6615: Device name

btk999, 2022-12-16:
Could you please add Tuya Smoke detector?
I already found a similar topic in issue 5143 but its not exactly the same i think and the PR 5310 is stil open.
Unfortunately im running deconz on windows only and dont have a linux machine, so i would be very happy if you could compile it for windows. @Smanar can you help here?

## Device

Product name: Haozee Tuya Smoke Alarm ZigBee Fire Detector 3C Fire Smoke Alarm
Manufacturer: _TZE200_ntcy3xu1
Model identifier: TS0601
Device type:
Other: Smoke detector

## Screenshots
[image]

### Basic
[image]

### Node Info
[image]

Mimiix, 2022-12-16:
Duplicate of : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6532

Closing this one.

Mimiix, 2022-12-16:
Closing the other one too, re-opening this one: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6236

closed 2022-12-16 by Mimiix