dresden-elektronik/deconz-rest-plugin #6617: Schneider Electric, W599001, Wiser Smoke Alarm – Battery

capristig, 2022-12-16:
## Device

- Product name: Smoke Alarm, Wiser, Battery, 10 years lifetime, white, IP20, CCT599001
- Manufacturer: Schneider Electric
- Model identifier: W599001
- Device type : Please remove all unrelated device types. 
  - Sensor

## Screenshots

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics
[image]

[image]

### Other clusters that are not mentioned above

[image]
[image]
[image]

BabaIsYou, 2023-01-10:
Hi. Have your screenshots been captured after clic on read button for each cluster ?

capristig, 2023-01-10:
> 

Hi. I cannot remember 100%, but I migh have read only once. Does that influence the results?

closed 2023-02-09 by github-actions[bot]