dresden-elektronik/deconz-rest-plugin #6618: Device name

Idaho947, 2022-12-17:
## Device

- Product name: Innr dimmer
- Manufacturer: Innr
- Model identifier: RC 250
- Device type : Switch

## Screenshots
[image]

### Basic
[image]
[image]

### Power
[image]

### Identify
[image]

### Groups
[image]

### On/Off
[image]

### Level Control
[image]

### Level color
[image]

### ZLL conmissionning
[image]

### Diagnostics
[image]

Smanar, 2022-12-17:
Hello

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "innr",
  "modelid": "RC 250",
  "product": "Battery switch",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 84000,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },        
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },    
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0008"
    }
  ]
}
```

But you will need to edit too the button_maps.json file, If I remember it's not your first battery switch ?
It's not possible on HA, because the file is not persistent.

```
        "innrRC110Map": {
            "vendor": "Innr",
            "doc": "RC 250 remote",
            "modelids": ["RC 250"],
            "map": [
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "OnOff"],
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "OnOff"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE", "0", "S_BUTTON_2", "S_BUTTON_ACTION_HOLD", "DimUp"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "DimUp"],
                [1, "0x01", "LEVEL_CONTROL", "STOP", "0", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "DimUp"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE", "1", "S_BUTTON_3", "S_BUTTON_ACTION_HOLD", "DimDown"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "1", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "DimDown"],
                [1, "0x01", "LEVEL_CONTROL", "STOP", "1", "S_BUTTON_3", "S_BUTTON_ACTION_LONG_RELEASED", "DimDown"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_TO_LEVEL_WITH_ON_OFF", "0x02", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "1"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_TO_LEVEL_WITH_ON_OFF", "0x34", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "2"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_TO_LEVEL_WITH_ON_OFF", "0x66", "S_BUTTON_6", "S_BUTTON_ACTION_SHORT_RELEASED", "3"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_TO_LEVEL_WITH_ON_OFF", "0x99", "S_BUTTON_7", "S_BUTTON_ACTION_SHORT_RELEASED", "4"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_TO_LEVEL_WITH_ON_OFF", "0xC2", "S_BUTTON_8", "S_BUTTON_ACTION_SHORT_RELEASED", "5"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_TO_LEVEL_WITH_ON_OFF", "0xFE", "S_BUTTON_9", "S_BUTTON_ACTION_SHORT_RELEASED", "6"]
            ]
        },
```

This part is taken from the RC 110, just tell me wich one are working or not to adapt it.

Idaho947, 2022-12-17:
Yes I ve got a lot of battery switch. Where can I found the bouton_map.json ?
I use deconz on a separate vm and link with jeedom so no worry about the persistent

Idaho947, 2022-12-18:
Ok found it.
Now I see the button in phoscon but nothing happen when I press it.

`                    {
    "config": {
        "battery": 10,
        "on": true,
        "reachable": true
    },
    "etag": "b3d1c9d398f573a0a7a2c0d3ea50aabf",
    "lastannounced": null,
    "lastseen": "2022-12-18T08:17Z",
    "manufacturername": "innr",
    "mode": 1,
    "modelid": "RC 250",
    "name": "RC 250",
    "state": {
        "buttonevent": null,
        "lastupdated": "none"
    },
    "swversion": "2.01.08",
    "type": "ZHASwitch",
    "uniqueid": "84:ba:20:ff:fe:a7:e2:52-01-0006"
}`

The rest api info stay like that.

Smanar, 2022-12-18:
>Now I see the button in phoscon but nothing happen when I press it.

Ok so the DDF part is done,
On deconz log with flag "info_l2" you will see some logs when using it

`DBG_Printf(DBG_INFO_L2, "[INFO] - No button handler for: %s%s, endpoint: 0x%02X, cluster: %s, command: %s, payload: %s, zclSeq: %u\n",`

the bouton_map.json is in the "/usr/share/deCONZ/devices" folder too, like DDF reserved to deconz.

Idaho947, 2022-12-18:
Yes I edit the botton_map jsn too.

Smanar, 2022-12-18:
`[INFO] - No button map` is when deconz don't found button map at all.

You are sure using the good bouton_map.json ? no typo inside ?

Idaho947, 2022-12-18:
Yes I use the right one.
I ve see that and change 100 to 250. Nothing change

Smanar, 2022-12-18:
There is some logs on star when loading the buttonmap but hard to catch.
Deconz use `"modelids": ["RC 250"],` to reconise the good buttonmap to use.

sure you haven't typo in your json ? I m using this kind of site to check them https://jsonlint.com/

Idaho947, 2022-12-18:
Json is valid for jsonlit

Can't see any typo :
[image]

Smanar, 2022-12-19:
>(normal ?)

We can set the value we want, previous one was random, the correct method is getting all command one by one using logs, take much time.

>Center : 1002

Ok for this one it's the on/off

> Up : 2002
> Down : 3002
> Left : 4002

It's button number and short released, can keep them

>Right : 9002
Heee, this one can be improved ^^

```
        "innrRC250Map": {
            "vendor": "Innr",
            "doc": "RC 250 remote",
            "modelids": ["RC 250"],
            "map": [
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "OnOff"],
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "OnOff"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "DimUp"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "1", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "DimDown"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_TO_LEVEL_WITH_ON_OFF", "0x02", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "1"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_TO_LEVEL_WITH_ON_OFF", "0xFE", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "2"]
            ]
        },
```

You haven't long press on your device ? (Can check with the old button map or reading logs when using it if you are using the new one)

Idaho947, 2022-12-19:
No just a quick click like the other.

Smanar, 2022-12-19:
Ha ?
And on the device documentation, they don't speak about a long press ?

The number you see in buttonevent are just the mapped button, if we miss one you will not see it in the buttonevent event. But it will be visible on log 
`
DBG_Printf(DBG_INFO_L2, "[INFO] - No button handler for: %s%s, endpoint: 0x%02X, cluster: %s, command: %s, payload: %s, zclSeq: %u\n"`

Idaho947, 2022-12-19:
There is an issue with the right bouton. I click one time it send 9002. I Click a second time non action and see that on logs :
[image]

I wait 30s an the right button send 9002 again. Click a second time and nothing happen.

EDIT : just see it's the same with the left bouton... Sometimes it send nothing, sometimes 9002 and sometimes 4002

Smanar, 2022-12-20:
From the documentation, some button have double clic  feature, and left and right are used for color, strange you have the "move to level with on/off" request.

As the device don't use other request when you use it (On thirs situation it's the ame request) we will use this one.

`[1, "0x01", "LEVEL_CONTROL", "MOVE_TO_LEVEL_WITH_ON_OFF", "0xFE", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "2"]`

The "0xFE" is the first byte of the payload you can see in the log.
Not possible use 0xFE as you have it for left and right

ATM the buttonmap use 0x02 for one and 0xFE for the other.
Just need to see the log and add the missing request for exemple according your last capture

`[1, "0x01", "LEVEL_CONTROL", "MOVE_TO_LEVEL_WITH_ON_OFF", "0xB2", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "2"]`

Idaho947, 2022-12-20:
I'm sorry but I don't understand what you expect from me this time

Smanar, 2022-12-20:
Just press left and right, but never double clic as it seem it's another feature (can add it later)
Sometime it will work, sometime no, if I m right you will have all the time "MOVE_TO_LEVEL_WITH_ON_OFF" but with different payload, and we need the first byte.

It seem you have 0x02, 0xFE and other thing for left.
And 0xB2, 0xFE and other thing for right.

Can't use 0xFE because you have it for all button, I hope for the "other thing" will be a not common value.

Idaho947, 2022-12-20:
Ok i've this for right :
`Ligne 1356: 17:59:19:281 [INFO] - No button handler for: RC 250, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04), payload: E50300, zclSeq: 39
Ligne 1388: 17:59:49:693 [INFO] - No button handler for: RC 250, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04), payload: B20300, zclSeq: 41
Ligne 1404: 18:00:16:394 [INFO] - No button handler for: RC 250, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04), payload: 4C0300, zclSeq: 43
	`

And this for left :
`Ligne 1506: 18:03:16:521 [INFO] - No button handler for: RC 250, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04), payload: 4C0300, zclSeq: 51
Ligne 1554: 18:04:04:204 [INFO] - No button handler for: RC 250, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04), payload: B20300, zclSeq: 53	Ligne 1573: 18:05:12:106 [INFO] - No button handler for: RC 250, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04), payload: E50300, zclSeq: 55`

Smanar, 2022-12-20:
>HS : Where can I find the log texte file in my VM. I think it'll be easier to search and copy\past the log.

No by defaut, the log are not stored (to help sd card perhaps), you need to run deconz with command line to send logs to a text file, but don't worry your captures are fine, and I just need the first bytes of pyaload.

I know the copy/paste is possible on VM, but I think I have same issue than you, I m using VirtualBox, and I m not able to use it ^^

Now the bad new.
Left and right button are using exactly same value, it 's a cycle, clockwise for one and anti clockwise for the second. And I don't see how to reconise it.

From the documentation short press cycle scene, and long press cycle color, but will be same, no possibilities to know if you are using left or right as you can have same values for both.

Idaho947, 2022-12-20:
Arf so no solution to use it ?

kasteleman, 2022-12-20:
I reading along in this issue and I know you can use the remote in 2
different modes:

1. Couple it with a bulb or more bulbs
2. Configure a attribute to send commands for key pressed

I have to look it up, had to use the second mode to implement the remote
for a other brand zigbee controller

Idaho947, 2022-12-20:
> 

Ok but in the documentation I can t see this information.

kasteleman, 2022-12-20:
That’s indeed not in the public documentation.
I’m not able to supply details today, but will get back to you tomorrow

Smanar, 2022-12-20:
>I ll try to understand how it works. I dont care if the bouton do the same thing but I have to understand when it send 4002 and 9002...

It'a a value cycle, you have E50300 > B20300 > E50300 >... > 0xFE0300

When the value is 0x02.... you have 4002, when the value is 0xFE0300 you have 9002, and nothing for others.
And you will have same value for left and right.

kasteleman, 2022-12-20:
I had a look at the initial provided screenshots and curious if you can read the FD01 cluster and the attributes within?

Idaho947, 2022-12-21:
I try but I can t.

kasteleman, 2022-12-21:
> I pair the switch in "bulb mode" (press up and middle 3s) but same issue.
>
> A strange thin, as smanar said, there is a rollover. I click on right :
> 4002 => nothing => nothing => nothing => 9002
> Then on the left :
> 9002 => nothing => nothing => nothing => 4002
>
> Except that if I press right (4002) then up (2002) and press right again I get 4002 again. But if I press right twice in a row, I get 4002 and then nothing.

FD01 is the device configuration cluster, which is a private cluster. In there you can use a attribute to set the device in Key event mode. But be aware if that is set, the device does not send ZigBee commands but only key events and you can not use as a touchline device.

But how do I read that that cluster and be able to write to an attribute?

Idaho947, 2022-12-21:
It seems to be impossible at the moment.

Smanar, 2022-12-21:
It's possible adding support for unknow cluster using this file https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/general.xml (can edit it with a text editor)

But I don't know the attribute/type value you need ? You have a link somewhere ?

>the device does not send ZigBee commands but only key events

kasteleman, 2022-12-21:
Cluster 0xFD01
ManufacturerId 0x1168
Attribute: 0x0004
Type: Uint8
Name: Key Report Mask Read/Write
Default value: 0

Smanar, 2022-12-21:
```
    <!-- INNR -->
    <cluster id="0xfc82" name="innr specific" mfcode="0x1166">
      <description>innr specific attributes.</description>
      <server>
        <attribute id="0x0006" name="Unknown" type="cstring" mfcode="0x1166" access="rw" required="m"> </attribute>
        <attribute id="0x0010" name="Unknown" type="u32" mfcode="0x1166" access="r" required="m"> </attribute>
        <attribute id="0x0011" name="Unknown" type="u32" mfcode="0x1166" access="r" required="m"> </attribute>
        <attribute id="0x0012" name="Unknown" type="u32" mfcode="0x1166" access="r" required="m"> </attribute>
        <attribute id="0x0013" name="Unknown" type="u32" mfcode="0x1166" access="rw" required="m"> </attribute>
        <attribute id="0x0014" name="Unknown" type="u32" mfcode="0x1166" access="rw" required="m"> </attribute>
        <attribute id="0x0015" name="Unknown" type="u32" mfcode="0x1166" access="rw" required="m"> </attribute>
        <attribute id="0xfffd" name="Cluster Revision" type="u16" mfcode="0x1166" access="r" required="m"> </attribute>
      </server>
      <client>
      </client>
    </cluster>
    <!-- Innr -->
    <cluster id="0xfd01" name="innr specific" mfcode="0x1168">
      <description>innr specific attributes.</description>
      <server>
        <attribute id="0x0004" name="Key Report Mask Read/Write" mfcode="0x1168" type="u8" default="0" access="rw" required="m">
        </attribute>
      </server>
      <client>
      </client>
    </cluster>
```

The first part is already in place, you can found the general.xml file in /usr/share/deCONZ/devices.

But check the manufacture number, it's on the "node info" panel, you need to have 0x1168 (it seem Innr use 2 differents)
And IDK wich ones values to use ?

Idaho947, 2022-12-21:
I ll try this tomorow.

kasteleman, 2022-12-21:
Ok, had to search for that file. It is located in /usr/share/deConz/zcl on my system.

Do I have to remove the remove the device and add again in order to get it working?

Idaho947, 2022-12-21:
I just check the node info and the manufacture code is 0x1166 :
[image]

kasteleman, 2022-12-21:
Yes I know, but for that specific cluster it differs

Smanar, 2022-12-21:
>Do I have to remove the remove the device and add again in order to get it working?

No just restrat deconz.

But my code willmake it appear only for the manufacture number 0x1168, can edit it to test.

Idaho947, 2022-12-22:
So I have to change 0x1168 by 0x1166 in your code ?

kasteleman, 2022-12-22:
According to my info and implementation for a other brand zigbee controller
you have to use 0x1168.

I tried it, but it does not work after testing and restarting deconz

Idaho947, 2022-12-22:
Ok same for me. Have you try with 0x1166 ?

kasteleman, 2022-12-22:
Yes, but that one I expected not to work.

The 0xFD01 cluster is still listed as unknown

kasteleman, 2022-12-22:
> You have set your mfcode here ?
> The file was not overwritted by deconz ?

After restart of deconz the changes are still in een place, so not
overwritten. But still in the device showing cluster 0xFD01 unknown

Smanar, 2022-12-23:
Perhaps because we have 2 time the same name for different cluster 

`name="innr specific"`

Try to use "innr specific 2"

And same for you @Idaho947 ?

Idaho947, 2022-12-23:
Same for me with 1168 and 1166.

Smanar, 2022-12-23:
And without attribute just with 

```
    <!-- Innr 2-->
    <cluster id="0xfd01" name="innr specific 2" mfcode="0x1166">
      <description>innr specific attributes 2.</description>
      <server>
      </server>
      <client>
      </client>
    </cluster>
```

Idaho947, 2022-12-23:
Ok sorry this one is working :
`  
    <cluster id="0xfc82" name="innr specific" mfcode="0x1166">
      <description>innr specific attributes.</description>
      <server>
        <attribute id="0x0006" name="Unknown" type="cstring" mfcode="0x1166" access="rw" required="m"> </attribute>
        <attribute id="0x0010" name="Unknown" type="u32" mfcode="0x1166" access="r" required="m"> </attribute>
        <attribute id="0x0011" name="Unknown" type="u32" mfcode="0x1166" access="r" required="m"> </attribute>
        <attribute id="0x0012" name="Unknown" type="u32" mfcode="0x1166" access="r" required="m"> </attribute>
        <attribute id="0x0013" name="Unknown" type="u32" mfcode="0x1166" access="rw" required="m"> </attribute>
        <attribute id="0x0014" name="Unknown" type="u32" mfcode="0x1166" access="rw" required="m"> </attribute>
        <attribute id="0x0015" name="Unknown" type="u32" mfcode="0x1166" access="rw" required="m"> </attribute>
        <attribute id="0xfffd" name="Cluster Revision" type="u16" mfcode="0x1166" access="r" required="m"> </attribute>
      </server>
      <client>
      </client>
    </cluster>
    
    <cluster id="0xfd01" name="innr specific" mfcode="0x1166">
      <description>innr specific attributes.</description>
      <server>
        <attribute id="0x0004" name="Key Report Mask Read/Write" mfcode="0x1166" type="u8" default="0" access="rw" required="m">
        </attribute>
      </server>
      <client>
      </client>
    </cluster>`

kasteleman, 2022-12-23:
Can you post a screenshot from the gui how it looks like for you?

Idaho947, 2022-12-23:
It's my first screenshot on my mast post.

kasteleman, 2022-12-23:
Sorry, missed that on my phone.

I’ll check tomorrow if I can reproduce your result on my remote.

kasteleman, 2022-12-24:
Finally got it right

Now the next step:

How to listen to the reports.

In the tech it states:

Commands generated:

ID: 0x12
Description: Custom Key Event Report

This command is used to send key events

With 4 payloads

Byte 0 is group id, but is reserved
Byte 1 is key id
Byte 2 is key event
Byte 4 is Expand field (not used)

Key id table:

1 Button none value 0x00
2 Button reset value 0x01
3 Button ok value 0x02
4 Button up value 0x03
5 Button down value 0x04
6 Button left value 0x05
7 Button right value 0x06

Key event table:

1 button press value 0x01
2 button long press value 0x02
3. button double press value 0x03
4. button short release value 0x04
5. button long release value 0x05
6. Button release value 0x06

Smanar, 2022-12-24:
But wih one value we need to put un the attribute 0x0004 to enable this mode ?
0 is the defaut value

kasteleman, 2022-12-24:
I used 63 to get all key events

Idaho947, 2022-12-24:
And now you see all the event ?
What software do you use it with deconz ?

Smanar, 2022-12-24:
It's possible with deconz, double clic on the attribute, then on the new popup you will be able to change the value

Idaho947, 2022-12-24:
I misspoke. I wanted to know which home automation software he uses with deconz (HA, jeedom, domotiz...).

For this switch i ve juste have to change the 0 for 63 and it works ?

kasteleman, 2022-12-24:
I use HA with deconz, but as I said before, next thing is to filter the key
reports after setting the attribute and convert it in a way that it is
useful. I’m new to the ddf part etc….so need help with that part

Smanar, 2022-12-24:
It's 63 decimal, not hexa ?

kasteleman, 2022-12-24:
The attribute is uint8 so in hex you should have 00111111 that is 63 in dec
and 0x3F in hex

kasteleman, 2022-12-24:
Sorry skip that first hex:
Attribute is uint8 you should set the first 6 so that would be 00111111
00111111 is 63 in dec and 0x3F in hex

Smanar, 2022-12-24:
I think the value will be in decimal in deconz (hexa is possible but need to be specified in the xml)

Idaho947, 2022-12-28:
Ok guys, thnaks for your work. I don t know if and how can I help you ?

Smanar, 2022-12-29:
Try to set the 63 value (and if the value stay), then with luck something will change in logs (the sames ones you show in your capture)

[78 more replies]