dresden-elektronik/deconz-rest-plugin #6619: Don't manage to get direct binding to work

ThorstenDtk, 2022-12-19:
I manage ~50 hue lamps with DeCONZ (on a raspberry). Most of them are controlled through iOBroker via Javascript from Homematic switches, some by hue dimmer switches and IKEA remotes. Everything is running fine for a couple of years now as long as the Raspis with iOBroker and DeCONZ are running. 
**But I want to be able to control at least a couple of lamps on each floor even if DeCONZ is down.** The idea is to create direct bindings between lamps/groups of lamps and a couple of additional zigbee remotes. 

This is how I tried:

Dagged and dropped the on/off node of the remote on source. Dragged and dropped the on/off node of the light to destination:

[image]

Result is success, but when I press the on/off buttons of the remote nothing happens.
Created additional binding for level control, problem persists. Even after waiting for an hour.

Then I removed the bindings and created a binding between the same remote and the group containing the light:

[image]

[image]

But nothing happens when using the remote. I tried the same with a hue remote, but that also doesn't work.

Is this supposed to work? And if yes - what am I doing wrong?

Mimiix, 2022-12-19:
Hi,

As this is not a bug report as described in #5113 i am closing this issue 

Please ask the question on the forums.

Kind regards

closed 2022-12-19 by Mimiix

ThorstenDtk, 2022-12-19:
> Hi,
> 
> As this is not a bug report as described in #5113 i am closing this issue
> 
> Please ask the question on the forums.
> 
> Kind regards

If it's only for bugs and device requests then you should remove the following link:

[image]

Mimiix, 2022-12-19:
I wish we could.

Mimiix, 2022-12-19:
I just found out we could! Thanks for the heads-up :)