dresden-elektronik/deconz-rest-plugin #6628: Frient Smart Plug Mini Type G (British) SPLZB-134

ReX1983, 2022-12-27:
## Device

- Product name: Frient Smart Plug Mini Type G (British)
- Manufacturer: friend A/S
- Model identifier: SPLZB-134
- Device type : Please remove all unrelated device types. 
  - Switch

The device is correctly paired and visible in both deCONZ and Phoscon (v2.19.3) but Phoscon lists it as under Lights, instead of Switches. It is exposed to Home Assistant (2022.12.8) but there is no meter / consumption attribute in Home Assistant, nor in Phoscon. I can see that deCONZ instead shows the meter capabilities (see cluster screenshots). 

Could you add support to expose the meter readings to Home Assistant as entity attributes?

## Screenshots

### Basic

[image]

### Node Info panel

[image]

### Simple Metering

[image]

[image]

[image]

Smanar, 2022-12-27:
Hello, can you try this DDF ?

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "friend A/S",
  "modelid": "SPLZB-134",
  "vendor": "Frient",
  "product": "Smart plug",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power"
        },
        {
          "name": "state/voltage"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 10,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 10,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 10,
          "max": 300,
          "change": "0x00000064"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 10,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

ReX1983, 2022-12-27:
I am trying to understand how to edit or create files in my installation (e.g. from [here](https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839/5)). deCONZ is installed as a Home Assistant plugin and I can't find a way to enter the docker image.

Is there an easy way?

Smanar, 2022-12-27:
The command on your link haven't worked ?

ReX1983, 2022-12-27:
I am having the same issue as [the user in this post](https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839/5). I am connected via ssh to the machine, I understand I might need to change user (You need to be logged in as user hassio) but it is not clear to me how to do so.

Is there a way to paste your DDF in the deCONZ UI exposed via VNC? I can see the editor but it doesn't let me type free text.

Smanar, 2022-12-27:
I don't have HA so I can't explain better. Perhaps https://community.home-assistant.io/t/hassos-docker-command-not-found/81477/4

You can use the editor and make the DDF yourelf from scratch up to have the same result.
And no you can't use free text https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

ReX1983, 2022-12-28:
> https://developers.home-assistant.io/docs/operating-system/debugging/ 

This requires physical access to the host machine (to plug in the USB stick), which is not very convenient. 

I might give it a go using the DDF editor instead, could you give me any hint on how to replicate your DDF using the UI?

Smanar, 2022-12-28:
>This section is not for end users. End users should use the [SSH add-on](https://github.com/home-assistant/hassio-addons/tree/master/ssh) to SSH into Home Assistant

And using this addon ?

Here you have the procedure to create DDF https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

ReX1983, 2022-12-29:
Unfortunately, the addon enables SSH to the Home Assistant docker container, but to get access to the deCONZ container you need to connect an USB drive to the server and load a fresh private key (this will give SSH access to the host). For me it is not an option now - but thanks for checking :-).

I tried the DDF construction but as you anticipated I had some troubles with the binding. 

Anyway, even without access to the deCONZ container, it was possible to navigate to the deCONZ/devices/develco folder and load and save as an existing DDF directly in the DDF editor.

I cloned the configuration of the same frient plug (the German version [here](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/develco/splzb-131_smart_plug.json)), updating the fields from SPLZB-131 to SPLZB-134. Saved and hot reloaded. When I select the plug node and do "Edit DDF" it actually shows the new DDF.

But in Phoscon (and HA) nothing has changed: in Phoscon, I don't see any power consumption / energy info under Help / API.

Am I supposed to do something else besides hot reload or it is just the DDF I have used wrong?

Smanar, 2022-12-29:
https://github.com/hassio-addons/addon-ssh/blob/main/ssh/DOCS.md

I see nothing about  USB drive !

>Am I supposed to do something else besides hot reload or it is just the DDF I have used wrong?

You have an option "Hot Reload" yes, that work immediatly, but for new sensors be added in the API deconz need to be in permit join ( add new light/sensor in phoscon). But HA are using docker, nothing is persistent, if you restart the docker you will lose all your modifications.
Dockers are not easy at all to use, better to find the problem with your SSH, you will need it another day.

ReX1983, 2022-12-29:
> I see nothing about USB drive !

Unfortunately, that is the plug-in that allows connection to the HA container. In order to access the deCONZ container I need SSH into the host machine (as explained [here](https://developers.home-assistant.io/docs/operating-system/debugging/)). How annoying!

> To choose DDF deconz check for model ID + Manufacture name.

Okay, so I assume that the correct DDF is picked up because these match between the file and the basic cluster and when I select edit DDF I go into my DDF. 

>  for new sensors be added in the API deconz need to be in permit join ( add new light/sensor in phoscon).

I tried to delete and pair again the device (both activities using the Phoscon UI) but I still don't get any new sensors.
The new device still shows the correct (my new) DDF in deconz.

What am I missing?

Was [this DDF](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/develco/splzb-131_smart_plug.json) the wrong starting point?

Smanar, 2022-12-29:
Nope your DDF seem fine, it's a standard one.
You just need to edit the manufacture name and the model id.
Then save your change.
And re-include the device after that is a good method too.

Have you take a look direclty in the API, perhaps it's HA that have skipped them, in phoscon / help / API Information / sensors

Smanar, 2022-12-29:
Oups, way  a little, there is a problem in the DDF

ReX1983, 2022-12-29:
There is something odd happening in my installation:
- in deCONZ Node List panel, the plug is not listed. Any other device is there
- in Phoscon all the devices (except the power consumptions we are trying to add here) are correctly listed under Devices (Lights, Switches and Sensors) but when I go to Help > API Information, I can see 6 sensors relative to plugs I don't have any more. Is it possible that the Sensors API config file/section of Phoscon is somehow compromised and it can't remove old sensors as well as add new sensors?

> It seem this DDF use the endpoint 0x02, and generaly device use 0x01

I can see that all the clusters of my plug are under 0x02 (0x01 has only two unknown clusters without any attribute inside), if I understand what you mean:

[image]

Smanar, 2022-12-29:
>I can see that all the clusters of my plug are under 0x02 (0x01 has only two unknown clusters without any attribute inside), if I understand what you mean:

Yeah, sorry, so you are fine.

Phoscon don't support power measurement sensor

>I can see 6 sensors relative to plugs I don't have any more

When you delete a device, all assciated sensors are deleted too.
You can compare the MAC adresss to be sure it's different devices

ReX1983, 2022-12-29:
Found it.

Playing with deconz, I saw the following:

[image]

If I understand the screenshot correctly, it means that only gold DDFs are used.
I updated the DDF to gold, hot reloaded and now I can see all the info in both the Phoscon APIs (under help) and Home Assistant! I am going to check the values over the next few days to confirm that the DDF is correct, if so I think you can clone the DDF from the 131 model. 

Am I supposed to leave the setting to use DDF gold only? 
I have many other devices in my network with DDFs in state either Bronze or Draft. How come they work?

> When you delete a device, all assciated sensors are deleted too.
> You can compare the MAC adresss to be sure it's different devices

No, in my case there is an issue, look here:

[image]

The 6 sensor with MAC addresses ending with B4, 7F, 76 are associated to devices that I have removed. Such devices were Xiaomi plugs and they are not listed any more as nodes in either Phoscon and deCONZ, but their sensors are somehow still reported by the API.

Smanar, 2022-12-30:
>Am I supposed to leave the setting to use DDF gold only?
I have many other devices in my network with DDFs in state either Bronze or Draft. How come they work?

Not normal ^^
And I have check the original DDF again, the status was "Gold" https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/develco/splzb-131_smart_plug.json

You have defaut setting, Draft are never used, Bronze are not used with your setting.

>No, in my case there is an issue, look here:

Yes effectively, and not possible to remove them using phoscon, not possible HA ?

ReX1983, 2023-01-02:
So, the power measurement is correct: it reports the correct watts. But the current attribute is wrong: it is 4 times the power, so clearly wrong. I had a device of 2000 watt connected, 2000 watt reported and 8000 as current!

> Not normal ^^
> And I have check the original DDF again, the status was "Gold

Yes, for SPLZB-131 the status was gold also on my end.

But I have in my network (this is what I see by selecting each node and looking at the content of the DDF editor):
- Ikea Tradfri Signal Repeater > Draft
- GLEDOPTO GL-C-006 > Draft
- Echostar Bell > Draft
- Ikea Tradfri bulb SU10 WS 400lm > Bronze
- Ikea Tradfri bulb E14 W op/ch 400lm > Draft
- Ikea Trasfri bulb E27 CWS opal 600lm > Draft
- GLEDOPTO GL-C-006P > Draft

Not a single device has the DDF in gold mode!

I am on deCONZ 2.19.3 installed as HA addon and I have never modified DDFs before this one.

How is this possible? How are these devices working if my setting is to accept only Gold DDFs?

Smanar, 2023-01-03:
Ikea are Bronze, not draft
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/ikea/e14_ws_opal_400lm_light.json
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/ikea/e27_cws_opal_600lm_light.json

But it mean they are not used with DDF core but legacy one, probably here for future or if you have issue with them.

>So, the power measurement is correct: it reports the correct watts. But the current attribute is wrong: it is 4 times the power, so clearly wrong. I had a device of 2000 watt connected, 2000 watt reported and 8000 as current!

The current is in mA, so the value seem good
P = U * I = 240 * 8000 / 1000 = 1920 W (around 2 000w)

ReX1983, 2023-01-07:
> The current is in mA, so the value seem good

That's great! 

I can confirm the plug continues working correctly, also acts as repeater. 

Is it possible to include the DDF in the next release?

Smanar, 2023-01-07:
Have added your model to previous one, if you want to check https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6651

ReX1983, 2023-01-07:
I suppose I should wait for the next deConz release as HA addon, as I still can't upload files in the container.

When is it planned?

Smanar, 2023-01-07:
On deconz realy soon, but idk for HA.

closed 2023-01-07 by manup