dresden-elektronik/deconz-rest-plugin #6629: Danfoss Ally Thermostat 014G2460

billy-boy, 2022-12-29:
Hello,

I have some Danfoss Ally Radiator Thermostat (014G2460) which are Zigbee Certified Products and also a Danfoss Ally Room Sensor (014G2480).

The Room Sensor is paired and working fine. The Radiator Thermostates are not pairing. Not in installed condition, nor laying beside the Raspberry. The Thermostates have one button, I tried all variants of pressing and factory reseted them multiple times.

I am using deconz REST API (gui-less) Version 2.19.03 / 19.9.2022 with Firmware 26780700.

Is there any option to get the Thermostates paired?

Regards
Markus

billy-boy, 2022-12-29:
As I saw in debug log a eTRV0103. I copied the etrv0100_thermostat.json from /usr/share/deCONZ/devices/danfoss to a etrv0103_thermostat.json and changed the modelid to eTRV0103. Now the Thermostate seems to work.

Mimiix, 2022-12-29:
Hi

As this is not an issue as described in #5113 i am closing this issue.

For any user questions, please use the forums.

Kind regards,

closed 2022-12-29 by Mimiix