dresden-elektronik/deconz-rest-plugin #6631: New clusters for Tuya LCD Multipurpose sensor (_TZ3000_qaaysllp)

rugarci, 2022-12-29:
Hi, 

This device was already partially supported but only for the lux measurements

[image]

After connecting the sensor to a Tuya gateway and coming back to deconz, new clusters for temperature and humidity appeared. However they are not displayed in web interface nor REST API

[image]

## Device

- Product name: Tuya  Zigbee Temperature Humidity Sensor And Lux Detector With LCD Screen
- Manufacturer: _TZ3000_qaaysllp
- Model identifier:  TS0201
- Device type:
  - Sensor

## Screenshots

### Device Temperature

[image]

### Device Humidity

[image]

Thanks!

Smanar, 2022-12-29:
Heeeee ???

I m searching the device _TZ3000_qaaysllp And I found nothing, in code and in DDF, how it can be reconised ......

rugarci, 2022-12-29:
No custom DDF, just using latest deconzcommunity/deconz:beta docker image

The temperature and humidity are correctly updated and reflects the same values as in the LCD

rugarci, 2022-12-29:
Other users were able to recognize it: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4790

Smanar, 2022-12-30:
Ok so from this link it seem we need to use the tuya unlock sequence to use this device, ad this is not possible yet with deconz.

closed 2023-01-28 by github-actions[bot]