dresden-elektronik/deconz-rest-plugin #6632: Luminea Home Control Tür- und Fensteralarm für ZigBee-Gateway, App, Sprachsteuerung

reichemn, 2022-12-29:
## Device

- Product name: Luminea Home Control Tür- und Fensteralarm für ZigBee-Gateway, App, Sprachsteuerung
- Manufacturer: _TZ3000_4ugnzsli
- Model identifier: TS0203
- Device type :
  - Sensor

[image]

## Screenshots

### Basic
[image]

### Identify
[image]

### IAS Zone
[image]

### OTAU
[image]

### Power Configuration
[image]

BabaIsYou, 2023-01-11:
May we can start with this [DDF](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_n2egfsli_door_sensor.json) after chnaging lines 2 and 3 by these : 
```
  "manufacturername": ["_TZ3000_n2egfsli", "_TZ3000_oxslv1c9", "_TZ3000_7tbsruql", "_TZ3000_7d8yme6f", "_TZ3000_rgchmad8", "_TZ3000_bzxlofth","_TZ3000_4ugnzsli"],
  "modelid": ["TS0203", "TS0203", "TS0203", "TS0203", "TS0203", "TS0203", "TS0203"],
```

reichemn, 2023-01-21:
I tested the mentioned modified DDF. The device is now recognized as door sensor in phoscon. Thank you!

wunni1, 2023-01-24:
Hello, I want to use the same sensors. Are they still working with phoscon? Or should I do an update? Current version of phoscon (runs in HomeAssistant): 2.20.01 / 19.9.2022; Firmware: 26720700

BabaIsYou, 2023-01-24:
@wunni1 you have to do the same as @reichemn did until the #6693 PR will be added in a next release.

wunni1, 2023-01-25:
How can I modify the DDF in Home Assistant. I was able to start the DDF Editor. But I can't add the new file. When will the next release published?

BabaIsYou, 2023-01-26:
> How can I modify the DDF in Home Assistant.

Think it could help [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6697#issuecomment-1404069986](url)

closed 2023-02-24 by github-actions[bot]

wunni1, 2023-03-06:
Is this Sensor (TZ3000_4ugnzsli) now implemented  in the last release 2.21.0?