dresden-elektronik/deconz-rest-plugin #6634: Tuya Smart Zigbee Radiator Actuator Trv

jossiee, 2022-12-29:
## Device

- Product name: Tuya Smart Zigbee Radiator Actuator Trv
- Manufacturer: _TZE200_hue3yfsn
- Model identifier: TS0601
- product page: [https://nl.aliexpress.com/item/1005003444955449.html](https://nl.aliexpress.com/item/1005003444955449.html)
- Device type : Please remove all unrelated device types. 
  - Thermostat

## Screenshots

### Basic
[image]

### Identify

### Alarms

### Device Temperature

### Groups
[image]

### Scenes
[image]

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]

[image]

basp1964, 2023-01-16:
Hallo,
hätte die gleiche Geräteanforderung

Smanar, 2023-01-16:
Hello, I don't see the cluster list, but I presume this device use the tuya clster and not the thermostat one ?
If yes, I need a tester https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6674

Tuya TRV need this PR to work

jossiee, 2023-01-18:
I took a screenshot of al the clusters that where available. Not sure if i can help with testing? What knowledge is needed?

Smanar, 2023-01-18:
Not a knowledge , but a hardware ^^
You need a linux machine with full/real OS to compile the code.

CHTHSCH, 2023-01-19:
Is this compatible with brennenstuhl®Connect Zigbee Heizkörperthermostat HT CZ 01?

Smanar, 2023-01-19:
Need the model id.
But if you are able to compile the code to test, I can try to make a DDF for the device if missing.

jossiee, 2023-01-19:
yeah i will try!
i have some spare hardware for a test setup.

will take some time before i got some time for it.

nilleiz, 2023-10-17:
I have this valve thermostat.
It seems that it isn't added yet.
Have a full Linux machine and can help.

Mimiix, 2023-11-15:
Asked @Smanar  to check in  :)

Smanar, 2023-11-15:
Here you have a starting point

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_sur6q7ko", "_TZE200_hue3yfsn", "_TZE200_e9ba97vf", "_TZE200_husqqvux", "_TZE200_lnbfnyxd", "_TZE200_lllliz3p", "_TZE200_mudxchsu", "_TZE200_7yoranx2", "_TZE200_kds0pmmv"],
  "modelid": ["TS0601", "TS0601", "TS0601", "TS0601", "TS0601", "TS0601", "TS0601", "TS0601", "TS0601"],
  "vendor": "Tuya",
  "product": "Tuya TRV",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "meta": {
        "values": {
          "config/mode": {"auto": 0, "heat": 1, "off": 3}
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/lowbattery",
          "parse": {"fn": "tuya", "dpid": 35, "eval": "Item.val = Attr.val != 0"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/tuya_unlock"
        },
        {
          "name": "config/heatsetpoint",
          "parse": {"fn": "tuya", "dpid": 16, "eval": "Item.val = Attr.val * 10;"},
          "write": {"fn": "tuya", "dpid": 16, "dt": "0x2b", "eval": "Item.val / 10;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid": 40, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 40, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/mode",
          "values": [
              ["auto", 0], ["heat", 1], ["off", 3]
          ],
          "parse": {"fn": "tuya", "dpid": 2, "eval": "if (Attr.val == 0) { Item.val = 'auto' } else if (Attr.val == 1) { Item.val = 'heat' } else { Item.val = 'off' }"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x30", "eval": "if (Item.val == 'auto') { 0 } else if (Item.val == 'heat') { 1 } else { 3 }"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/windowopen_set",
          "parse": {"fn": "tuya", "dpid": 8, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 8, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 24, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        }
      ]
    }
  ]
}
```

You will just need probably some adjsutement for values convertion (*10 or /10 for exemple) and not sure how will work the mode.

Smanar, 2023-11-16:
Bad luck ^^. It's read only.
It depend of OS, you just need to put a text file called whatyouwant.json with this contain in a "devices" folder.
A little more complex on the official docker and worse on HA.

What is your OS ^^ ?

rensbr, 2023-11-16:
> Bad luck :). It's read only. It depend of OS, you just need to put a text file called whatyouwant.json with this contain in a "devices" folder. A little more complex on the official docker and worse on HA.
> 
> What is your OS ^^ ?

I'm using Debian as OS on a ChromeBox and I'm running everything inside docker. And furthermore I'm using Domoticz.

Smanar, 2023-11-16:
So you are using the official deconz docker ?
You can use the opt/deCONZ/devices/tuya folder for exemple https://github.com/deconz-community/deconz-docker/issues/163

Just put a test file inside with the DDF contain, rename the file in *.json
And restart the deconz docker, and next restart deconz will be able to use the DDF.

rensbr, 2023-11-17:
> So you are using the official deconz docker ? You can use the opt/deCONZ/devices/tuya folder for exemple [deconz-community/deconz-docker#163](https://github.com/deconz-community/deconz-docker/issues/163)
> 
> Just put a text file inside with the DDF contain, rename the file in *.json And restart the deconz docker, and next restart deconz will be able to use the DDF.

That seems to work! Thanks! I have added your file to the designed location and now it is visible in Phoscon and Domoticz!

I can see the temprature and set temperature and there is a mode switcher. But the mode switcher does not seem to work very well. When I press 'Off' it goes to holiday mode for a few seconds and then it goes to Auto mode. It does not really matter to be, because setting the temperature is the only thing I need, but it would be nice to have it working for 100%.

Can I do something to make it work?

And also there are some features like Window Detection and that seems not possible to set or do anything with it.

Smanar, 2023-11-18:
>And also there are some features like Window Detection and that seems not possible to set or do anything with it.

You have it in config/windowopen_set

There is a dpid I want to test about mode, on the DDF can you edit 
```
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid": 40, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 40, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
```

And use the dpid 107 instead of 40 ?
After that the config/locked will act a strange dpid that concern mode/heating, perhaps usefull for the mode problem.

rensbr, 2023-11-21:
> "name": "config/locked",

I have done the last part, checked info_l2 and see what happens if I switch modes. But this happens so fast that it is really hard for me to see what happens. Can you do anything with the below jiberish:

[code]
08:49:59:810 INSERT INTO resource_items (sub_device_id,item,value,source,timestamp) SELECT id, 'config/mode', 'off', 'dev', 1700552999 FROM sub_devices WHERE uniqueid = 'cc:86:ec:ff:fe:ca:a4:37-01-ef00'
08:49:59:819 Websocket 172.17.0.1:57812 send message: {"attr":{"id":"68","lastannounced":null,"lastseen":"2023-11-21T07:49Z","manufacturername":"_TZE200_hue3yfsn","modelid":"TS0601","name":"Thermostat 68","swversion":null,"type":"ZHAThermostat","uniqueid":"cc:86:ec:ff:fe:ca:a4:37-01-ef00"},"e":"changed","id":"68","r":"sensors","t":"event","uniqueid":"cc:86:ec:ff:fe:ca:a4:37-01-ef00"} (ret = 332)
08:49:59:820 Websocket 172.17.0.1:57812 send message: {"config":{"heatsetpoint":2250,"locked":false,"mode":"off","on":true,"reachable":true,"windowopen_set":false},"e":"changed","id":"68","r":"sensors","t":"event","uniqueid":"cc:86:ec:ff:fe:ca:a4:37-01-ef00"} (ret = 205)
08:49:59:937 TY_DATA_RESPONSE: seq 23, dpid: 0x10, type: 0x02, length: 4, val: 170
08:49:59:938 TY_DATA_RESPONSE: seq 23, dpid: 0x10, type: 0x02, length: 4, val: 170
08:49:59:939 TY_DATA_RESPONSE: seq 23, dpid: 0x10, type: 0x02, length: 4, val: 170
08:49:59:939 TY_DATA_RESPONSE: seq 23, dpid: 0x10, type: 0x02, length: 4, val: 170
08:49:59:939 TY_DATA_RESPONSE: seq 23, dpid: 0x10, type: 0x02, length: 4, val: 170
08:49:59:940 TY_DATA_RESPONSE: seq 23, dpid: 0x10, type: 0x02, length: 4, val: 170
08:49:59:941 INSERT INTO resource_items (sub_device_id,item,value,source,timestamp) SELECT id, 'config/heatsetpoint', '1700', 'dev', 1700552999 FROM sub_devices WHERE uniqueid = 'cc:86:ec:ff:fe:ca:a4:37-01-ef00'
08:49:59:949 Websocket 172.17.0.1:57812 send message: {"config":{"heatsetpoint":1700,"locked":false,"mode":"off","on":true,"reachable":true,"windowopen_set":false},"e":"changed","id":"68","r":"sensors","t":"event","uniqueid":"cc:86:ec:ff:fe:ca:a4:37-01-ef00"} (ret = 205)
08:50:00:022 poll node f0:d1:b8:00:00:1c:8e:a7-01
08:50:00:022 Poll light node Lamp WC
08:50:00:073 Poll APS request to 0xF0D1B800001C8EA7 cluster: 0x0006 dropped, values are fresh enough
08:50:00:176 Idle timer triggered
08:50:00:177 Force read attributes for ZHAThermostat SensorNode Thermostat 68
08:50:00:177 Force binding of attribute reporting for node Thermostat 68
08:50:01:101 poll node 84:2e:14:ff:fe:3d:ac:f2-01-0702
08:50:01:102 Poll ZHAConsumption sensor node Consumption 54
08:50:01:363 Poll APS request to 0x842E14FFFE3DACF2 cluster: 0x0702 dropped, values are fresh enough
08:50:01:383 TY_DATA_RESPONSE: seq 23, dpid: 0x02, type: 0x04, length: 1, val: 0
08:50:01:384 TY_DATA_RESPONSE: seq 23, dpid: 0x02, type: 0x04, length: 1, val: 0
08:50:01:384 TY_DATA_RESPONSE: seq 23, dpid: 0x02, type: 0x04, length: 1, val: 0
08:50:01:385 TY_DATA_RESPONSE: seq 23, dpid: 0x02, type: 0x04, length: 1, val: 0
08:50:01:385 TY_DATA_RESPONSE: seq 23, dpid: 0x02, type: 0x04, length: 1, val: 0
08:50:01:385 TY_DATA_RESPONSE: seq 23, dpid: 0x02, type: 0x04, length: 1, val: 0
08:50:01:386 INSERT INTO resource_items (sub_device_id,item,value,source,timestamp) SELECT id, 'config/mode', 'auto', 'dev', 1700553001 FROM sub_devices WHERE uniqueid = 'cc:86:ec:ff:fe:ca:a4:37-01-ef00'
08:50:01:394 Websocket 172.17.0.1:57812 send message: {"config":{"heatsetpoint":1700,"locked":false,"mode":"auto","on":true,"reachable":true,"windowopen_set":false},"e":"changed","id":"68","r":"sensors","t":"event","uniqueid":"cc:86:ec:ff:fe:ca:a4:37-01-ef00"} (ret = 206)
08:50:01:500 TY_DATA_RESPONSE: seq 24, dpid: 0x10, type: 0x02, length: 4, val: 210
08:50:01:501 TY_DATA_RESPONSE: seq 24, dpid: 0x10, type: 0x02, length: 4, val: 210
08:50:01:501 TY_DATA_RESPONSE: seq 24, dpid: 0x10, type: 0x02, length: 4, val: 210
08:50:01:502 TY_DATA_RESPONSE: seq 24, dpid: 0x10, type: 0x02, length: 4, val: 210
08:50:01:502 TY_DATA_RESPONSE: seq 24, dpid: 0x10, type: 0x02, length: 4, val: 210
08:50:01:503 TY_DATA_RESPONSE: seq 24, dpid: 0x10, type: 0x02, length: 4, val: 210
08:50:01:503 INSERT INTO resource_items (sub_device_id,item,value,source,timestamp) SELECT id, 'config/heatsetpoint', '2100', 'dev', 1700553001 FROM sub_devices WHERE uniqueid = 'cc:86:ec:ff:fe:ca:a4:37-01-ef00'
08:50:01:517 Websocket 172.17.0.1:57812 send message: {"config":{"heatsetpoint":2100,"locked":false,"mode":"auto","on":true,"reachable":true,"windowopen_set":false},"e":"changed","id":"68","r":"sensors","t":"event","uniqueid":"cc:86:ec:ff:fe:ca:a4:37-01-ef00"} (ret = 206)
08:50:02:099 poll node 84:2e:14:ff:fe:35:a2:0d-01-0702
08:50:02:100 Poll ZHAConsumption sensor node Consumption 26
08:50:02:175 Idle timer triggered
08:50:02:357 Poll APS request to 0x842E14FFFE35A20D cluster: 0x0702 dropped, values are fresh enough
08:50:02:408 TY_DATA_RESPONSE: seq 24, dpid: 0x02, type: 0x04, length: 1, val: 1
08:50:02:409 TY_DATA_RESPONSE: seq 24, dpid: 0x02, type: 0x04, length: 1, val: 1
08:50:02:409 TY_DATA_RESPONSE: seq 24, dpid: 0x02, type: 0x04, length: 1, val: 1
08:50:02:410 TY_DATA_RESPONSE: seq 24, dpid: 0x02, type: 0x04, length: 1, val: 1
08:50:02:410 TY_DATA_RESPONSE: seq 24, dpid: 0x02, type: 0x04, length: 1, val: 1
08:50:02:410 TY_DATA_RESPONSE: seq 24, dpid: 0x02, type: 0x04, length: 1, val: 1
[... cut ...]
08:50:05:694 TY_DATA_RESPONSE: seq 28, dpid: 0x10, type: 0x02, length: 4, val: 225
08:50:05:694 INSERT INTO resource_items (sub_device_id,item,value,source,timestamp) SELECT id, 'config/heatsetpoint', '2250', 'dev', 1700553005 FROM sub_devices WHERE uniqueid = 'cc:86:ec:ff:fe:ca:a4:37-01-ef00'
08:50:05:714 Websocket 172.17.0.1:57812 send message: {"config":{"heatsetpoint":2250,"locked":false,"mode":"heat","on":true,"reachable":true,"windowopen_set":false},"e":"changed","id":"68","r":"sensors","t":"event","uniqueid":"cc:86:ec:ff:fe:ca:a4:37-01-ef00"} (ret = 206)[/code]

Smanar, 2023-11-21:
`08:50:03:372 TY_DATA_RESPONSE: seq 25, dpid: 0x02, type: 0x04, length: 1, val: 3`

You use the dpid 0x02 to set the mode 0x03 > off

But immediatly after

`08:50:04:557 TY_DATA_RESPONSE: seq 26, dpid: 0x02, type: 0x04, length: 1, val: 0`

The mode return to 0x00 > auto

So if you have done the procedure manually, it mean it's the same reaction ? The device stay in "off" mode during your test ?
We can see to the temperature moving from 22.5 > 21 > 17 °C ?

rensbr, 2023-11-22:
> `08:50:03:372 TY_DATA_RESPONSE: seq 25, dpid: 0x02, type: 0x04, length: 1, val: 3`
>
> You use the dpid 0x02 to set the mode 0x03 > off
>
> But immediatly after
>
> `08:50:04:557 TY_DATA_RESPONSE: seq 26, [...]

I switched between the modes 'Manual' -> 'Holiday' -> 'Auto' -> 'Manual'.

The switch between Holiday and auto happens automaticly.

closed 2023-12-22 by github-actions[bot]

Feldi1606, 2024-04-04:
I also got the thermostat and put it on with the DDF from above. Unfortunately the battery status is not displayed correctly for me. And I also have the problem that vacation mode automatically switches back to Auto after a few seconds. Is there still a solution?

Smanar, 2024-04-05:
From the DDF, this device don't have battery level but a low battery warning. You have the reverse state ?
For the auto return, it's something native on the device.
I need you make the procedure using the device itself (and hopping for the device don't thave the "auto return"), get logs as described previously for I make something to mimic them.

Feldi1606, 2024-04-05:
Hello Smanar,

the battery level is schown as low.( New device with new battery, an the device itself is not show an empty battery.) So I think the battery status ist noch shown correctly.
Is it possible to send a weekly program to the TRV? And how can I pair the Door Sensor and the Multisensor with the TRV? 
  
I started the LOG and Switch the TRV from Manual to Holiday mode. After a few seconds the TRV Switches to Home mode with 17°C . Maybe it is normal. Is this the LOG sequens you want to see? 

> 21:58:58:090 TY_DATA_RESPONSE: seq 166, dpid: 0x02, type: 0x04, length: 1, val: 3
> 21:58:58:091 TY_DATA_RESPONSE: seq 166, dpid: 0x02, type: 0x04, length: 1, val: 3
> 21:58:58:092 TY_DATA_RESPONSE: seq 166, dpid: 0x02, type: 0x04, length: 1, val: 3
> 21:58:58:093 TY_DATA_RESPONSE: seq 166, dpid: 0x02, type: 0x04, length: 1, val: 3
> 21:58:58:094 TY_DATA_RESPONSE: seq 166, dpid: 0x02, type: 0x04, length: 1, val: 3
> 21:58:58:095 TY_DATA_RESPONSE: seq 166, dpid: 0x02, type: 0x04, length: 1, val: 3
> 21:58:58:164 Websocket 127.0.0.1:50218 send message: {"config":{"heatsetpoint":2000,"locked":false,"mode":"off","on":true,"reachable":true,"windowopen_set":false},"e":"changed","id":"14","r":"sensors","t":"event","uniqueid":"e8:e0:7e:ff:fe:ce:3d:a7-01-ef00"} (ret = 2130660672)
> 21:58:58:320 [Master] read param with arg 0x19
> 21:58:58:440 CTRL read param resp id: 0x26, status: 0x04
> 21:58:58:540 TY_DATA_RESPONSE: seq 167, dpid: 0x10, type: 0x02, length: 4, val: 170
> 21:58:58:542 TY_DATA_RESPONSE: seq 167, dpid: 0x10, type: 0x02, length: 4, val: 170
> 21:58:58:543 TY_DATA_RESPONSE: seq 167, dpid: 0x10, type: 0x02, length: 4, val: 170
> 21:58:58:544 TY_DATA_RESPONSE: seq 167, dpid: 0x10, type: 0x02, length: 4, val: 170
> 21:58:58:545 TY_DATA_RESPONSE: seq 167, dpid: 0x10, type: 0x02, length: 4, val: 170
> 21:58:58:546 TY_DATA_RESPONSE: seq 167, dpid: 0x10, type: 0x02, length: 4, val: 170
> 21:58:58:596 Websocket 127.0.0.1:50218 send message: {"config":{"heatsetpoint":1700,"locked":false,"mode":"off","on":true,"reachable":true,"windowopen_set":false},"e":"changed","id":"14","r":"sensors","t":"event","uniqueid":"e8:e0:7e:ff:fe:ce:3d:a7-01-ef00"} (ret = 2130660672)
> 21:58:59:305 poll node a4:c1:38:15:ca:a9:3a:8a-01
> 21:58:59:306 Poll light node Bett Samuel1
> 21:58:59:357 read attributes of 0xA4C13815CAA93A8A cluster: 0x0006: [ 
> 21:58:59:358 0x0000 
> 21:58:59:359 ]
> 21:58:59:360 add task 412992 type 19 to 0xA4C13815CAA93A8A cluster 0x0006 req.id 221
> 21:58:59:361 Poll APS request 221 to 0xA4C13815CAA93A8A cluster: 0x0006
> 21:58:59:520 Poll APS confirm 221 status: 0x00
> 21:58:59:523 Erase task req-id: 221, type: 19 zcl seqno: 2 send time 0, profileId: 0x0104, clusterId: 0x0006
> 21:58:59:580 0xA4C13815CAA93A8A: update ZCL value 0x01/0x0006/0x0000 after 0 s
> 21:59:00:605 poll node 28:2c:02:bf:ff:ed:a4:b3-01
> 21:59:00:606 Poll light node Smart plug 1
> 21:59:00:651 TY_DATA_RESPONSE: seq 
> 167, dpid: 0x10, type: 0x02, length: 4, val: 170
> 21:59:00:653 TY_DATA_RESPONSE: seq 167, dpid: 0x10, type: 0x02, length: 4, val: 170
> 21:59:00:654 TY_DATA_RESPONSE: seq 167, dpid: 0x10, type: 0x02, length: 4, val: 170
> 21:59:00:655 TY_DATA_RESPONSE: seq 167, dpid: 0x10, type: 0x02, length: 4, val: 170
> 21:59:00:656 TY_DATA_RESPONSE: seq 167, dpid: 0x10, type: 0x02, length: 4, val: 170
> 21:59:00:656 TY_DATA_RESPONSE: seq 167, dpid: 0x10, type: 0x02, length: 4, val: 170
> 21:59:00:658 Websocket 127.0.0.1:50218 send message: {"config":{"heatsetpoint":1700,"locked":false,"mode":"off","on":true,"reachable":true,"windowopen_set":false},"e":"changed","id":"14","r":"sensors","t":"event","uniqueid":"e8:e0:7e:ff:fe:ce:3d:a7-01-ef00"} (ret = 2130660672)
> 21:59:00:745 read attributes of 0x282C02BFFFEDA4B3 cluster: 0x0006: [ 
> 21:59:00:749 0x0000 
> 21:59:00:751 ]
> 21:59:00:754 add task 412995 type 19 to 0x282C02BFFFEDA4B3 cluster 0x0006 req.id 227
> 21:59:00:756 Poll APS request 227 to 0x282C02BFFFEDA4B3 cluster: 0x0006
> 21:59:00:960 Poll APS confirm 227 status: 0x00
> 21:59:00:962 Erase task req-id: 227, type: 19 zcl seqno: 3 send time 0, profileId: 0x0104, clusterId: 0x0006
> 21:59:02:280 Daylight now: nauticalDusk, status: 220, daylight: 0, dark: 1
> 21:59:02:380 TY_DATA_RESPONSE: seq 167, dpid: 0x02, type: 0x04, length: 1, val: 0
> 21:59:02:382 TY_DATA_RESPONSE: seq 167, dpid: 0x02, type: 0x04, length: 1, val: 0
> 21:59:02:383 TY_DATA_RESPONSE: seq 167, dpid: 0x02, type: 0x04, length: 1, val: 0
> 21:59:02:386 TY_DATA_RESPONSE: seq 167, dpid: 0x02, type: 0x04, length: 1, val: 0
> 21:59:02:388 TY_DATA_RESPONSE: seq 167, dpid: 0x02, type: 0x04, length: 1, val: 0
> 21:59:02:390 TY_DATA_RESPONSE: seq 167, dpid: 0x02, type: 0x04, length: 1, val: 0
> 21:59:02:457 Websocket 127.0.0.1:50218 send message: {"config":{"heatsetpoint":1700,"locked":false,"mode":"auto","on":true,"reachable":true,"windowopen_set":false},"e":"changed","id":"14","r":"sensors","t":"event","uniqueid":"e8:e0:7e:ff:fe:ce:3d:a7-01-ef00"} (ret = 2130660672)
> 21:59:02:833 poll node b4:35:22:ff:fe:c1:75:8e-01
> 21:59:02:834 Poll light node Huhn_ak_Klappe
> 21:59:03:180 Poll APS request to 0xB43522FFFEC1758E cluster: 0x0008 dropped, cluster doesn't exist
> 21:59:04:905 poll node bc:02:6e:ff:fe:c3:da:66-01
> 21:59:04:906 Poll light node Jalousie_Samuel

Smanar, 2024-04-06:
>I started the LOG and Switch the TRV from Manual to Holiday mode. After a few seconds the TRV Switches to Home mode with 17°C . Maybe it is normal. Is this the LOG sequens you want to see?

Yes exactly, but if the device do same without deconz intervention, IDK what to do, perhaps need another setting somewhere ?
If I m right "holiday" is the "program" mode ?

> And how can I pair the Door Sensor and the Multisensor with the TRV?

Tuya support direct bind for this feature ? Never see it on my side. And I don't think it's something possible using battery device.

>Is it possible to send a weekly program to the TRV?

Good question, have started using c++ code, but using DDF will be another story, honnestly I don't have enought time to do that on my side.

>the battery level is schown as low.

And no way to know if it's not working or reversed, as the device make too few report for it to see it in logs.
Wait somes days for the device make report.
If it's reversed :

```
        {
          "name": "state/lowbattery",
          "parse": {"fn": "tuya", "dpid": 35, "eval": "Item.val = Attr.val == 0"},
          "read": {"fn": "none"}
        },
```