dresden-elektronik/deconz-rest-plugin #6635: Hama Smart Radiator _TZE200_h4cgnbzg

Timo-Weike, 2022-12-30:
## Device

- Product name: Hama Smart Radiator Thermostat
- Manufacturer: _TZE200_h4cgnbzg
- Model identifier: TS0601
- Device type :
  - Thermostat

## Screenshots

[image]

### Basic
[image]

### Groups
[image]

### Scenes
[image]

### Other clusters that are not mentioned above
#### Tuya specific
[image]

#### OTAU
[image]
[image]

#### Time
[image]

Timo-Weike, 2023-01-08:
[Someone](https://github.com/zigpy/zha-device-handlers/issues/1901#issuecomment-1374482799) found a solution to get this working with ZHA. Maybe this is also the problem with deconz.

closed 2023-02-06 by github-actions[bot]

CaptainFuture78, 2023-02-08:
Hello,
I have the same problem with the HAMA Radiator Thermostat.
If there is a soulution for this issue, does anybody know where I can find it?
More information on this issue would be very helpful.

ralfhille, 2023-04-08:
Hello,
I have the same problem with the HAMA Radiator Thermostat.

Have unfortunately seen too late that there is a problem with the Hama radiator thermostats in version 3. My purchase decision was based on some articles with Home Assistant and deconz that the devices can be recognized and controlled.

Is then planned on the part of Dresden Elektonik here still offer support for the devices?

More information on this issue would be very helpful.

Mimiix, 2023-04-08:
More often then not, its the community adding support for devices 😅. @Smanar are you able to help out ?

Smanar, 2023-04-08:
There is a new Pr for this devices https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6859