dresden-elektronik/deconz-rest-plugin #6637: Aqara smart plug (EU) not reporting kWh and W through deCONZ

gvssr, 2022-12-30:
## Describe the bug

Aqara smart plug (EU) version with power metering is not reporting kWh and W sensors, even though you can see the stats being updated in deCONZ REST API. This issue is forwarded from https://github.com/home-assistant/core/issues/84867 by request due to https://github.com/home-assistant/core/issues/84867#issuecomment-1368066490

## Steps to reproduce the behavior

1. Add Aqara smart plug in deCONZ
2. Read all required attributes
3. Add to Home Assistant through official integration
4. Plug in device that draws power

## Expected behavior

Current W draw is reported, like it used to before the introduction of DDF
## Screenshots

[image]
[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26720700
 - deCONZ version: 2.19.03 / 9/19/2022
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Mimiix, 2022-12-30:
What's the manufacturer I'd and model id? I believe this is a duplicate.

gvssr, 2022-12-31:
> What's the manufacturer I'd and model id? I believe this is a duplicate.

[image]

Mimiix, 2022-12-31:
Seems that this should be fixed before by a pr by @SwoopX .

cappadanna, 2023-01-20:
Same issue here with same plug and firmware

gvssr, 2023-01-24:
@ondrackajan I've noticed for me while trying to add / remove / add the plug there's a huge delay in reported Wattage in deCONZ API vs. what shows up in the Home Assistant frontend.

Through DDF: 5-10 minute delay before changed values show up
Without DDF: Updated every 1-5 seconds

BabaIsYou, 2023-01-27:
> Through DDF: 5-10 minute delay before changed values show up

With  `"refresh.interval": 300` in DDF it seems normal tome to have a refresh at 5mn

Rob787, 2023-01-27:
Same issue here with 0w and 0 kwH readings

Mimiix, 2023-01-27:
Did all of you check https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-smart-plugs-not-working-correctly

Rob787, 2023-01-27:
@Mimiix Yep, I'm on latest version already and resetted multiple times:

[image]

Rob787, 2023-01-27:
I did notice following behaviour:

Initially HASS only get the Temp and Power (wattage) entity. Then the Power seems to get updated. After a while, and a HASS restart, the Consumption entity also returns, and then Power and Consumption remain on 0.

Mimiix, 2023-01-27:
> @Mimiix Yep, I'm on latest version already and resetted multiple times:
> 
> [image]

Also the conbee / raspbee ?

Rob787, 2023-01-27:
> > @Mimiix Yep, I'm on latest version already and resetted multiple times:
> > [image]
> 
> Also the conbee / raspbee ?

Yep Conbee II:

[image]

Mimiix, 2023-01-27:
> > > @Mimiix Yep, I'm on latest version already and resetted multiple times:
> > > [image]
> > 
> > 
> > Also the conbee / raspbee ?
> 
> Yep Conbee II:
> 
> [image]

In that case I'm out of options 😅

Smanar, 2023-01-27:
Can you check using the flag "DDF" in logs if the device use the DDF or not for power ?

Some device use native cluster/attribute, some others use the Xiaomi one.

On the actual DDF for "lumi.plug.maeu01" , the DDF chek for the presence of cluster 0x1f and 0x15 to be used
It use the Xiaomi cluster for current, voltage, and cluster 0x000C /attribute 0x0055 (from cluister 0x1F) for power and consumption (with the same value).

So first what is working ?

gvssr, 2023-02-13:
This issue actually pushed me over to migrating to Zigbee2Mqtt, solving my problem. I'll keep this issue open for others, but am unsubscribing as it's no longer applicable for me.

closed 2023-03-15 by github-actions[bot]