dresden-elektronik/deconz-rest-plugin #6639: Philips Hue Tap Dial Switch: rotary events sometimes negative for right rotation

cmuellner, 2023-01-01:
I've joined a new Philips Tap Dial Switch to my home network (bought at Amazon).
Deconz acts as ZigBee bridge and forwards events to Home Assistant.

Observed behavior:
When rotating right the Philips Tap Dial Switch, the reported event includes a positive or negative value in the rotation field.
When rotating left the Philips Tap Dial Switch, the reported event includes a negative value in the rotation field.

Expected behavior:
When rotating right the Philips Tap Dial Switch, the reported event includes a positive value in the rotation field.

The values are observed with HA (listen to events -> "deconz_relative_rotary_event").
The negative values when rotating right appear when rotating moderately or fast.

Deconz and HA run in containers (managed by Podman) on an x86_64 machine with Fedora Linux 37.
Deconz Version: 2.19.03 / 9/19/2022 (FW 26660700)
Home Assistant 2022.12.8

Mimiix, 2023-01-01:
Hi,

As this issue is not as described in #5113 (and not following the template) I'm closing this issue.

If you believe it's a bug, open a bug report following the template or post on the forums if it's only related to your environment.

Kind regards,
Dennis

closed 2023-01-01 by Mimiix

ebaauw, 2023-01-27:
Probably same root cause as #6670.

cmuellner, 2023-01-27:
Thanks for letting me know!