dresden-elektronik/deconz-rest-plugin #6641: Danfoss eTRV0100 thermostat missing mandatory SystemMode (0x001C) attribute in json api

danieldemus, 2023-01-02:
## Describe the bug
The following json response from a Danfoss Ally eTRV0100 thermostat should contain a "mode" attribute containing "off" or "heat".
```
{
    "config": {
        "battery": 63,
        "displayflipped": false,
        "externalsensortemp": -8000,
        "externalwindowopen": false,
        "heatsetpoint": 2000,
        "locked": false,
        "mountingmode": false,
        "offset": 0,
        "on": true,
        "reachable": true,
        "schedule": {},
        "schedule_on": false
    },
    "ep": 1,
    "etag": "b735466c24b7cd120b4bf45b595cd4ae",
    "lastannounced": null,
    "lastseen": "2023-01-02T19:21Z",
    "manufacturername": "Danfoss",
    "modelid": "eTRV0100",
    "name": "Thermostat, Stue højre",
    "state": {
        "errorcode": "0",
        "lastupdated": "2023-01-02T18:30:21.028",
        "mountingmodeactive": false,
        "on": true,
        "temperature": 1997,
        "valve": 25,
        "windowopen": "Quarantine"
    },
    "swversion": "01.18.0008 01.18",
    "type": "ZHAThermostat",
    "uniqueid": "0c:43:14:ff:fe:c7:1c:27-01-0201"
}
```
## Steps to reproduce the behavior
Observe in deconz that the thermostat has a value for the 0x0201 cluster 0x001C attribute.
Using the phoscon app get the a json response for a eTRV0100.
Observe the lack of such a value in the json response.

## Expected behavior
There is a "mode" attribute matching the value in the 0x0201 Thermostat cluster for 0x001C System mode.

## Screenshots
This is the relevant part of the cluster info for one of the thermostats
[image]

## Environment
 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26720700
 - deCONZ version: 2.19.03
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
HTTP logs the above json response.

## Additional con
deconz-rest-plugin/devices/danfoss/etrv0100_thermostat.json doesn't include "config/mode" in it's items, thermostat.json does.

SwoopX, 2023-01-02:
The reason not to include that attribute is that it is pointless to expose something which only allows for a single value, `heat`. In other words: there are no other valid values than heat.

closed 2023-01-02 by SwoopX

danieldemus, 2023-01-03:
It's a mandatory attribute that is expected to be there consumers. Fx. the openhab deconz plugin expects it, so all danfoss thermostats are always in the unknown state.
All the mandatory attributes from the zigbee specification should be exported in the json.