dresden-elektronik/deconz-rest-plugin #6644: TMZ-02 mini a.k.a Tmz02L-16A-W ( Manu: _TZ3000_v7sopte0, Model: TS0011 )

abrakhim, 2023-01-04:
I searched previous Device Request and found some similar ones but with different Manufacturer-Id, starting _TZ3000_ but different Suffix. 

## Device

- Product name: TMZ-02 Mini / Tmz02L-16A-W
- Manufacturer: _TZ3000_v7sopte0
- Model identifier: TS0011
- Device type : Please remove all unrelated device types. 
  - Switch

## Screenshots

Node Info panel
[image]

Note that it states 'Battery' as Power Source but it is connected to Mains! 
### Basic
[image]
### Identify
[image]
### Groups
[image]
### Scenes
[image]
### On/Off
[image]

BabaIsYou, 2023-01-17:
I don't really know if state/on is based on Tuya cluster on not then I suggest to try this DDF that could be adapted if On/Off state doesn't respond.

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_v7sopte0",
  "modelid": "TS0011",
  "product": "TMZ-02 Mini",
  "sleeper": false,
  "status": "Silver",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 300,
          "read": {
            "fn": "tuya"
          },
          "write": {
            "dpid": 1,
            "dt": "0x10",
            "eval": "Item.val == 1 ? 1 : 0;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 1,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}

closed 2023-02-15 by github-actions[bot]

abrakhim, 2023-04-04:
@BabaIsYou Thanks for your reply, i edited the ddl, the light shows up in phoscon, but i can't toggle it, nor does it show the current state. Any other ideas?

BabaIsYou, 2023-04-05:
It will be easier than I initialy thought because there is already a DDF file for this kind of device (#6764 )
Then could you try this one ? 
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZ3000_txpirhfq","_TZ3000_qmi1cfuq", "_TZ3000_ji4araar","_TZ3000_v7sopte0"],
  "modelid": ["TS0011", "TS0011", "TS0011", "TS0011"],
  "product": "1 gang switch module ",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "refresh.interval": 86400,
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

abrakhim, 2023-04-05:
Hahaha, i just read through https://dresden-elektronik.github.io/deconz-dev-doc/modules/ddf/#ddf-file-content
and figured it out myself (without the sw-version-script... ), come back to report here and find your reply...

So, dunno about your attr/swversion-Part, i went with state/on -> looks like this and works:
`{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_v7sopte0",
  "modelid": "TS0011",
  "vendor": "Tuya Industries",
  "product": "TS0011",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/ts0011.json",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5,
          "read": {
            "at": "0x0000",
            "cl": "0x0006",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0006",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        }
      ]
    }
  ]
}`

Thanks!

BabaIsYou, 2023-04-05:
Fine !
Information you added to `state/on` are useless because they are already included in the generic definition of that attribute : https://github.com/dresden-elektronik/deconz-rest-plugin/blob/6c7ce32e403641e91e5746a18734651d73868881/devices/generic/items/state_on_item.json . That's why they are not in mine.

For indentation, I pass my json thru an online free json validator that do the trick (prettyu sure you can find one by yourself).

abrakhim, 2023-04-05:
Thanks again, i feel a bit more enlightened now, escpecially in my kitchen, when the ceiling lights turn on as if by magic ;-)
Will format my code snippet properly and incorporate your suggested changes.

BabaIsYou, 2023-04-11:
> Indentation was just the markdown, i used the button <> instead of writing three ticks. That removed the correct indentation like this:

Just do a "return" after the ''' won't do the trick ?

PR #6866 (2023-04-17): DDF for Tuya TMZ02L-16A-W (_TZ3000_v7sopte0)
Adding "_TZ3000_v7sopte0" and renaming to be more "generic" for TS0011 tuya device.
Fix #6644